emelianov/modbus-esp8266 #1: Modbus TCP client Write Multiple Register --- strange behavior ---

elsabz, 2018-04-13:
Hi emelianov,

thanks for sharing this library...
I have tested an ESP32 with the following program, I am very happy to see that it works very well. I tried several modbus TCP client (master) and unfortunately I discovered a problem.

When I write from client bit1 true, with a polling time of 0.1 seconds, the reading of the state of this bit is not stable, in the sense that sometimes it works well, but very often it starts to malfunction. The strange behavior is that only when I read (in the arduino program) the bit1 of the first word 110 that is not written to 1 by the client is read to 0 by the arduino program see `bitRead (TgrPak_RX.CMD, 1)`

If I change the BIT1 with any other word BIT, it works perfectly even with 0.1 second polling.

I do not understand where the problem is, could you help me?

`#ifdef ESP8266
 #include <ESP8266WiFi.h>
#else
 #include <WiFi.h>
#endif
#include <ModbusIP_ESP8266.h>

// Modbus Registers Offsets (0-9999)
const int TEST_HREG = 100;
//ModbusIP object
ModbusIP mb;

const int LED_PIN = 2;

typedef struct
{
  uint16_t CMD; // Word di comando (16 bit comandi)
  unsigned long Tempo; // Tempo in millisecondi
  uint16_t ris1; // Riserva
  uint16_t ris2; // Riserva 
}
TelegrammaDef_TX;
TelegrammaDef_TX TgrPak_TX;

typedef struct
{
  uint16_t CMD; // Word di comando (16 bit comandi)
  unsigned long Tempo; // Tempo in millisecondi
  uint16_t ris1; // Riserva
  uint16_t ris2; // Riserva 
}
TelegrammaDef_RX;
TelegrammaDef_RX TgrPak_RX;

 // NETWORK: Static IP details...
  IPAddress ip(192, 168, 2, 191);
  IPAddress gateway(192, 168, 2, 1);
  IPAddress subnet(255, 255, 255, 0);

static bool MFP1 ; // Merker Fronte di salita
unsigned long MemoTempo;
uint16_t buff_TX[sizeof(TgrPak_TX)/2];
uint16_t buff_RX[sizeof(TgrPak_RX)/2];

void setup() {
  Serial.begin(74880);
  pinMode(LED_PIN,OUTPUT);
  
  // Static IP Setup Info Here...
  WiFi.config(ip, gateway, subnet);
  WiFi.begin("SSID", "password");
  
  while (WiFi.status() != WL_CONNECTED) {
    delay(500);
    Serial.print(".");
  }

   Serial.println("");
  Serial.println("WiFi connected");  
  Serial.println("IP address: ");
  Serial.println(WiFi.localIP());

  mb.begin();
  mb.addHreg(TEST_HREG, 0x0000, 20);
}
 
void loop() {
   //Call once inside loop() - all magic here
   mb.task();

   // Write
   uint16_t word_app=TgrPak_TX.Tempo>>16;
   mb.Hreg(100, TgrPak_TX.CMD);
   mb.Hreg(101, word_app); 
   mb.Hreg(102, (uint16_t)TgrPak_TX.Tempo); 
   
   // Read
   TgrPak_RX.CMD = mb.Hreg(110);

  // Watchdog
  if(bitRead(TgrPak_RX.CMD, 0)) bitSet(TgrPak_TX.CMD, 0);
  else bitClear(TgrPak_TX.CMD, 0);

  // Test 1
  if(bitRead(TgrPak_RX.CMD, 1)){
    bitSet(TgrPak_TX.CMD, 1);
    digitalWrite(LED_PIN, HIGH);
  }
  else {
    bitClear(TgrPak_TX.CMD, 1);
    digitalWrite(LED_PIN, LOW);  
  }

  if(!MFP1 && bitRead(TgrPak_RX.CMD, 1)){
    MemoTempo = millis();   
    MFP1=true;
  }
  if(!bitRead(TgrPak_RX.CMD, 1)) MFP1=false;
  if(MFP1) TgrPak_TX.Tempo = millis() - MemoTempo ;
  
} // Loop`

emelianov, 2018-04-13:
Hello,
As quick suggestion try to add
delay(10);
to loop (you can play with delay value).
I note that at least with ESP8266 instant pooling of tcp state may lead to unexpected errors.

emelianov, 2018-06-03:
No responce. Closed.

closed 2018-06-03 by emelianov