emelianov/modbus-esp8266 #3: Modbus TCP/IP over Ethernet

bertybert, 2018-09-04:
Does this library allow modbus tcp/ip over ethernet or is it ONLY wifi?

Currently using https://github.com/andresarmento/modbus-arduino but want to move to an ESP32 with an ethernet shield.

Thanks

emelianov, 2018-09-05:
It's complex question not only related to the library.
1. Andrea's library you are using should work with ethernet shield (WizNet) using ethernet library. Not sure about ESP32 but ethernet library port works with ESP8266.
2. It seems to me using ethernet shield is not very best idea. Take look at [Ethernet PHY compatible with ESP32](https://hackaday.com/2017/04/18/enabling-ethernet-on-the-esp32/). This way you will use full TCP stack from ESP SDK (instead of WizNet's).
3. If you decide to keep ethernet shield, ESP32 looks overkilling -- use ESP8266 instead.
4. Arduino core for ESP8266 works rock solid. ESP32's one is not good (at least by now).

bertybert, 2018-09-05:
thanks for the response.

my issue is that the arduino isn't fast enough for my project at only 8bit/16mhz
I would switch to the esp8266 but I need ethernet.
I could perhaps add ethernet to an esp8266 but your library examples do not show how to use ethernet only wifi.

closed 2018-09-20 by emelianov