emelianov/modbus-esp8266 #4: Examples not compiling

bertybert, 2018-09-10:
I'm getting this type of error when trying to compile the examples

`error: 'vector' in namespace 'std' does not name a template type`

emelianov, 2018-09-10:
Fixed. <vector> and <algorithm> libraries was missing in headers. Now code can be compiled against ESP32.

closed 2018-09-20 by emelianov

EdenSilveira, 2018-11-01:
Hello mister !

First of all, congratulations for your initiative and work.
I'm very excited to test you libraryes for a project I´m doing here.
but...
I´m having some problems when trying compiling any of the aplication examples with "Arduino IDE 1.8.7.

"
In file included from C:\Program Files (x86)\Arduino\libraries\modbus-esp8266-master\src/ModbusIP_ESP8266.h:8:0,

                 from C:\Users\Eden\Documents\Arduino\Programas\Led-modbus-esp8266\Led-modbus-esp8266.ino:18:

C:\Program Files (x86)\Arduino\libraries\modbus-esp8266-master\src/Modbus.h:37:16: warning: 'typedef' was ignored in this declaration [enabled by default]

 typedef struct TRegister;

                ^

C:\Program Files (x86)\Arduino\libraries\modbus-esp8266-master\src/Modbus.h:65:2: warning: 'typedef' was ignored in this declaration [enabled by default]

 };

  ^

C:\Program Files (x86)\Arduino\libraries\modbus-esp8266-master\src/Modbus.h:72:1: warning: 'typedef' was ignored in this declaration [enabled by default]

 };

 ^

C:\Program Files (x86)\Arduino\libraries\modbus-esp8266-master\src/Modbus.h:82:2: warning: 'typedef' was ignored in this declaration [enabled by default]

 };

  ^

In file included from C:\Users\Eden\Documents\Arduino\Programas\Led-modbus-esp8266\Led-modbus-esp8266.ino:18:0:

C:\Program Files (x86)\Arduino\libraries\modbus-esp8266-master\src/ModbusIP_ESP8266.h:38:2: warning: 'typedef' was ignored in this declaration [enabled by default]

 };

  ^

C:\Program Files (x86)\Arduino\libraries\modbus-esp8266-master\src/ModbusIP_ESP8266.h:50:2: warning: 'typedef' was ignored in this declaration [enabled by default]

  };

  ^
"
and...

"
In file included from C:\Program Files (x86)\Arduino\libraries\modbus-esp8266-master\src\Modbus.cpp:6:0:

C:\Program Files (x86)\Arduino\libraries\modbus-esp8266-master\src\Modbus.h:37:16: warning: 'typedef' was ignored in this declaration [enabled by default]

 typedef struct TRegister;

                ^

C:\Program Files (x86)\Arduino\libraries\modbus-esp8266-master\src\Modbus.h:65:2: warning: 'typedef' was ignored in this declaration [enabled by default]

 };

  ^

C:\Program Files (x86)\Arduino\libraries\modbus-esp8266-master\src\Modbus.h:72:1: warning: 'typedef' was ignored in this declaration [enabled by default]

 };

 ^

C:\Program Files (x86)\Arduino\libraries\modbus-esp8266-master\src\Modbus.h:82:2: warning: 'typedef' was ignored in this declaration [enabled by default]

 };

  ^

In file included from C:\Program Files (x86)\Arduino\libraries\modbus-esp8266-master\src\ModbusIP_ESP8266.h:8:0,

                 from C:\Program Files (x86)\Arduino\libraries\modbus-esp8266-master\src\ModbusIP_ESP8266.cpp:6:

C:\Program Files (x86)\Arduino\libraries\modbus-esp8266-master\src/Modbus.h:37:16: warning: 'typedef' was ignored in this declaration [enabled by default]

 typedef struct TRegister;

                ^

C:\Program Files (x86)\Arduino\libraries\modbus-esp8266-master\src/Modbus.h:65:2: warning: 'typedef' was ignored in this declaration [enabled by default]

 };

  ^

C:\Program Files (x86)\Arduino\libraries\modbus-esp8266-master\src/Modbus.h:72:1: warning: 'typedef' was ignored in this declaration [enabled by default]

 };

 ^

C:\Program Files (x86)\Arduino\libraries\modbus-esp8266-master\src/Modbus.h:82:2: warning: 'typedef' was ignored in this declaration [enabled by default]

 };

  ^

C:\Program Files (x86)\Arduino\libraries\modbus-esp8266-master\src\Modbus.cpp: In member function 'void Modbus::masterPDU(uint8_t*, uint8_t*, TAddress, void*)':

C:\Program Files (x86)\Arduino\libraries\modbus-esp8266-master\src\Modbus.cpp:489:39: warning: pointer of type 'void *' used in arithmetic [-Wpointer-arith]

                     output = output + 2;

                                       ^

C:\Program Files (x86)\Arduino\libraries\modbus-esp8266-master\src\Modbus.cpp:539:39: warning: pointer of type 'void *' used in arithmetic [-Wpointer-arith]

                     output = output + 2;

                                       ^

In file included from C:\Program Files (x86)\Arduino\libraries\modbus-esp8266-master\src\ModbusIP_ESP8266.cpp:6:0:

C:\Program Files (x86)\Arduino\libraries\modbus-esp8266-master\src\ModbusIP_ESP8266.h:38:2: warning: 'typedef' was ignored in this declaration [enabled by default]

 };

  ^

C:\Program Files (x86)\Arduino\libraries\modbus-esp8266-master\src\ModbusIP_ESP8266.h:50:2: warning: 'typedef' was ignored in this declaration [enabled by default]

  };

  ^
"

I´m sorry if this text became to long.
I think the problem could be related with the "TRegister" struct declaration.

thank you in advance

regards

EdenSilveira, 2018-11-02:
Hello again

Problem solved here. There is some diferences betwen C and C++, this is why I was confused and asking for help.
Originally the 'typedef struc" definititions were like:

typedef struct Point
{
    int x;
    int y;
};

Then, I've modified to something like this:

typedef struct Point
{
    int x;
    int y;
}
Point;

I think that there is no need to use "typedef struct TRegister"
Could be "struct TRegister"

What you think ?

I still have some warnings in compilation:
C:\Program Files (x86)\Arduino\libraries\modbus-esp8266-master\src\Modbus.cpp: In member function 'void Modbus::masterPDU(uint8_t*, uint8_t*, TAddress, void*)':

C:\Program Files (x86)\Arduino\libraries\modbus-esp8266-master\src\Modbus.cpp:489:39: warning: pointer of type 'void *' used in arithmetic [-Wpointer-arith]

                     output = output + 2;

but it´s another problem, I will work about.

The modified libraries are attached.
[modbus-esp8266-emelianov.zip](https://github.com/emelianov/modbus-esp8266/files/2540616/modbus-esp8266-emelianov.zip)

Fell free to comment.

regards

emelianov, 2018-11-05:
Hello Eden,

You are right. I’ll remove ’typedef’ to make code a bit clearer.