emelianov/modbus-esp8266 #7: ESP8266 MEMORY ADDRESS

methaqali, 2019-01-17:
Thank you for this great library.
I have ESP8266 programmed to operate at MODBUS TCP / IP and PLC (LSIS) connected to router by Ethernet MODBUS. I made connection between them successfully (using P2P service found in PLC) through router, but the connection produce error. P2P service used to make connection between LSIS devices and other devices. This service requires inter the read and write memory address for ESP8266 with condition that the address must has 5 digit. Knowing that I use simple diode connection on GPIO pin-14. The problem is when I search for ESP8266 memory map addresses, I find it (0x60000360) 8 digit and this return error from PLC side. I dont know how to make reduction for ESP8266 register address .
Please help me.

emelianov, 2019-01-18:
Hello,
I'm not exactly sure that understanding you needs right, but it seems to me that you need to create at ESP side Modbus Coil with address 12345 (whatever 0..65535) and add callbacks to get/set pin value on external modbus requests. (Code is very alike Callback sample just need to add onGetCoil).
From other hand your PLC probably assume using Hreg not Coil. So you can try to implement code described above for Hreg (just replace Coil with Hreg).

methaqali, 2019-01-18:
Thank you emelianov for replay,
I tried above code, it return illegal data error when use modbus poll and CAS Mod scaner softwares (the blue led in ESP8266 continues glow).

methaqali, 2019-01-18:
I confused what the mean of length and CAS modbus already add 1 offset to address it refusing change offset to (0)

emelianov, 2019-01-18:
For code from Callback.ino example (coil address is 100) you need following CAS settings:
[image]

methaqali, 2019-01-18:
I tried to connect use callback.ino but connection failed.Are you sure this library working well?
[image]

methaqali, 2019-01-18:
this is the error seen in the IDE serial 
[image]

emelianov, 2019-01-18:
I've got simular exception with Arduino/Core **2.5.Beta2**.
Try to use **2.4.2** -- Callback.ino runs stable that case. I will try to investigate the problem.

methaqali, 2019-01-18:
I was try this code its working perfect, my problem only in know to inter the correct plc address in modbus poll software to operate diode(say in GPIO 14)

/*
 * More Tutorials: 
 * Website http://trialcommand.com
 * In English: http://en.trialcommand.com
 * En Español: http://en.trialcommand.com 
 */
#include <ESP8266WiFi.h>

const char* ssid = "MH_Ext";
const char* password = "74375325";
int ModbusTCP_port = 502;

//////// Required for Modbus TCP / IP /// Requerido para Modbus TCP/IP /////////
#define maxInputRegister 20
#define maxHoldingRegister 20

#define MB_FC_NONE 0
#define MB_FC_READ_REGISTERS 3 //implemented
#define MB_FC_WRITE_REGISTER 6 //implemented
#define MB_FC_WRITE_MULTIPLE_REGISTERS 16 //implemented
//
// MODBUS Error Codes
//
#define MB_EC_NONE 0
#define MB_EC_ILLEGAL_FUNCTION 1
#define MB_EC_ILLEGAL_DATA_ADDRESS 2
#define MB_EC_ILLEGAL_DATA_VALUE 3
#define MB_EC_SLAVE_DEVICE_FAILURE 4
//
// MODBUS MBAP offsets
//
#define MB_TCP_TID 0
#define MB_TCP_PID 2
#define MB_TCP_LEN 4
#define MB_TCP_UID 6
#define MB_TCP_FUNC 7
#define MB_TCP_REGISTER_START 8
#define MB_TCP_REGISTER_NUMBER 10

byte ByteArray[260];
unsigned int MBHoldingRegister[maxHoldingRegister];

//////////////////////////////////////////////////////////////////////////

WiFiServer MBServer(ModbusTCP_port);

void setup() {

 pinMode(14, OUTPUT);
 
 Serial.begin(9600);
 delay(100) ;
 WiFi.begin(ssid, password);
 delay(100) ;
 Serial.println(".");
 while (WiFi.status() != WL_CONNECTED) {
 delay(500);
 Serial.print(".");
 }
 MBServer.begin();
 Serial.println("Connected ");
 Serial.print("ESP8266 Slave Modbus TCP/IP ");
 Serial.print(WiFi.localIP());
 Serial.print(":");
 Serial.println(String(ModbusTCP_port));
 Serial.println("Modbus TCP/IP Online");
 
 }
void loop() {
 // Check if a client has connected // Modbus TCP/IP
 WiFiClient client = MBServer.available();
 if (!client) {
 return;
 }
  boolean flagClientConnected = 0;
 byte byteFN = MB_FC_NONE;
 int Start;
 int WordDataLength;
 int ByteDataLength;
 int MessageLength;
  // Modbus TCP/IP
 while (client.connected()) {
  if(client.available())
 {
 flagClientConnected = 1;
 int i = 0;
 while(client.available())
 {
 ByteArray[i] = client.read();
 i++;
 }
 client.flush();
///// code here --- codigo aqui
 ///////// Holding Register [0] A [9] = 10 Holding Registers Escritura
 ///////// Holding Register [0] A [9] = 10 Holding Registers Writing
  MBHoldingRegister[0] = random(0,12);
 MBHoldingRegister[1] = random(0,12);
 MBHoldingRegister[2] = random(0,12);
 MBHoldingRegister[3] = random(0,12);
 MBHoldingRegister[4] = random(0,12);
 MBHoldingRegister[5] = random(0,12);
 MBHoldingRegister[6] = random(0,12);
 MBHoldingRegister[7] = random(0,12);
 MBHoldingRegister[8] = random(0,12);
 MBHoldingRegister[9] = random(0,12);
 ///////// Holding Register [10] A [19] = 10 Holding Registers Lectura
 ///// Holding Register [10] A [19] = 10 Holding Registers Reading
  int Temporal[10];
  Temporal[0] = MBHoldingRegister[10];
 Temporal[1] = MBHoldingRegister[11];
 Temporal[2] = MBHoldingRegister[12];
 Temporal[3] = MBHoldingRegister[13];
 Temporal[4] = MBHoldingRegister[14];
 Temporal[5] = MBHoldingRegister[15];
 Temporal[6] = MBHoldingRegister[16];
 Temporal[7] = MBHoldingRegister[17];
 Temporal[8] = MBHoldingRegister[18];
 Temporal[9] = MBHoldingRegister[19];
 /// Enable Output 14
 digitalWrite(14, MBHoldingRegister[14] );
 //// debug
 for (int i = 0; i < 10; i++) {
 Serial.print("[");
 Serial.print(i);
 Serial.print("] ");
 Serial.print(Temporal[i]);
  }
 Serial.println("");
//// end code - fin 
  //// rutine Modbus TCP
 byteFN = ByteArray[MB_TCP_FUNC];
 Start = word(ByteArray[MB_TCP_REGISTER_START],ByteArray[MB_TCP_REGISTER_START+1]);
 WordDataLength = word(ByteArray[MB_TCP_REGISTER_NUMBER],ByteArray[MB_TCP_REGISTER_NUMBER+1]);
 }
  // Handle request
 switch(byteFN) {
 case MB_FC_NONE:
 break;
  case MB_FC_READ_REGISTERS: // 03 Read Holding Registers
 ByteDataLength = WordDataLength * 2;
 ByteArray[5] = ByteDataLength + 3; //Number of bytes after this one.
 ByteArray[8] = ByteDataLength; //Number of bytes after this one (or number of bytes of data).
 for(int i = 0; i < WordDataLength; i++)
 {
 ByteArray[ 9 + i * 2] = highByte(MBHoldingRegister[Start + i]);
 ByteArray[10 + i * 2] = lowByte(MBHoldingRegister[Start + i]);
 }
 MessageLength = ByteDataLength + 9;
 client.write((const uint8_t *)ByteArray,MessageLength);
  byteFN = MB_FC_NONE;
  break;
  case MB_FC_WRITE_REGISTER: // 06 Write Holding Register
 MBHoldingRegister[Start] = word(ByteArray[MB_TCP_REGISTER_NUMBER],ByteArray[MB_TCP_REGISTER_NUMBER+1]);
 ByteArray[5] = 6; //Number of bytes after this one.
 MessageLength = 12;
 client.write((const uint8_t *)ByteArray,MessageLength);
 byteFN = MB_FC_NONE;
 break;
  case MB_FC_WRITE_MULTIPLE_REGISTERS: //16 Write Holding Registers
 ByteDataLength = WordDataLength * 2;
 ByteArray[5] = ByteDataLength + 3; //Number of bytes after this one.
 for(int i = 0; i < WordDataLength; i++)
 {
 MBHoldingRegister[Start + i] = word(ByteArray[ 13 + i * 2],ByteArray[14 + i * 2]);
 }
 MessageLength = 12;
 client.write((const uint8_t *)ByteArray,MessageLength); 
 byteFN = MB_FC_NONE;
  break;
 }
 }
}

emelianov, 2019-01-18:
This code should work as code you provided above.

`#ifdef ESP8266
 #include <ESP8266WiFi.h>
#else //ESP32
 #include <WiFi.h>
#endif
#include <ModbusIP_ESP8266.h>

//Modbus Registers Offsets (0-65535)
const int LED_HREG = 14;
//Used Pins
#ifdef ESP8266
 const int ledPin = 14; // Builtin ESP8266 LED
#else
 const int ledPin = TX; // ESP32 TX LED
#endif
//ModbusIP object
ModbusIP mb;

// Callback function for write (set) Coil. Returns value to store.
uint16_t cbWrite(TRegister* reg, uint16_t val) {
  digitalWrite(ledPin, val);
  return val;
}

uint16_t cbRead(TRegister* reg, uint16_t val) {
  return digitalRead(ledPin);
}

// Callback function for client connect. Returns true to allow connection.
bool cbConn(IPAddress ip) {
  Serial.println(ip);
  return true;
}
 
void setup() {
 #ifdef ESP8266
  Serial.begin(74880);
 #else
  Serial.begin(115200);
 #endif
  WiFi.begin("EW", "iMpress6264");
  
  while (WiFi.status() != WL_CONNECTED) {
    delay(500);
    Serial.print(".");
  }
 
  Serial.println("");
  Serial.println("WiFi connected");  
  Serial.print("IP address: ");
  Serial.println(WiFi.localIP());
  
  mb.onConnect(cbConn);   // Add callback on connection event
  mb.begin();

  pinMode(ledPin, OUTPUT);
  mb.addHreg(LED_HREG);
  mb.onSetHreg(LED_HREG, cbWrite);
  //mb.onGetHreg(LED_HREG, cbRead); // This line links register value to real pin status.
}

void loop() {
   //Call once inside loop() - all magic here
   mb.task();
   delay(100);
}`

methaqali, 2019-01-18:
ok thanks , what about pre-mapping between the MODBUS data model and ESP8266 device application. Are you have document for this issue.

emelianov, 2019-01-19:
Library responsible only for Modbus layer. It may function absolutely separate from application logic.
Application should use call backs or checking/setting registers from main loop.

methaqali, 2019-01-20:
Thats right I have succeed in send address from cas modbus to ESP make diode on - off but same address send by PLC with out make the same diode to operate, I dont know  why axcactly!!

emelianov, 2019-01-21:
Try to catch request data that PLC sends to slave. You can install for example PeakHMI Modbus Slave Simulator  (https://www.hmisys.com) and look Communication Monitor output.

emelianov, 2019-01-21:
> error: Failed to open COM5
Arduino IDE doesn't see ESP board at COM5 or port is busy by another application. May be you need just retry FW upload attempt.
Check Tools\Port settings. Probably board port has been changed.

methaqali, 2019-01-21:
I mean this type of error "Invalid library found in C:\Users\lenovo\Documents\Arduino\libraries\OneWire-master: no headers files (.h)"
its not effect the uploading of program but annoy me.

methaqali, 2019-01-21:
emelianov, Could you help me:
I have the following code to make MODBUS tcp/ip(its work good) the other issue in the code is ultrasonic sensor (trig on D1& echo on D2 and output distance on serial port)I want to write distance on one of memory register say (3). I use the command "digitalWrite(distance, MBHoldingRegister[3] )" its fail then I use analogWrite also fail what's the solution?

/*
 * More Tutorials: 
 * Website http://trialcommand.com
 * In English: http://en.trialcommand.com
 * En Español: http://en.trialcommand.com 
 */
#include <ESP8266WiFi.h>
#define trigPin D1
#define echoPin D2
#define analog_output D5

float distance;
long duration;
int dataIn;
int m = 0;
int h = 0;
int t = 0;

const char* ssid = "MH_Ext";
const char* password = "74375325";
int ModbusTCP_port = 502;

//////// Required for Modbus TCP / IP /// Requerido para Modbus TCP/IP /////////
#define maxInputRegister 20
#define maxHoldingRegister 20

#define MB_FC_NONE 0
#define MB_FC_READ_REGISTERS 3 //implemented
#define MB_FC_WRITE_REGISTER 6 //implemented
#define MB_FC_WRITE_MULTIPLE_REGISTERS 16 //implemented
//
// MODBUS Error Codes
//
#define MB_EC_NONE 0
#define MB_EC_ILLEGAL_FUNCTION 1
#define MB_EC_ILLEGAL_DATA_ADDRESS 2
#define MB_EC_ILLEGAL_DATA_VALUE 3
#define MB_EC_SLAVE_DEVICE_FAILURE 4
//
// MODBUS MBAP offsets
//
#define MB_TCP_TID 0
#define MB_TCP_PID 2
#define MB_TCP_LEN 4
#define MB_TCP_UID 6
#define MB_TCP_FUNC 7
#define MB_TCP_REGISTER_START 8
#define MB_TCP_REGISTER_NUMBER 10

byte ByteArray[260];
unsigned int MBHoldingRegister[maxHoldingRegister];

//////////////////////////////////////////////////////////////////////////

WiFiServer MBServer(ModbusTCP_port);

void setup() {

 pinMode(14, OUTPUT);
 pinMode(trigPin, OUTPUT); 
 pinMode(echoPin, INPUT);
 Serial.begin(9600);
 delay(100) ;
 WiFi.begin(ssid, password);
 delay(100) ;
 Serial.println(".");
 while (WiFi.status() != WL_CONNECTED) {
 delay(500);
 Serial.print(".");
 }
 MBServer.begin();
 Serial.println("Connected ");
 Serial.print("ESP8266 Slave Modbus TCP/IP ");
 Serial.print(WiFi.localIP());
 Serial.print(":");
 Serial.println(String(ModbusTCP_port));
 Serial.println("Modbus TCP/IP Online");
 
 }
void loop() {
distance = getDistance();
Serial.println(distance);
delay(1000);
 // Check if a client has connected // Modbus TCP/IP
 WiFiClient client = MBServer.available();
 if (!client) {
 return;
 }
  boolean flagClientConnected = 0;
 byte byteFN = MB_FC_NONE;
 int Start;
 int WordDataLength;
 int ByteDataLength;
 int MessageLength;
  // Modbus TCP/IP
 while (client.connected()) {
  if(client.available())
 {
 flagClientConnected = 1;
 int i = 0;
 while(client.available())
 {
 ByteArray[i] = client.read();
 i++;
 }
 client.flush();
///// code here --- codigo aqui
 ///////// Holding Register [0] A [9] = 10 Holding Registers Escritura
 ///////// Holding Register [0] A [9] = 10 Holding Registers Writing
 
 /// Enable Output 14
 digitalWrite(14, MBHoldingRegister[14] );
 digitalWrite(distance, MBHoldingRegister[12] );

//// end code - fin 
  //// rutine Modbus TCP
 byteFN = ByteArray[MB_TCP_FUNC];
 Start = word(ByteArray[MB_TCP_REGISTER_START],ByteArray[MB_TCP_REGISTER_START+1]);
 WordDataLength = word(ByteArray[MB_TCP_REGISTER_NUMBER],ByteArray[MB_TCP_REGISTER_NUMBER+1]);
 }
  // Handle request
 switch(byteFN) {
 case MB_FC_NONE:
 break;
  case MB_FC_READ_REGISTERS: // 03 Read Holding Registers
 ByteDataLength = WordDataLength * 2;
 ByteArray[5] = ByteDataLength + 3; //Number of bytes after this one.
 ByteArray[8] = ByteDataLength; //Number of bytes after this one (or number of bytes of data).
 for(int i = 0; i < WordDataLength; i++)
 {
 ByteArray[ 9 + i * 2] = highByte(MBHoldingRegister[Start + i]);
 ByteArray[10 + i * 2] = lowByte(MBHoldingRegister[Start + i]);
 }
 MessageLength = ByteDataLength + 9;
 client.write((const uint8_t *)ByteArray,MessageLength);
  byteFN = MB_FC_NONE;
  break;
  case MB_FC_WRITE_REGISTER: // 06 Write Holding Register
 MBHoldingRegister[Start] = word(ByteArray[MB_TCP_REGISTER_NUMBER],ByteArray[MB_TCP_REGISTER_NUMBER+1]);
 ByteArray[5] = 6; //Number of bytes after this one.
 MessageLength = 12;
 client.write((const uint8_t *)ByteArray,MessageLength);
 byteFN = MB_FC_NONE;
 break;
  case MB_FC_WRITE_MULTIPLE_REGISTERS: //16 Write Holding Registers
 ByteDataLength = WordDataLength * 2;
 ByteArray[5] = ByteDataLength + 3; //Number of bytes after this one.
 for(int i = 0; i < WordDataLength; i++)
 {
 MBHoldingRegister[Start + i] = word(ByteArray[ 13 + i * 2],ByteArray[14 + i * 2]);
 }
 MessageLength = 12;
 client.write((const uint8_t *)ByteArray,MessageLength); 
 byteFN = MB_FC_NONE;
  break;
 }
 }

 }
 //===== getDistance - Custom Function
int getDistance() {
  // Clears the trigPin
  digitalWrite(trigPin, LOW);
  delayMicroseconds(2);
  // Sets the trigPin on HIGH state for 10 micro seconds
  digitalWrite(trigPin, HIGH);
  delayMicroseconds(10);
  digitalWrite(trigPin, LOW);
  // Reads the echoPin, returns the sound wave travel time in microseconds
  duration = pulseIn(echoPin, HIGH);
  // Calculating the distance
  distance = duration * 0.034 / 2; // distance in cm
  return distance;
}

methaqali, 2019-01-22:
emelianov, Could you help me:
I have the following code to make MODBUS tcp/ip(its work good) the other issue in the code is ultrasonic sensor (trig on D1& echo on D2 and output distance on serial port)I want to write distance on one of memory register say (3). I use the command "digitalWrite(distance, MBHoldingRegister[3] )" its fail then I use analogWrite also fail what's the solution?

/*
 * More Tutorials: 
 * Website http://trialcommand.com
 * In English: http://en.trialcommand.com
 * En Español: http://en.trialcommand.com 
 */
#include <ESP8266WiFi.h>
#define trigPin D1
#define echoPin D2
#define analog_output D5

float distance;
long duration;
int dataIn;
int m = 0;
int h = 0;
int t = 0;

const char* ssid = "MH_Ext";
const char* password = "74375325";
int ModbusTCP_port = 502;

//////// Required for Modbus TCP / IP /// Requerido para Modbus TCP/IP /////////
#define maxInputRegister 20
#define maxHoldingRegister 20

#define MB_FC_NONE 0
#define MB_FC_READ_REGISTERS 3 //implemented
#define MB_FC_WRITE_REGISTER 6 //implemented
#define MB_FC_WRITE_MULTIPLE_REGISTERS 16 //implemented
//
// MODBUS Error Codes
//
#define MB_EC_NONE 0
#define MB_EC_ILLEGAL_FUNCTION 1
#define MB_EC_ILLEGAL_DATA_ADDRESS 2
#define MB_EC_ILLEGAL_DATA_VALUE 3
#define MB_EC_SLAVE_DEVICE_FAILURE 4
//
// MODBUS MBAP offsets
//
#define MB_TCP_TID 0
#define MB_TCP_PID 2
#define MB_TCP_LEN 4
#define MB_TCP_UID 6
#define MB_TCP_FUNC 7
#define MB_TCP_REGISTER_START 8
#define MB_TCP_REGISTER_NUMBER 10

byte ByteArray[260];
unsigned int MBHoldingRegister[maxHoldingRegister];

//////////////////////////////////////////////////////////////////////////

WiFiServer MBServer(ModbusTCP_port);

void setup() {

 pinMode(14, OUTPUT);
 pinMode(trigPin, OUTPUT); 
 pinMode(echoPin, INPUT);
 Serial.begin(9600);
 delay(100) ;
 WiFi.begin(ssid, password);
 delay(100) ;
 Serial.println(".");
 while (WiFi.status() != WL_CONNECTED) {
 delay(500);
 Serial.print(".");
 }
 MBServer.begin();
 Serial.println("Connected ");
 Serial.print("ESP8266 Slave Modbus TCP/IP ");
 Serial.print(WiFi.localIP());
 Serial.print(":");
 Serial.println(String(ModbusTCP_port));
 Serial.println("Modbus TCP/IP Online");
 
 }
void loop() {
distance = getDistance();
Serial.println(distance);
delay(1000);
 // Check if a client has connected // Modbus TCP/IP
 WiFiClient client = MBServer.available();
 if (!client) {
 return;
 }
  boolean flagClientConnected = 0;
 byte byteFN = MB_FC_NONE;
 int Start;
 int WordDataLength;
 int ByteDataLength;
 int MessageLength;
  // Modbus TCP/IP
 while (client.connected()) {
  if(client.available())
 {
 flagClientConnected = 1;
 int i = 0;
 while(client.available())
 {
 ByteArray[i] = client.read();
 i++;
 }
 client.flush();
///// code here --- codigo aqui
 ///////// Holding Register [0] A [9] = 10 Holding Registers Escritura
 ///////// Holding Register [0] A [9] = 10 Holding Registers Writing
 
 /// Enable Output 14
 digitalWrite(14, MBHoldingRegister[14] );
 digitalWrite(distance, MBHoldingRegister[12] );

//// end code - fin 
  //// rutine Modbus TCP
 byteFN = ByteArray[MB_TCP_FUNC];
 Start = word(ByteArray[MB_TCP_REGISTER_START],ByteArray[MB_TCP_REGISTER_START+1]);
 WordDataLength = word(ByteArray[MB_TCP_REGISTER_NUMBER],ByteArray[MB_TCP_REGISTER_NUMBER+1]);
 }
  // Handle request
 switch(byteFN) {
 case MB_FC_NONE:
 break;
  case MB_FC_READ_REGISTERS: // 03 Read Holding Registers
 ByteDataLength = WordDataLength * 2;
 ByteArray[5] = ByteDataLength + 3; //Number of bytes after this one.
 ByteArray[8] = ByteDataLength; //Number of bytes after this one (or number of bytes of data).
 for(int i = 0; i < WordDataLength; i++)
 {
 ByteArray[ 9 + i * 2] = highByte(MBHoldingRegister[Start + i]);
 ByteArray[10 + i * 2] = lowByte(MBHoldingRegister[Start + i]);
 }
 MessageLength = ByteDataLength + 9;
 client.write((const uint8_t *)ByteArray,MessageLength);
  byteFN = MB_FC_NONE;
  break;
  case MB_FC_WRITE_REGISTER: // 06 Write Holding Register
 MBHoldingRegister[Start] = word(ByteArray[MB_TCP_REGISTER_NUMBER],ByteArray[MB_TCP_REGISTER_NUMBER+1]);
 ByteArray[5] = 6; //Number of bytes after this one.
 MessageLength = 12;
 client.write((const uint8_t *)ByteArray,MessageLength);
 byteFN = MB_FC_NONE;
 break;
  case MB_FC_WRITE_MULTIPLE_REGISTERS: //16 Write Holding Registers
 ByteDataLength = WordDataLength * 2;
 ByteArray[5] = ByteDataLength + 3; //Number of bytes after this one.
 for(int i = 0; i < WordDataLength; i++)
 {
 MBHoldingRegister[Start + i] = word(ByteArray[ 13 + i * 2],ByteArray[14 + i * 2]);
 }
 MessageLength = 12;
 client.write((const uint8_t *)ByteArray,MessageLength); 
 byteFN = MB_FC_NONE;
  break;
 }
 }

 }
 //===== getDistance - Custom Function
int getDistance() {
  // Clears the trigPin
  digitalWrite(trigPin, LOW);
  delayMicroseconds(2);
  // Sets the trigPin on HIGH state for 10 micro seconds
  digitalWrite(trigPin, HIGH);
  delayMicroseconds(10);
  digitalWrite(trigPin, LOW);
  // Reads the echoPin, returns the sound wave travel time in microseconds
  duration = pulseIn(echoPin, HIGH);
  // Calculating the distance
  distance = duration * 0.034 / 2; // distance in cm
  return distance;
}
Duplicate of #

closed 2019-01-22 by methaqali

emelianov, 2019-01-22:
Just replace
distance = getDistance();
with 
MBHoldingRegister[< desired HReg # to store distance (< 20) >] = getDistance();

methaqali, 2019-01-22:
I'm so grateful for you my friend.

methaqali, 2019-01-23:
Hi emelianov, I need to use static IP for ESP8266 and add the following commands to upper code after that ESP connected with router but plc and modbus software both not connected with ESP and give error "time out"
IPAddress staticIP224_60(192,168,1,60);
IPAddress gateway224_60(192,168,1,1);
IPAddress subnet224_60(255,255,255,0);

void setup()
{
  Serial.begin(9600);
  WiFi.config(staticIP224_60, gateway224_60, subnet224_60)
what is the wrong?