emelianov/modbus-esp8266 #8: static IPfor ESP8266

methaqali, 2019-01-25:
I need to use static IP for ESP8266 and add the following  commands: 
 staticIP224_60(192,168,1,60); 
IPAddress gateway224_60(192,168,1,1); 
IPAddress subnet224_60(255,255,255,0); 
void setup() { Serial.begin(9600); 
WiFi.config(staticIP224_60, gateway224_60, subnet224_60) what is the wrong?

after that ESP connected with router but plc and  modbus poll software both not connected with ESP and give error "time out" IPAddress.  
Can you help me?

emelianov, 2019-01-25:
Following code should work:
```
IPAddress staticIP224_60(192,168,1,60);
IPAddress gateway224_60(192,168,1,1);
IPAddress subnet224_60(255,255,255,0);
void setup() {
 Serial.begin(9600);
 WiFi.config(staticIP224_60, gateway224_60, subnet224_60);
 WiFi.begin(ssid, password);
 //rest of init
}
```

methaqali, 2019-01-26:
I doing the same but ESP after connect to router can't connect to plc!

emelianov, 2019-01-26:
Verify that address (192.168.1.60) not used by another device. Check ESP device availability from PC using CLI command
ping 192.168.1.60

methaqali, 2019-01-27:
I checked it there ping available from plc to ESP

methaqali, 2019-01-28:
I find the mistake, its when change IP of ESP inside PLC mod bus Chanel , I forget to write change to PLC.
Thanks for cooperation my friend.

closed 2019-01-28 by methaqali