emelianov/modbus-esp8266 #12: ModbusTCP readHreg failed, but connection to slave is ok

RottenVanSuti, 2019-03-19:
Hi, @emelianov  thanks for your library. I'm doing some testing code. I'm setting esp32 as Master and connects to the slave address just fine. However, when I try to read Holding registers the cbTransaction triggers and shows error 0X04 which I read it means fail to proces request. I'm using modsim as slave and it shows the connection with the esp32 and the "read data" state. I've put values on modsim registers as float32, decimal,hex and none of those seems to be read by the esp.
`
#ifdef ESP8266
#include <ESP8266WiFi.h>
#else
#include <WiFi.h>
#endif
#include <ModbusIP_ESP8266.h>

const int REG = 0;                   // Modbus Hreg Offset
IPAddress remote(192, 168, 43, 107); // 
const int LOOP_COUNT = 10;

ModbusIP test; 

void setup()
{

    Serial.begin(115200);

    WiFi.begin("RottenVanSuit", "12345678");

    while (WiFi.status() != WL_CONNECTED)
    {
        delay(500);
        Serial.print(".");
    }

    Serial.println("");
    Serial.println("WiFi connected");
    Serial.println("IP address: ");
    Serial.println(WiFi.localIP());

    test.master();
}
bool cb(Modbus::ResultCode event, uint16_t transactionId, void *data)
{
    if (event != Modbus::EX_SUCCESS)
        Serial.printf("Modbus result: %02X\n", event);
    if (event == Modbus::EX_TIMEOUT)
    {
        Serial.println("Timeout");
        //mb.disconnect(remote);
        //mb.dropTransactions();
    }
    return true;
}
uint16_t res = 0;
uint8_t show = LOOP_COUNT;
uint32_t ctr = 0;
void loop()
{
    if (test.isConnected(remote))
    { // Check if connection to Modbus Slave is established
        if (!test.readHreg(remote, REG, &res, 1, cb))
            Serial.print("#"); // Initiate Read Coil from Modbus Slave
        ctr++;
    }
    else
    {
        test.connect(remote); 
        Serial.println("reconnecting");
    }
    test.task();  
    //delay(100); // Pulling interval
    if (!show--)
    { 
        Serial.printf("\n%u: %d",ctr,res);
        res = 0;
        show = LOOP_COUNT;
    }
    delay(1000);
}
`

RottenVanSuti, 2019-03-19:
`#ifdef ESP8266
#include <ESP8266WiFi.h>
#else
#include <WiFi.h>
#endif
#include <ModbusIP_ESP8266.h>

const int REG = 0;                   // Modbus Hreg Offset
IPAddress remote(192, 168, 43, 107); // 
const int LOOP_COUNT = 10;

ModbusIP test; 

void setup()
{

    Serial.begin(115200);

    WiFi.begin("RottenVanSuit", "12345678");

    while (WiFi.status() != WL_CONNECTED)
    {
        delay(500);
        Serial.print(".");
    }

    Serial.println("");
    Serial.println("WiFi connected");
    Serial.println("IP address: ");
    Serial.println(WiFi.localIP());

    test.master();
}
bool cb(Modbus::ResultCode event, uint16_t transactionId, void *data)
{
    if (event != Modbus::EX_SUCCESS)
        Serial.printf("Modbus result: %02X\n", event);
    if (event == Modbus::EX_TIMEOUT)
    {
        Serial.println("Timeout");
        //mb.disconnect(remote);
        //mb.dropTransactions();
    }
    return true;
}
uint16_t res = 0;
uint8_t show = LOOP_COUNT;
uint32_t ctr = 0;
void loop()
{
    if (test.isConnected(remote))
    { // Check if connection to Modbus Slave is established
        if (!test.readHreg(remote, REG, &res, 1, cb))
            Serial.print("#"); // Initiate Read Coil from Modbus Slave
        ctr++;
    }
    else
    {
        test.connect(remote); 
        Serial.println("reconnecting");
    }
    test.task();  
    //delay(100); // Pulling interval
    if (!show--)
    { 
        Serial.printf("\n%u: %d",ctr,res);
        res = 0;
        show = LOOP_COUNT;
    }
    delay(1000);
}
``

emelianov, 2019-03-20:
Hello.
Decrease delay(1000) to say delay(100).
The reason is that transaction timeout is 1000mS so pause between test.task() calls should be less than 1000mS.

closed 2019-04-27 by emelianov

RottenVanSuti, 2019-04-28:
Thank you @emelianov

RottenVanSuti, 2019-07-26:
Hi Again @emelianov  I still have an issue with the timeout. The ESP32 as master trying to read holding registers from software ModSim, can't get the values. This software starts with 40001. It connects correctly to ModSim, but Timeout still appears even putting delay with 100 ms. I post the code.

RottenVanSuti, 2019-07-26:
`#include <Arduino.h>
#include <Preferences.h>
#include <WiFi.h>
#include <ModbusIP_ESP8266.h>
#include "soc/timer_group_struct.h"
#include "soc/timer_group_reg.h"
IPAddress remote(192, 168, 20, 231);
const int LOOP = 10;
bool Setup = false;
ModbusIP test;
void ondisconnect()
{
    Serial.println("disconnected from server");
}
TaskHandle_t Modbus;
void setup()
{
    Serial.begin(115200);
    WiFi.begin("RED_INTER_TIC", "R3dt1c2017");

    while (WiFi.status() != WL_CONNECTED)
    {
        delay(500);
        Serial.print(".");
    }
    test.master();
    // test.onDisconnect();
    // test.onConnect();
}
uint8_t show = LOOP;
bool cb(Modbus::ResultCode event, uint16_t transactionId, void *data)
{
    if (event != Modbus::EX_SUCCESS)
        Serial.printf("Modbus result: %02X\n", event);
    if (event == Modbus::EX_TIMEOUT)
    {
        Serial.println("Timeout");
        //test.disconnect(remote);
        //test.dropTransactions();
    }
    return true;
}
void loop()
{
    if (test.isConnected(remote))
    {
        test.readHreg(remote, 40001, &res, 6, cb);
    }
    else
    {
        test.connect(remote, 502);
    }
    if (!show--)
    {
        Serial.println(res);
        show = LOOP;
    }
    test.task();
    delay(100);
}`