emelianov/modbus-esp8266 #15: Call input register by modbus  from ESP8266 or Arduino D1

methaqali, 2019-05-23:
Dear emelianov
Thank you very much for your replay- I am so grateful for you.
I apply your advice in my code shown below, but I get this error message:
(( exit status 1
'MB' was not declared in this scope ))
When declare MB as " unsigned int MB;"
The compiler give me
((request for member 'Ireg' in 'MB', which is of non-class type 'unsigned int'
))
My code is:

#include <ESP8266WiFi.h>
#define analog_output D5
#include "Adafruit_VL53L0X.h"
Adafruit_VL53L0X lox = Adafruit_VL53L0X();
float distance;
long duration;
int dataIn;
const char* ssid = "MH_Ext";
const char* password = "74375325";
int ModbusTCP_port = 502;
//////// Required for Modbus TCP / IP ///
#define maxInputRegister 20
#define maxHoldingRegister 20
#define MB_FC_NONE 0
#define MB_FC_READ_REGISTERS 3 //implemented
#define MB_FC_WRITE_REGISTER 6 //implemented
#define MB_FC_WRITE_MULTIPLE_REGISTERS 16 //implemented
// MODBUS Error Codes
#define MB_EC_NONE 0
#define MB_EC_ILLEGAL_FUNCTION 1
#define MB_EC_ILLEGAL_DATA_ADDRESS 2
#define MB_EC_ILLEGAL_DATA_VALUE 3
#define MB_EC_SLAVE_DEVICE_FAILURE 4
// MODBUS MBAP offsets
#define MB_TCP_TID 0
#define MB_TCP_PID 2
#define MB_TCP_LEN 4
#define MB_TCP_UID 6
#define MB_TCP_FUNC 7
#define MB_TCP_REGISTER_START 8
#define MB_TCP_REGISTER_NUMBER 10
byte ByteArray[260];
unsigned int MBHoldingRegister[maxHoldingRegister];
//////////////////////////////////////////////////////////////////////////
WiFiServer MBServer(ModbusTCP_port);
IPAddress staticIP224_60(192,168,1,60);
IPAddress gateway224_60(192,168,1,1);
IPAddress subnet224_60(255,255,255,0);
void setup() {
//pinMode(14, OUTPUT);
Serial.begin(115200);
// wait until serial port opens for native USB devices
while (! Serial) {
delay(1);
}

Serial.println("Adafruit VL53L0X test");
if (!lox.begin()) {
Serial.println(F("Failed to boot VL53L0X"));
while(1);
}
// power
Serial.println(F("VL53L0X API Simple Ranging example\n\n"));
WiFi.config(staticIP224_60, gateway224_60, subnet224_60);
WiFi.begin(ssid, password); //rest of init
delay(100) ;
WiFi.begin(ssid, password);
delay(100) ;
Serial.println(".");
while (WiFi.status() != WL_CONNECTED) {
delay(500);
Serial.print(".");
}
MBServer.begin();
Serial.println("Connected ");
Serial.print("ESP8266 Slave Modbus TCP/IP ");
Serial.print(WiFi.localIP());
Serial.print(":");
Serial.println(String(ModbusTCP_port));
Serial.println("Modbus TCP/IP Online");
}
void loop() { //read from VL53LOX
VL53L0X_RangingMeasurementData_t measure;

Serial.print("Reading a measurement... ");
lox.rangingTest(&measure, false); // pass in 'true' to get debug data printout!

if (measure.RangeStatus != 4) { // phase failures have incorrect data
Serial.print("Distance (mm): "); Serial.println(measure.RangeMilliMeter);
MB.Ireg(6, measure.RangeMilliMeter); // Assign value to Ireg
} else {
Serial.println(" out of range ");
MB.Ireg(6, 0); // Set 0 on error
}
delay(1000);
// Check if a client has connected // Modbus TCP/IP
WiFiClient client = MBServer.available();
if (!client) {
return;
}
boolean flagClientConnected = 0;
byte byteFN = MB_FC_NONE;
int Start;
int WordDataLength;
int ByteDataLength;
int MessageLength;
// Modbus TCP/IP
while (client.connected()) {
if(client.available())
{
flagClientConnected = 1;
int i = 0;
while(client.available())
{
ByteArray[i] = client.read();
i++;
}
client.flush();
///////// Holding Register [0] A [9] = 10 Holding Registers Writing
//// end code - fin
//// rutine Modbus TCP
byteFN = ByteArray[MB_TCP_FUNC];
Start = word(ByteArray[MB_TCP_REGISTER_START],ByteArray[MB_TCP_REGISTER_START+1]);
WordDataLength = word(ByteArray[MB_TCP_REGISTER_NUMBER],ByteArray[MB_TCP_REGISTER_NUMBER+1]);
}
// Handle request
switch(byteFN) {
case MB_FC_NONE:
break;
case MB_FC_READ_REGISTERS: // 03 Read Holding Registers
ByteDataLength = WordDataLength * 2;
ByteArray[5] = ByteDataLength + 3; //Number of bytes after this one.
ByteArray[8] = ByteDataLength; //Number of bytes after this one (or number of bytes of data).
for(int i = 0; i < WordDataLength; i++)
{
ByteArray[ 9 + i * 2] = highByte(MBHoldingRegister[Start + i]);
ByteArray[10 + i * 2] = lowByte(MBHoldingRegister[Start + i]);
}
MessageLength = ByteDataLength + 9;
client.write((const uint8_t *)ByteArray,MessageLength);
byteFN = MB_FC_NONE;
break;
case MB_FC_WRITE_REGISTER: // 06 Write Holding Register
MBHoldingRegister[Start] = word(ByteArray[MB_TCP_REGISTER_NUMBER],ByteArray[MB_TCP_REGISTER_NUMBER+1]);
ByteArray[5] = 6; //Number of bytes after this one.
MessageLength = 12;
client.write((const uint8_t *)ByteArray,MessageLength);
byteFN = MB_FC_NONE;
break;
case MB_FC_WRITE_MULTIPLE_REGISTERS: //16 Write Holding Registers
ByteDataLength = WordDataLength * 2;
ByteArray[5] = ByteDataLength + 3; //Number of bytes after this one.
for(int i = 0; i < WordDataLength; i++)
{
MBHoldingRegister[Start + i] = word(ByteArray[ 13 + i * 2],ByteArray[14 + i * 2]);
}
MessageLength = 12;
client.write((const uint8_t *)ByteArray,MessageLength);
byteFN = MB_FC_NONE;
break;
}
}

}

emelianov, 2019-05-27:
I'm afraid issue is not the Modbus library related. Code you provided used library-less implementation of ModbusIP so you need to modify code itself it to implement features required. It seems to me it'll be easer for you to find somone to make this changes or guide you on how to do it.
In case you want to use the Modbus library code needed to be rewritten from scratch.

methaqali, 2019-05-27:
The same modbus code used with ultrasonic sensor without problem if you remember https://github.com/emelianov/modbus-esp8266/issues/7

emelianov, 2019-05-31:
It's absolutaly out of the library scope.
According to your code implementation holding regs are used.
```c
#include <ESP8266WiFi.h>
#define analog_output D5
#include <Adafruit_VL53L0X.h>
Adafruit_VL53L0X lox = Adafruit_VL53L0X();
float distance;
long duration;
int dataIn;
const char* ssid = "MH_Ext";
const char* password = "74375325";
int ModbusTCP_port = 502;
//////// Required for Modbus TCP / IP ///
#define maxInputRegister 20
#define maxHoldingRegister 20
#define MB_FC_NONE 0
#define MB_FC_READ_REGISTERS 3 //implemented
#define MB_FC_WRITE_REGISTER 6 //implemented
#define MB_FC_WRITE_MULTIPLE_REGISTERS 16 //implemented
// MODBUS Error Codes
#define MB_EC_NONE 0
#define MB_EC_ILLEGAL_FUNCTION 1
#define MB_EC_ILLEGAL_DATA_ADDRESS 2
#define MB_EC_ILLEGAL_DATA_VALUE 3
#define MB_EC_SLAVE_DEVICE_FAILURE 4
// MODBUS MBAP offsets
#define MB_TCP_TID 0
#define MB_TCP_PID 2
#define MB_TCP_LEN 4
#define MB_TCP_UID 6
#define MB_TCP_FUNC 7
#define MB_TCP_REGISTER_START 8
#define MB_TCP_REGISTER_NUMBER 10
byte ByteArray[260];
unsigned int MBHoldingRegister[maxHoldingRegister];
//////////////////////////////////////////////////////////////////////////
WiFiServer MBServer(ModbusTCP_port);
IPAddress staticIP224_60(192,168,1,60);
IPAddress gateway224_60(192,168,1,1);
IPAddress subnet224_60(255,255,255,0);
void setup() {
//pinMode(14, OUTPUT);
Serial.begin(115200);
// wait until serial port opens for native USB devices
while (! Serial) {
delay(1);
}

Serial.println("Adafruit VL53L0X test");
if (!lox.begin()) {
Serial.println(F("Failed to boot VL53L0X"));
while(1);
}
// power
Serial.println(F("VL53L0X API Simple Ranging example\n\n"));
WiFi.config(staticIP224_60, gateway224_60, subnet224_60);
WiFi.begin(ssid, password); //rest of init
delay(100) ;
WiFi.begin(ssid, password);
delay(100) ;
Serial.println(".");
while (WiFi.status() != WL_CONNECTED) {
delay(500);
Serial.print(".");
}
MBServer.begin();
Serial.println("Connected ");
Serial.print("ESP8266 Slave Modbus TCP/IP ");
Serial.print(WiFi.localIP());
Serial.print(":");
Serial.println(String(ModbusTCP_port));
Serial.println("Modbus TCP/IP Online");
}
void loop() { //read from VL53LOX
VL53L0X_RangingMeasurementData_t measure;

Serial.print("Reading a measurement... ");
lox.rangingTest(&measure, false); // pass in 'true' to get debug data printout!

if (measure.RangeStatus != 4) { // phase failures have incorrect data
Serial.print("Distance (mm): "); Serial.println(measure.RangeMilliMeter);
//MB.Ireg(6, measure.RangeMilliMeter); // Assign value to Ireg
MBHoldingRegister[6] = measure.RangeMilliMeter;
} else {
Serial.println(" out of range ");
//MB.Ireg(6, 0); // Set 0 on error
MBHoldingRegister[6] = 0;
}
delay(1000);
// Check if a client has connected // Modbus TCP/IP
WiFiClient client = MBServer.available();
if (!client) {
return;
}
boolean flagClientConnected = 0;
byte byteFN = MB_FC_NONE;
int Start;
int WordDataLength;
int ByteDataLength;
int MessageLength;
// Modbus TCP/IP
while (client.connected()) {
if(client.available())
{
flagClientConnected = 1;
int i = 0;
while(client.available())
{
ByteArray[i] = client.read();
i++;
}
client.flush();
///////// Holding Register [0] A [9] = 10 Holding Registers Writing
//// end code - fin
//// rutine Modbus TCP
byteFN = ByteArray[MB_TCP_FUNC];
Start = word(ByteArray[MB_TCP_REGISTER_START],ByteArray[MB_TCP_REGISTER_START+1]);
WordDataLength = word(ByteArray[MB_TCP_REGISTER_NUMBER],ByteArray[MB_TCP_REGISTER_NUMBER+1]);
}
// Handle request
switch(byteFN) {
case MB_FC_NONE:
break;
case MB_FC_READ_REGISTERS: // 03 Read Holding Registers
ByteDataLength = WordDataLength * 2;
ByteArray[5] = ByteDataLength + 3; //Number of bytes after this one.
ByteArray[8] = ByteDataLength; //Number of bytes after this one (or number of bytes of data).
for(int i = 0; i < WordDataLength; i++)
{
ByteArray[ 9 + i * 2] = highByte(MBHoldingRegister[Start + i]);
ByteArray[10 + i * 2] = lowByte(MBHoldingRegister[Start + i]);
}
MessageLength = ByteDataLength + 9;
client.write((const uint8_t *)ByteArray,MessageLength);
byteFN = MB_FC_NONE;
break;
case MB_FC_WRITE_REGISTER: // 06 Write Holding Register
MBHoldingRegister[Start] = word(ByteArray[MB_TCP_REGISTER_NUMBER],ByteArray[MB_TCP_REGISTER_NUMBER+1]);
ByteArray[5] = 6; //Number of bytes after this one.
MessageLength = 12;
client.write((const uint8_t *)ByteArray,MessageLength);
byteFN = MB_FC_NONE;
break;
case MB_FC_WRITE_MULTIPLE_REGISTERS: //16 Write Holding Registers
ByteDataLength = WordDataLength * 2;
ByteArray[5] = ByteDataLength + 3; //Number of bytes after this one.
for(int i = 0; i < WordDataLength; i++)
{
MBHoldingRegister[Start + i] = word(ByteArray[ 13 + i * 2],ByteArray[14 + i * 2]);
}
MessageLength = 12;
client.write((const uint8_t *)ByteArray,MessageLength);
byteFN = MB_FC_NONE;
break;
}
}

}
```

methaqali, 2019-05-31:
Could you submit a solution for the problem?

emelianov, 2019-05-31:
Try code from my previous post

methaqali, 2019-05-31:
I apply the above code and get the following error in serial monitor!

methaqali, 2019-05-31:
14:02:44.146 -> >>>stack>>>
14:02:44.146 -> 3fff0030:  3fffdad0 00000000 3ffef044 402021ac  
14:02:44.188 -> 3fff0040:  feefeffe feefeffe feefeffe feefeffe  
14:02:44.188 -> 3fff0050:  feefeffe feefeffe feefeffe feefeffe  
14:02:44.188 -> 3fff0060:  feefeffe feefeffe feefeffe feefeffe  
14:02:44.188 -> 3fff0070:  feefeffe feefeffe feefeffe 3ffef070  
14:02:44.188 -> 3fff0080:  3fffdad0 00000000 3ffef068 4020743c  
14:02:44.188 -> 3fff0090:  feefeffe feefeffe 3ffef080 40100718  
14:02:44.188 -> <<<stack<<<
14:02:44.228 -> 
14:02:44.228 ->  ets Jan  8 2013,rst cause:2, boot mode:(3,7)
14:02:44.228 -> 
14:02:44.228 -> load 0x4010f000, len 1384, room 16 
14:02:44.228 -> tail 8
14:02:44.228 -> chksum 0x2d
14:02:44.228 -> csum 0x2d
14:02:44.228 -> v3ffe8ee8
14:02:44.228 -> ~ld
14:02:44.508 -> ⸮Adafruit VL53L0X test
14:02:44.508 -> Failed to boot VL53L0X

emelianov, 2019-05-31:
Output you've sent is only last lines of debug data. Scrollup and find location of code resulting crash. There are lot of debug output points in your code so you can easily do it.
Anyway debug is impossible without having real VL53L0X device.

methaqali, 2019-05-31:
Sorry, there is problem in wiring connection but after correct it, I still facing the same problem(at first VL53LOX) read and stop reading as soon as the PLC connect to it.

emelianov, 2019-05-31:
try to decrease
delay(1000);
to, say, 100 or 200.
Have no other quick ideas.

methaqali, 2019-05-31:
I change the delay as you instructed with same problem see picture below

[image]

emelianov, 2019-06-03:
As far as i can understand from your code measurements stops as ESP device got incoming modbus connection.
You can try to replace (not sure that code will work this way)
*while (client.connected()) {*
with
*if (client.connected()) {*
Otherwise modbus processing code should be refactored.

closed 2019-06-09 by emelianov