emelianov/modbus-esp8266 #17: Help please

ltonini80, 2019-06-08:
I have a plc to ip 192.168.1.200 and id 3. I need to read the holding registry on 30775 address, but is a 32 bit value?? How i can do it??

emelianov, 2019-06-08:
Modbus operates 16bit registers so i suspect that target 32bit value is stored in two sequential holding regs staring from 30775. Probably you'll need to swap high/low words depending of source device settings.  
```c
uint32_t value32;
IPAddress ip(192, 168, 1, 200);
mb.readHreg(ip, 30775, (uint16_t*)value32, 2, nullptr, 3); // Last '3' is device id
//value32 = (value32>>16) | (value32<<16); // Uncomment for words swap
```

ltonini80, 2019-06-08:
Thank so much for fast replay.. 
I've a nodemcu device ("amica version).

I've modify SimpleMasterExemple adding my parameter and the loop function in this mode
....
uint32_t value32;
void loop() { 
  if (mb.isConnected(remote)) {   // Check if connection to Modbus Slave is established   
    mb.readHreg(remote, REG, (uint16_t*)value32,2,nullptr,3);  // Initiate Read Coil from Modbus Slave
  } else {
    mb.connect(remote);           // Try to connect if no connection
  }
  mb.task();                      // Common local Modbus task
  delay(100);                     // Pulling interval
  if (!show--) {                   // Display Slave register value one time per second (with default settings)
    value32 = (value32>>16) | (value32<<16); // Uncomment for words swap
    Serial.println( printf("%" PRIu32 "\n",value32));
    show = LOOP_COUNT;
  }
}

My value expected on serial print is 9301 but I receive 0 or 2.
If i comment the "words swap" not change.
With a program like "modbus master simulator" i read 9301

emelianov, 2019-06-08:
Ops! It's my fault. I've missed & sign. Should be
```c
mb.readHreg(ip, 30775, (uint16_t*)&value32, 2, nullptr, 3); 
```

ltonini80, 2019-06-08:
Fantastic!! Now work perfectly!!
Thank you so much!!

emelianov, 2019-06-09:
You're welcome

closed 2019-06-09 by emelianov