emelianov/modbus-esp8266 #19: Modbus TCP to RTU bridge

andig, 2019-07-03:
I'm interested in using this library as TCP to RTU modbus bridge, i.e.:

  - have modbus TCP master on a different system with any modbus TCP standard library
  - use esp8266 as modbus TCP slave over Wifi
  - have esp8266 as modbus RTU master against slave device connected to esp8266
  - implement bridging behaviour on esp8266

I'm not sure about any timing constraints in modbus, but is this a scenario that could be supported by this library?

emelianov, 2019-07-03:
Hello,
Yes this functionality available out-of-box as registers set is shared between Modbus* instances. I'm planning to ship release version with modbus bridge example.
In general it should be something alike:
```c
#include <ModbusIP_ESP8266.h>
#include <ModbusRTU.h.>
ModbusRTU rtu;
ModbusIP ip;
void setup() {
// network, etc init should be there
 Serial.begin(9600);
 rtu.begin(&Serial);
 rtu.master();
 ip.slave();
 ip.addCoil(0, 10);
}
void loop() {
 if (!rtu.slave())
  rtu.pullCoil(1, 0, 0, 10); // read 10 slave coils starting from 0 and put values to local coils start from 0
 delay(100);
 rtu.task();
 ip.task();
}
```

andig, 2019-07-04:
Great, thank you for the example! It would be great (even if I don't have a use case for that) to allow a tee-like connection, i.e. give the slave to map different device ids to different RTU slaves, not sure this is possible?

One thing I've noticed is that the `ModbusIP_ESP8266` files look odd since you support esp32, too. Wouldn't the familiar name be ModbusTCP?

emelianov, 2019-07-05:
First of all two notes:
- Real bridge code should be a bit more complex (need to manage multiple separate push/pull calls to the same slave)
- Bridge assumed to be working in buffered mode (i.e. IP-side responses and receives data from/to local registers and distributing/receiving to/from RTU slaves in background.

Using Modbus Unit ID as slave ID is impossible without internal changes. It's not big deal. Real problem that i see no way to build clear API for that feature.
So for multiple slave interaction i suggest to use different register offset (i mean IP-side/internal storage).
Also it'll possible to use multiple instances of ModbusIP on different ports (to be implemented).

emelianov, 2020-06-17:
https://github.com/emelianov/modbus-esp8266/blob/master/examples/IP-bridge/IP-bridge.ino

khoa200399, 2021-01-11:
Hello? Can you explain about function pullCoil? I just learned about it

emelianov, 2021-01-11:
Hi
```c
uint16_t pullCoil(uint8_t slaveId, uint16_t from, uint16_t to, uint16_t numregs = 1, cbTransaction cb = nullptr);
uint16_t pullCoil(IPAddress serverIP, uint16_t from, uint16_t to, uint16_t numregs = 1, cbTransaction cb = nullptr);
```

- `slaveId`/`serverIP` read coils from
- `from` first remote register
- `to` first local register to write result to
- `numbers` count of registers to read
- `cb` transaction callback function

Returns:

- `0` on error
- transaction id (for IP) or non zero (for RTU) on success request sent
 
Reads number of coils from remote server.
Result is saved to local coils. Method returns `0` on error, corresponding transaction id (for IP).
Method exits just after request sent. Response data are filled later by .task() method.

khoa200399, 2021-01-11:
Thanks for reply. It's very helpfull. But Can I ask one more question? I try to print the result from 
`readCoil(uint8_t slaveId, uint16_t offset, bool* value, uint16_t numregs, cbTransaction cb)`
I use 
` for(int i = 0; i < 10; i++){
    Serial.print(coils[i],BIN);
    Serial.print("\n");`
but it doesn't work.

emelianov, 2021-01-12:
It's hard to say. Probably wiring problem. Start with [RTU Master Sync](https://github.com/emelianov/modbus-esp8266/blob/master/examples/RTU/masterSync/masterSync.ino) example.

khoa200399, 2021-01-12:
> Hi
>
> ```c
> uint16_t pullCoil(uint8_t slaveId, uint16_t from, uint16_t to, uint16_t numregs = 1, cbTransaction cb = nullptr);
> uint16_t pullCoil(IPAddress serverIP, uint16_t from, uint16_t to, uint16_t [...]

How can I print results to Serial Monitor or what's local coils/registers? 
If I pullCoil from modbus RTU, local coils is on modbusIP-server coils?

emelianov, 2021-01-12:
Read local reg

```c
uint16_t Hreg(uint16_t offset);
bool Coil(uint16_t offset);
bool Ists(uint16_t offset);
uint16_t Ireg(uint16_t offset);
```

- `offset` Address of the register to read

Returns current value of the local register.

Local registers are shared for all ModbusRTU and ModbusTCP instances so you can access it using any instance.
Refere examples for detailes.

khoa200399, 2021-01-12:
Thanks for you help. <3

khoa200399, 2021-01-21:
Can you help me a little problem? The IP bridge example just only can read registers. How can I write to this?

emelianov, 2021-01-21:
If it's needed to only write just replace `pullHreg` with `pushHreg`.

```c
void loop() {
 if (!rtu.slave())
  rtu.pushHreg(SLAVE_ID, REM_HREG, LOC_HREG, 10);
 delay(100);
 rtu.task();
 ip.task();
}
```

In case you need bidirectional data exchange it will be a bit more compelx.

khoa200399, 2021-01-21:
yes, I need to bidirectional data exchange, my project is: The server can read/write the data from/to the modbus devices(slave). How I do it?

emelianov, 2021-01-21:
Talke a look issue #101

khoa200399, 2021-01-21:
'TRegister* ModbusMap::searchRegister(TAddress)' marked override, but does not override
 I compile and this error?

emelianov, 2021-01-21:
Code from last messages of that issue requires latest library commits from master branch.
Probably sketch from first message will be suitable.

khoa200399, 2021-01-21:
How can I download the version 4.0.0?

emelianov, 2021-01-21:
`git clone https://github.com/emelianov/modbus-esp8266.git`
or `Download ZIP` under `Code` button menu

khoa200399, 2021-01-21:
I have done it. Thanks, and how can I read or write Coils?
I try 'uint16_t callbackGet(TRegister* reg, uint16_t val) {
  uint16_t res = 0xBEEF;
  if ( !mbRTU.slave() ) {
    mbRTU.readHreg(SLAVE_ID, mbTCP.lastRegister.address, &res);
    mbRTU.readCoil(SLAVE_ID, mbTCP.lastRegister.address, &res);
    while( mbRTU.slave() ) {
      mbRTU.task();
    }
  }
  return res;'
but it not work?

khoa200399, 2021-01-21:
Thank you, it work. But how can I read/write Coils?

emelianov, 2021-01-21:
Simplest way is to create another pair of callbacks for coil. And also add coil and coil mapping the same way as it done for hregs.

khoa200399, 2021-01-21:
I can write coils but, read it not work

> uint16_t callbackGetCoil(TRegister* reg, uint16_t val) {
  uint16_t res;
  bool res1;
  if ( !mbRTU.slave() ) {
    mbRTU.readCoil(SLAVE_ID, mbTCP.lastRegister.address,&res1,res);
    while( mbRTU.slave() ) {
      mbRTU.task();
    }
  }
  return res;
}

khoa200399, 2021-01-21:
How can I fix that?

emelianov, 2021-01-21:
Callback operates modbus's representation of coil on/off.
Should use macro to convert bool to required form:

```c
return COIL_VAL(res);
```

also there is macro to convert to bool

```c
bool x = COIL_BOOL(val);
```

closed 2021-02-05 by emelianov

fahara02, 2023-03-08:
Hi i have question in a tcp client say a plc ,trying to connect to multple rtu slave (different slave id)  through a single tcp rtu bridge .how send data to different slave as tcp command does not have a slave id in built. am i missing something?

emelianov, 2023-03-31:
Typically UnitId is used to pass slaveId to send request to.