emelianov/modbus-esp8266 #22: Reading multible Holding Registers (Question)

Smis2, 2019-10-20:
I'm trying to read some Holding registers.
for example Register 152. I want to read a 4 bit float. so I have to read numregs=2. But I didnt get it right now. 
Is it possible to do that with the Library?

emelianov, 2019-10-21:
Sure.
```c
uint16_t writeHreg(IPAddress ip, uint16_t offset, uint16_t* value, uint16_t numregs = 1, cbTransaction cb = nullptr, uint8_t unit = MODBUSIP_UNIT);
```
Something alike below. Just probably need to change byte order depending on one at remote side.
```c
union fl {
  float f;
  uint16_t ui[2];
};
fl res;
...
mb.readHreg(remote_ip, 152, res.ui, 2);
...
```

Smis2, 2019-10-22:
It worked just fine,
thank you so much.

closed 2019-10-22 by Smis2