emelianov/modbus-esp8266 #24: Modbus RTU Master and Slave implementation in ESP32

krutarthtrivedi, 2019-11-01:
Hello there,

Thanks for an amazing workaround on providing Modbus RTU support for ESP32.

For one of my project, I need to implement one ESP32 as a Modbus master and another one as a Modbus slave. I tried RTU-master and RTU-slave example by slightly modifying them. I stored values in Slave holding register and tried to read the same by calling readHreg in master. But it's showing some garbage values. I have verified connections and everything looks good. Kindly help me out to resolve this.

Attaching programs for your reference.

This one is of Modbus Slave.

```
#include <ModbusRTU.h>

#define REGN 10
#define SLAVE_ID 1

#define RE_DE 4

ModbusRTU mb;

void setup() {
  Serial.begin(115200);
  mb.begin(&Serial,RE_DE);
  mb.slave(SLAVE_ID);
  mb.addHreg(REGN);
  mb.Hreg(REGN, 100);
  Serial.println(mb.Hreg(REGN));
  Serial.println(mb.slave());
}

void loop() {
  mb.task();
  yield();
}
```

Following one is of Modbus Master.

```
#include <ModbusRTU.h>

ModbusRTU mb;

bool cbWrite(Modbus::ResultCode event, uint16_t transactionId, void* data) {
  Serial.printf_P("Request result: 0x%02X, Mem: %d\n", event, ESP.getFreeHeap());
  return true;
}

void setup() {
  Serial.begin(115200);
  mb.begin(&Serial);
  mb.master();
}

uint16_t coils[20];

void loop() {
  if (!mb.slave()) {
    mb.readHreg(1, 10, coils, 1, cbWrite);
  }
  mb.task();
  yield();
}
```

Thanks,
Krutarth Trivedi.

emelianov, 2019-11-17:
Hello,
1. You are using (at least within code above) Serial for Modbus communication as well aIt s for debug output. This may affect data transfer. For ESP32 it's better to configure and use Serial1, Serial2 for Modbus.
2. It looks like yiu are using RS-485 transiver with auto RX/TX. I had problems with TX speed above 9600 with low-cost such modules type.

Never the less i'll try to simulate the problem in my lab next days (i was busy last weeks).

krutarthtrivedi, 2019-11-19:
Hello @emelianov,

Thanks for the response. It seems there was a problem associated with RS485 transceiver module. After changing it with the new one, everything works well. 

Besides, I have some feedback to share as I am using this library for almost 20 days.

1. I found if there's nothing connected on AB lines and Master tries to pass on any request, Master ESP32 crashes. Eventually, this thing leads to force restart of ESP32.
2. There might be something with event scheduler because it causes force restart when Modbus events are not properly being addressed. 
3. Alongside, if Master is requesting for any register read from Slave and if Slave is having any blocking delay then it also causes Master to force restart.

At this point of time, I am not sure whether all these problems are related to event schedulers or not but just sharing my thoughts on what I found.

Thanks,
Krutarth Trivedi.

emelianov, 2019-11-23:
Krutarth,
I've performed some testing and unable to reproduce problems you described.
Can you write piece of code causing ESP32 crash?
Also can you tell me if you are using RS485 transceiver with auto RX/TX sensing or with manual line control?

emelianov, 2019-11-25:
Krutarth,
Check your code with last commit to master branch. It was bug in write multiple Hregs fixed. Probably crashes was related to the bug.

krutarthtrivedi, 2019-11-27:
> Krutarth,
> Check your code with last commit to master branch. It was bug in write multiple Hregs fixed. Probably crashes was related to the bug.

Hey @emelianov ,
Thanks for this.

I'll try with the last commit and will share the results asap.

Regards,
Krutarth Trivedi.

krutarthtrivedi, 2019-12-02:
Hey @emelianov ,

I tried last commit with my program and I didn't find any issues with my program. 

I have a doubt to discuss with you. 

I am writing Holding registers of two slaves using the following piece of code and It's writing both the slaves properly.
```
    if (!Modbus_Master.slave())
    {
      Modbus_Master.writeHreg(NODE_1_SLAVE_ID, HREG_ADDRESS, dataToNode_1, NULL);
      delay(100);
      Modbus_Master.writeHreg(NODE_2_SLAVE_ID, HREG_ADDRESS, dataToNode_2, NULL);
      delay(100);
    }
```

Now, after writing to slaves, They perform operations on those data and then they set their respective Input status registers (Ists). During that period, the Master is reading respective Ists from both slaves at the certain time interval set using millis() by using the following piece of code. What I found is, Master is detecting only one slave at a time Like reading them individually works good but reading them simultaneously gives the output of very first readIsts request only. 

```
  while (waitingForACK)
  {

    timer_function();

    if (ackCheckTimeout)
    {
      if (!Modbus_Master.slave())
      {
        Modbus_Master.readIsts(NODE_1_SLAVE_ID, ISTS_ADDRESS, &Node_1_ackStatus, 1, NULL);
        delay(100);
        Modbus_Master.readIsts(NODE_2_SLAVE_ID, ISTS_ADDRESS, &Node_2_ackStatus, 1, NULL);
        delay(100);
      }

      Serial.println(Node_1_ackStatus);
      Serial.println(Node_2_ackStatus);

      if (!Node_1_ackStatus && !Node_2_ackStatus)
      {
           readyForNextData = true;
           waitingForACK = false;
      }
      ackCheckTimeout = false;
    }

    Modbus_Master.task();
    yield();
  }
```

The above program returns this:  Modbus_Master.readIsts(NODE_1_SLAVE_ID, ISTS_ADDRESS, &Node_1_ackStatus, 1, NULL); only and not returning value of Modbus_Master.readIsts(NODE_2_SLAVE_ID, ISTS_ADDRESS, &Node_2_ackStatus, 1, NULL); 

Instead of reading both simultaneously, I use the following only, it returns proper output.

```
      if (!Modbus_Master.slave())
      {
        Modbus_Master.readIsts(NODE_1_SLAVE_ID, ISTS_ADDRESS, &Node_1_ackStatus, 1, NULL);
        delay(100);
      }
```

Now, instead of reading the first slave only, I read the second slave only, It returns proper output also.
```
      if (!Modbus_Master.slave())
      {
        Modbus_Master.readIsts(NODE_2_SLAVE_ID, ISTS_ADDRESS, &Node_2_ackStatus, 1, NULL);
        delay(100);
      }
```

I need to read acknowledgement from both the slave devices. So, Is this problem associated with the library or not? Kindly look into it.

Thanks,
Krutarth Trivedi.

emelianov, 2019-12-02:
Hello @krutarthtrivedi ,
Sequental readIsts (readHreg/writeHreg/whatever) unable to run because of ModbusRTU nature. That is request is unable to run until got responce for previuose one. (That's protocol limitation not the library) No requests queue inside library is implemented.
If it suitable to lock execution till responce or timeout you can write:
```c
  while (waitingForACK)
  {

    timer_function();

    if (ackCheckTimeout)
    {
      if (!Modbus_Master.slave())
      {
        Modbus_Master.readIsts(NODE_1_SLAVE_ID, ISTS_ADDRESS, &Node_1_ackStatus, 1, NULL);
        while (Modbus_Master.slave()) {
           Modbus_Master.task();
           delay(1);
        }
        Modbus_Master.readIsts(NODE_2_SLAVE_ID, ISTS_ADDRESS, &Node_2_ackStatus, 1, NULL);
        while (Modbus_Master.slave()) {
           Modbus_Master.task();
           delay(1);
        }
      }

      Serial.println(Node_1_ackStatus);
      Serial.println(Node_2_ackStatus);

      if (!Node_1_ackStatus && !Node_2_ackStatus)
      {
           readyForNextData = true;
           waitingForACK = false;
      }
      ackCheckTimeout = false;
    }

    //Modbus_Master.task();
    yield();
  }
```

krutarthtrivedi, 2019-12-02:
> Hello @krutarthtrivedi ,
> Sequental readIsts (readHreg/writeHreg/whatever) unable to run because of ModbusRTU nature. That is request is unable to run until got responce for previuose one. (That's [...]

Hey @emelianov ,
Thanks for the quick response. I tried the suggested way and ESP32 crashes every time. Find the below-attached image. 

[image]

emelianov, 2019-12-02:
It's my fault. Wrong timeout processing. Fix pushed to master.
Thanks!

krutarthtrivedi, 2019-12-02:
I understood your point that we don't have the request queue feature due to ModbusRTU nature.
Now, I tried the same updated commit and it still gets crashed with the same error.

Besides, I tired the same algorithm for writeHreg and it is also getting crashed the same way as readIsts crash. Find below piece of code I updated for writeHreg, which causes force restart of ESP32.

```
    if (!Modbus_Master.slave())
    {
      Modbus_Master.writeHreg(NODE_1_SLAVE_ID, HREG_ADDRESS, dataToNode_1, NULL);

      while (Modbus_Master.slave()) {
        Modbus_Master.task();
        delay(1);
      }

      Modbus_Master.writeHreg(NODE_2_SLAVE_ID, HREG_ADDRESS, dataToNode_2, NULL);

      while (Modbus_Master.slave()) {
        Modbus_Master.task();
        delay(1);
      }
    }
```

krutarthtrivedi, 2019-12-02:
Hey @emelianov,

I suppose that was due to Modbus traffic. Because I tried adding delay(100); after every Modbus read/write request and it didn't get crash. I am still working on this before concluding anything from my side. I'll keep you posted.

Thanks,
Krutarth Trivedi.

emelianov, 2019-12-02:
Hmm... Probably some problems with ESP32 Serial implementation.
With synthetic testing (see tests) unable to get crash at this point.

emelianov, 2019-12-27:
Hope problem is fixed somehow.
You are welcome to reopen this or create new issue if problem persists.

closed 2019-12-27 by emelianov

LorenzoCerda, 2022-11-16:
Hello everybody,

  I know it has been a while, but nevertheless. Concerning the first post of @krutarthtrivedi, I tried that code (with a MAX485, that uses the DE_RE input) and it didn't work (at the beginning). I checked with an oscilloscope and I was getting something similar to what is asked in this question: https://electronics.stackexchange.com/questions/315078/strange-voltage-levels-in-a-multi-slave-full-duplex-rs485-bus

  Then I realized that the problem might be the setup in the master side. I changed one line and it worked (also checked with the oscilloscope):

```
mb.begin(&Serial1);                           // Original line

mb.begin(&Serial1, PIN_DE_RE);      // The one that worked for me. 
```

  That's it. I leave it here in case it is useful for someone. 

All the best.

lab14-iot, 2024-08-17:
@LorenzoCerda i tried the one workedfor you but while reading with qmodbus it shows error
[image]