emelianov/modbus-esp8266 #26: holding register

methaqali, 2019-11-22:
Hi Emelianov,
Thank you for this great work.
Could you advise me, how to holding the value of [get distance ()] in register 100 of ESP8266 in the following code:

#ifdef ESP8266
 #include <ESP8266WiFi.h>
#else //ESP32
 #include <WiFi.h>
#endif
#include <ModbusIP_ESP8266.h>
const int trigPin = 5;    // pin no. 5 is D1 ESP8266
const int echoPin = 4;    // pin no. 4 is D2 ESP8266
int distance;
long duration;
// Modbus Registers Offsets
const int TEST_HREG = 100;
//ModbusIP object
ModbusIP mb;
IPAddress staticIP224_60(192,168,1,70);
IPAddress gateway224_60(192,168,1,1);
IPAddress subnet224_60(255,255,255,0);
void setup() {
 #ifdef ESP8266
  Serial.begin(9600);
 #else
  Serial.begin(115200);
 #endif
 pinMode(trigPin, OUTPUT); 
 pinMode(echoPin, INPUT);
 WiFi.config(staticIP224_60, gateway224_60, subnet224_60);
  WiFi.begin("----", "------");
  
  while (WiFi.status() != WL_CONNECTED) {
    delay(500);
    Serial.print(".");
  }
 
  Serial.println("");
  Serial.println("WiFi connected");  
  Serial.println("IP address: ");
  Serial.println(WiFi.localIP());

  mb.slave();
  mb.addHreg(TEST_HREG, 0xABCD);
}
 
void loop() {
   //Call once inside loop() - all magic here
   mb.task();
   delay(10);
}

int getDistance() {
// Clears the trigPin
digitalWrite(trigPin, LOW);
delayMicroseconds(2);
// Sets the trigPin on HIGH state for 10 micro seconds
digitalWrite(trigPin, HIGH);
delayMicroseconds(10);
digitalWrite(trigPin, LOW);
// Reads the echoPin, returns the sound wave travel time in microseconds
duration = pulseIn(echoPin, HIGH);
// Calculating the distance
distance = duration * 0.034 / 2; // distance in cm
return distance;
}

emelianov, 2019-11-23:
I think just adding mb.Hreg(dist); should work;
```c
void loop() {
//Call once inside loop() - all magic here
mb.task();
// Clears the trigPin
digitalWrite(trigPin, LOW);
delayMicroseconds(2);
// Sets the trigPin on HIGH state for 10 micro seconds
digitalWrite(trigPin, HIGH);
delayMicroseconds(10);
digitalWrite(trigPin, LOW);
// Reads the echoPin, returns the sound wave travel time in microseconds
duration = pulseIn(echoPin, HIGH);
// Calculating the distance
dist= duration*0.034/2;
// Prints the distance on the Serial Monitor
Serial.print("Distance: ");
Serial.println(dist);
mb.Hreg(dist); // Store distance to hreg
delay(100); // Should decrease loop delay for reliable network communication
}
```

Tomat7, 2019-11-23:
...
Serial.println(dist);
mb.Hreg(TEST_HREG, dist); // Store distance to hreg :-)
delay(100); // Should decrease loop delay for reliable network communication
}

emelianov, 2019-11-25:
@Tomat7 you are right ```mb.Hreg(TEST_HREG, dist);```. Thanks.
I've omitted register address by accident.

methaqali, 2019-11-26:
Thank you very much @emelianov &@Tomat7 for your feed back. 
The code working well and it's better than old modbus code.
In fact, I'm facing new challenge which is the combine above code with following code, in order to make uploading for data to Ubidots in normal case and switching the data to local PLC in case interruption of internet. Any advise is appreciated.

/****************************************
 * Include Libraries
 ****************************************/
#include "UbidotsESPMQTT.h"
/****************************************
 * Define Constants
 ****************************************/
#define TOKEN "....." // Your Ubidots TOKEN
#define WIFINAME "....." //Your SSID
#define WIFIPASS "....." // Your Wifi Pass
Ubidots client(TOKEN);
/****************************************
 * Auxiliar Functions
 ****************************************/
void callback(char* topic, byte* payload, unsigned int length) {
  Serial.print("Message arrived [");
  Serial.print(topic);
  Serial.print("] ");
  for (int i=0;i<length;i++) {
    Serial.print((char)payload[i]);
  }
  Serial.println();
}
/****************************************
 * Main Functions
 ****************************************/
void setup() {
  // put your setup code here, to run once:
  Serial.begin(115200);
  client.setDebug(true); // Pass a true or false bool value to activate debug messages
  client.wifiConnection(WIFINAME, WIFIPASS);
  client.begin(callback);
  }
void loop() {
  // put your main code here, to run repeatedly:
  if(!client.connected()){
      client.reconnect();
      }
  
  // Publish values to 2 different data sources
  client.add("stuff", 10.2); //Insert your variable Labels and the value to be sent
  client.ubidotsPublish("source1");
  client.add("stuff", 10.2);
  client.add("more-stuff", 120.2);
  client.ubidotsPublish("source2");
  client.loop();
  }

* Are there a way to make this code connect to wifi by static IP?

Tomat7, 2019-11-26:
try...
1) put before setup()
```cpp
IPAddress staticIP224_60(192,168,1,70);
IPAddress gateway224_60(192,168,1,1);
IPAddress subnet224_60(255,255,255,0);
```
2) put before client.wifiConnection(WIFINAME, WIFIPASS);
```cpp
WiFi.config(staticIP224_60, gateway224_60, subnet224_60);
```

methaqali, 2019-11-26:
I tried it but caused prevent MQTT from connect to Ubidots, I thing it need some modify for "UbidotsESPMQTT.h" library.

emelianov, 2019-11-27:
Try
```c
IPAddress staticIP224_60(192,168,1,70);
IPAddress gateway224_60(192,168,1,1);
IPAddress subnet224_60(255,255,255,0);
IPAddress dns224_60(192,168,1,1);// DNS server IP
WiFi.config(staticIP224_60, gateway224_60, subnet224_60, dns224_60);
```

methaqali, 2019-11-27:
@emelianov as I said to Tomat7 I tried it but caused prevent MQTT from connect to Ubidots, I thing it need some modify for "UbidotsESPMQTT.h" library.

emelianov, 2019-11-27:
Note it's required to specify DNS server address.
```c
void setup() {
IPAddress staticIP224_60(192,168,1,70);
IPAddress gateway224_60(192,168,1,1);
IPAddress subnet224_60(255,255,255,0);
IPAddress dns224_60(192,168,1,1);// DNS server IP
Serial.begin(115200);
client.setDebug(true); // Pass a true or false bool value to activate debug messages
client.wifiConnection(WIFINAME, WIFIPASS);
WiFi.config(staticIP224_60, gateway224_60, subnet224_60, dns224_60);
client.begin(callback);
```

Tomat7, 2019-11-27:
> I tried it but caused prevent MQTT from connect to Ubidots

Is the Wi-Fi connection still OK? Is it possible to ping your device?
What is your network configuration? What is GW, MASK?
What static address leased for **your** device?
Are you sure that there is no other device with the same address on the network?

Tomat7, 2019-11-27:
> I tried it but caused prevent MQTT from connect to Ubidots

**methaqali,**
Is the Wi-Fi connection still OK with static address? Is it possible to ping **your** device?
What is your network configuration? What is GW, MASK?
What static address leased for **your** device?
Are you sure that there is no other device with the same address on the network?

By the way, you will get more qualified answer here https://github.com/ubidots/ubidots-mqtt-esp/issues

methaqali, 2019-11-27:
@emelianov when tried the first code (above) not effect on operation and use DHIP. But when tried second part its use static IP but MQTT refuse connections to Ubidots the following errors appeared:
[image]

methaqali, 2019-11-27:
@Tomat7  
1) ping is ok
2) static IP operating well with modbus code but when using ubidots code make MQTT error.
3) Network configuration the router is on range extender except both dynamic and static IP
 staticIP(192,168,1,70);
 gateway(192,168,1,1);
 subnet(255,255,255,0);
 dns(192,168,1,1);
4) In fact, I used separated router for my experiments I have two devices connected with it PLC and esp.

emelianov, 2019-11-28:
I've take a look on UbidotsESPMQTT source code.
They using some unsafe processing. To avoid unexpected behaviour try this variant:
```c
const char* token = "TOKEN";
char macAddr[18];

Ubidots* client;

void setup() {
byte mac[6];
WiFi.macAddress(mac);
sprintf(macAddr, "%2X%2X%2X%2X%2X%2X", mac[0], mac[1], mac[2], mac[3], mac[4], mac[5]);
client = new Ubidots(token, macAddr);

IPAddress staticIP224_60(192,168,1,70);
IPAddress gateway224_60(192,168,1,1);
IPAddress subnet224_60(255,255,255,0);
IPAddress dns224_60(192,168,1,1);// DNS server IP
Serial.begin(115200);
client->setDebug(true); // Pass a true or false bool value to activate debug messages
WiFi.begin(WIFINAME, WIFIPASS);
WiFi.config(staticIP224_60, gateway224_60, subnet224_60, dns224_60);
while (WiFi.status() != WL_CONNECTED) {
        delay(1000);
        Serial.print(".");
}
Serial.println(F("WiFi connected"));
Serial.println(F("IP address: "));
Serial.println(WiFi.localIP());
client->begin(callback);
```

methaqali, 2019-11-28:
Hi @emelianov  when compile it retune error message on this statment;
client = new Ubidots(token, macAddr);
error: invalid conversion from 'const char*' to 'char*' [-fpermissive]

emelianov, 2019-11-28:
```char* token = "TOKEN";```
or
```client = new Ubidots((char*)token, macAddr);```

methaqali, 2019-11-28:
Hi @emelianov when compile it retun error message on this statement;
client->begin(callback);
error: 'callback' was not declared in this scope

emelianov, 2019-11-28:
Look, it's just code fragment based on your code in https://github.com/emelianov/modbus-esp8266/issues/26#issuecomment-558517105. callback was declared there.

methaqali, 2019-11-28:
It's compiled with these errors in serial port:
[image]

methaqali, 2019-11-28:
Hi @emelianov  
Are you still thinking there is possible solution for current problem?

emelianov, 2019-11-29:
Sure. I'll try to find some spare time to play with Ubidots.

methaqali, 2019-11-29:
Great my friend, I'm sure you will find the solution. If you succeeded in this issue, will submit good contribution for IoT technology that is "increasing the reliability of IoT"

Best wishes

emelianov, 2019-12-02:
Unable to connect to things.ubidots.com but no crash.
```
ip:192.168.31.70,mask:255.255.255.0,gw:192.168.31.5
wifi evt: 0
wifi evt: 3
[hostByName] Host: things.ubidots.com IP: 50.23.124.68
failed, rc=-2 try again in 3 seconds
Attempting MQTT connection...[hostByName] request IP for: things.ubidots.com
[hostByName] Host: things.ubidots.com IP: 50.23.124.68
pm open,type:2 0
failed, rc=-2 try again in 3 seconds
```

```c
/****************************************
 * Include Libraries
 ****************************************/
#include "UbidotsESPMQTT.h"

/****************************************
 * Define Constants
 ****************************************/
#define TOKEN "BBFF-TbJcovrE6kyDs6CiVzT6XJRXWRPKdb" // Your Ubidots TOKEN
#define WIFINAME "E2" //Your SSID
#define WIFIPASS "fOlissio92" // Your Wifi Pass

//Ubidots client(TOKEN);

/****************************************
 * Auxiliar Functions
 ****************************************/

void callback(char* topic, byte* payload, unsigned int length) {
  Serial.print("Message arrived [");
  Serial.print(topic);
  Serial.print("] ");
  for (int i=0;i<length;i++) {
    Serial.print((char)payload[i]);
  }
  Serial.println();
}

/****************************************
 * Main Functions
 ****************************************/

char* token = TOKEN;
char macAddr[18];

Ubidots* client;

void setup() {
  byte mac[6];
  WiFi.macAddress(mac);
  sprintf(macAddr, "%2X%2X%2X%2X%2X%2X", mac[0], mac[1], mac[2], mac[3], mac[4], mac[5]);
  client = new Ubidots(token, macAddr);

  IPAddress staticIP224_60(192,168,31,70);
  IPAddress gateway224_60(192,168,31,5);
  IPAddress subnet224_60(255,255,255,0);
  IPAddress dns224_60(192,168,30,1);// DNS server IP
  Serial.begin(74880);
  client->setDebug(true); // Pass a true or false bool value to activate debug messages
  WiFi.begin(WIFINAME, WIFIPASS);
  WiFi.config(staticIP224_60, gateway224_60, subnet224_60, dns224_60);
  while (WiFi.status() != WL_CONNECTED) {
        delay(1000);
        Serial.print(".");
  }
  Serial.println(F("WiFi connected"));
  Serial.println(F("IP address: "));
  Serial.println(WiFi.localIP());
  client->begin(callback);
}

void loop() {
  // put your main code here, to run repeatedly:
  if(!client->connected()){
      client->reconnect();
      }
  
  // Publish values to 2 different data sources
  
  client->add("stuff", 10.2); //Insert your variable Labels and the value to be sent
  client->ubidotsPublish("source1");
  client->add("stuff", 10.2);
  client->add("more-stuff", 120.2);
  client->ubidotsPublish("source2");
  client->loop();
  }
```

methaqali, 2019-12-02:
Hi @emelianov  
Yes I think the crash because the sensor VL53L0X was attached to ESP.
Now I disconnect it and try exactly same code above I get this result;

ets Jan  8 2013,rst cause:2, boot mode:(3,7)
load 0x4010f000, len 1384, room 16 
tail 8
chksum 0x2d
csum 0x2d
v8b899c12
~ld
WiFi connected
IP address: 
192.168.1.70
Attempting MQTT connection...failed, rc=-2 try again in 3 seconds
Attempting MQTT connection...failed, rc=-2 try again in 3 seconds
Attempting MQTT connection...failed, rc=-2 try again in 3 seconds

methaqali, 2019-12-12:
Hi @emelianov, Any update for this issue?

emelianov, 2019-12-14:
Hello @methaqali 
Look, Problem is related to Ubidots. As far as I understand it's commercial product. I believe you'd better to address question to the Ubidots.

closed 2019-12-14 by emelianov

methaqali, 2019-12-14:
I used education part of Ubidots which is free. Any way, I sent inquiry for them and they answer me that their library don't support static IP. Can make some modification to the general MQTT library with  to connect with any cloud platform with supporting for static IP?

emelianov, 2019-12-16:
```Attempting MQTT connection...failed, rc=-2 try again in 3 seconds`` means that some problem with credentionals.
Do you able to connect using dynamic address? I beleave code i provided above should connect with rignt token.

methaqali, 2019-12-16:
No problem in connecting by dynamic IP. I think the problem not in the credentionals because when using dynamic IP with same credentionals it's work well.