emelianov/modbus-esp8266 #27: Write multiple holding register, func 16

petrb8, 2019-11-23:
Hi, I'm having difficulties with writing to multiple registers. I'm using mb.writeHreg function and testing the transfer with Modbus slave Windows utility. Function is transferring every even number correctly, but odd numbers are random and they're changing. Please find screenshot and code bellow. Thanks for advice

[image]

` #include <ESP8266WiFi.h>
#include <ModbusIP_ESP8266.h>

// Modbus Registers Offsets
const int REG = 256;                // Modbus register offset
const int COUNT = 19;                    // Count of Coils

IPAddress remote(192, 168, 1, 10);    // Address of Modbus Slave device

//ModbusIP object
ModbusIP mb;
  
void setup() {
  Serial.begin(74880);
 
  WiFi.begin("xxxx", "xxxx");
   
  mb.master();
  }

uint16_t res1,res2,res3,res4 = 0;
uint16_t zero = 0;
uint16_t res[COUNT] = {3232,3211,3039,3732,3030,4531,3643,3631,3639,4236,3944,4242,3535,3842,3436,3833,3631,3746,0};;

void loop() {
  if (mb.isConnected(remote)) {   // Check if connection to Modbus Slave is established
    mb.writeHreg(remote, REG, res, COUNT);
    
    mb.readHreg(remote, 16448, &res1);  
    mb.readHreg(remote, 16449, &res2);  
    mb.readHreg(remote, 16450, &res3);
    mb.readHreg(remote, 16451, &res4);   
  
  } else {
    mb.connect(remote);           // Try to connect if no connection
  }
   
   mb.task();
   delay(100);
     
    Serial.println(res1);
    Serial.println(res2);
    Serial.println(res3);
    Serial.println(res4);
    
}`

emelianov, 2019-11-25:
Hello,
It was the bug in writeHreg(). I've refactored related internals calls in 6daf1cc
So try with current master branch.

petrb8, 2019-11-29:
Works great now, thanks a lot, great job.

closed 2019-11-29 by petrb8