emelianov/modbus-esp8266 #28: How to receive response from Slave

petrb8, 2019-11-29:
Hi, 

when I sent writeHreg to TCP Slave, is there way to read response from Slave to this message?

When I'm reading multiple registers using readIreg, do I need to use addIreg function for results, or it it's enough to define uint16_t array?

Thanks

Petr

emelianov, 2019-11-29:
> when I sent writeHreg to TCP Slave, is there way to read response from Slave to this message?

Only using callback function. Something alike RTU-Master example (functionality is the same for TCP).

> When I'm reading multiple registers using readIreg, do I need to use addIreg function for results, or it it's enough to define uint16_t array?

Array is enough
```uint16_t readIreg(IPAddress ip, uint16_t offset, uint16_t* value, uint16_t numregs = 1, cbTransaction cb = nullptr, uint8_t unit = MODBUSIP_UNIT);```

petrb8, 2019-11-29:
Awesome thanks

closed 2019-11-29 by petrb8