emelianov/modbus-esp8266 #31: Why #define MB_MAX_REGS 32 ?

wladwnt, 2019-12-20:
Why the number of registers is hard limited to 32? Do you expect perfromance problems for more registers on ESP8266?

emelianov, 2019-12-20:
For historical reasons ;). 
You can just comment/remove #define MB_MAX_REGS line to escape the limit at all.

wladwnt, 2019-12-20:
Ok, thank you for clarification! Maybe you can mention this Limit in Readme, it took for me more than one hour to find out why my program crashes at setup. It seems that somethimes trying to add more than MB_MAX_REGS registers lead to an crash, somethimes not (I am using Modbus TCP Slave).

emelianov, 2019-12-26:
Just started implementing WiKi with more detailed API explanation.
https://github.com/emelianov/modbus-esp8266/wiki/Hardcoded-Constants

closed 2019-12-26 by emelianov