emelianov/modbus-esp8266 #32: Read Holding Registers from Modbus Master (Solaredge)

Chris685, 2020-01-22:
Good Morning, 

unluckily i need a little support to get my project working.  I am using CAS MODBUS SCANNER and get the Result of the inverter.

Attached two pictures

[image]
[image]

With this informations i created the code

```c++
#include <ESP8266WiFi.h>
#include <ModbusIP_ESP8266.h>

const int REG = 205;               // Modbus Hreg Offset
IPAddress remote(192, 168, 1, 26);  // Address of Modbus Slave device
const int LOOP_COUNT = 10;

ModbusIP mb;  //ModbusIP object

void setup() {
  Serial.begin(115200);
 
  WiFi.begin("SSID", "PASSWORD");
  
  while (WiFi.status() != WL_CONNECTED) {
    delay(500);
    Serial.print(".");
  }
 
  Serial.println("");
  Serial.println("WiFi connected");  
  Serial.println("IP address: ");
  Serial.println(WiFi.localIP());

  mb.master();
}

uint16_t res = 0;
uint8_t show = LOOP_COUNT;

void loop() {
  if (mb.isConnected(remote)) {   // Check if connection to Modbus Slave is established
    mb.readHreg(remote, REG, &res);  // Initiate Read Coil from Modbus Slave
  } else {
    mb.connect(remote);           // Try to connect if no connection
  }
  mb.task();                      // Common local Modbus task
  delay(100);                     // Pulling interval
  if (!show--) {                   // Display Slave register value one time per second (with default settings)
    Serial.println(res);
    show = LOOP_COUNT;
  }
}
```

But my result posted on the serial console is just 0 and not 4998 as supposed by the modbus scanner.

Do you have an hint for me?

Thanks and Regards,
Christian

emelianov, 2020-01-22:
Hello Christian,
Basing on screenshots you attached it's required to read register 40205.
```c
const int REG = 40205;
```

Chris685, 2020-01-22:
Thanks for reply. I wil try this in the evening.

Chris685, 2020-01-22:
Hi,

i have changed the code, but i only get the result = 0.

```c++
uint16_t res = 0;
uint8_t show = LOOP_COUNT;

void loop() {
  if (mb.isConnected(remote)) {   // Check if connection to Modbus Slave is established
    mb.readHreg(remote, REG, &res);  // Initiate Read Coil from Modbus Slave
    Serial.println("verbunden");
  } else {
    mb.connect(remote);           // Try to connect if no connection
    Serial.println("nicht verbunden");
  }
  mb.task();                      // Common local Modbus task
  delay(100);                     // Pulling interval
  if (!show--) {                   // Display Slave register value one time per second (with default settings)
    Serial.print("Ergebniss: ");
    Serial.println(res);
    show = LOOP_COUNT;
  }
}
```

```
Ergebniss: 0
verbunden
verbunden
verbunden
verbunden
verbunden
verbunden
verbunden
verbunden
verbunden
verbunden
verbunden
Ergebniss: 0
```

so i am connected and in the right loop, but result is not fitting to CAS MODBUS SCANNER.

Any suggestion?

emelianov, 2020-01-23:
Sorry i was wrong about address.
You should use
**204**
as register address.

```c
#include <ESP8266WiFi.h>
#include <ModbusIP_ESP8266.h>
#define SSID "xxx" //Your SSID
#define PASS "xxx" // Your Wifi Pass

const int REG = 204;               // Modbus Hreg Offset
IPAddress remote(192, 168, 1, 26);  // Address of Modbus Slave device
const int LOOP_COUNT = 10;

ModbusIP mb;  //ModbusIP object

void setup() {
  Serial.begin(115200);
 
  WiFi.begin(SSID, PASS);
  
  while (WiFi.status() != WL_CONNECTED) {
    delay(500);
    Serial.print(".");
  }
 
  Serial.println("");
  Serial.println("WiFi connected");  
  Serial.println("IP address: ");
  Serial.println(WiFi.localIP());

  mb.master();
}

uint16_t res = 0;
uint8_t show = LOOP_COUNT;

void loop() {
  if (mb.isConnected(remote)) {   // Check if connection to Modbus Slave is established
    mb.readHreg(remote, REG, &res);  // Initiate Read Coil from Modbus Slave
    Serial.println("verbunden");
  } else {
    mb.connect(remote);           // Try to connect if no connection
    Serial.println("nicht verbunden");
  }
  mb.task();                      // Common local Modbus task
  delay(100);                     // Pulling interval
  if (!show--) {                   // Display Slave register value one time per second (with default settings)
    Serial.print("Ergebniss: ");
    Serial.println(res);
    show = LOOP_COUNT;
  }
}
```

Chris685, 2020-01-23:
Sorry, the Result is still 0. I dont understand, the difference between the tool and the script.

One Register up or down wouldn't matter, this i can fix, but i guess i get no response, anything more to debug?

Chris685, 2020-01-24:
Now it is working. I had to add the slave id:

````c++
mb.readHreg(remote, 40204, &res3, 1, cb, 1);  //Frequenz Zähler

closed 2020-01-24 by Chris685