emelianov/modbus-esp8266 #33: Complete code - working example

TelleRen2k, 2020-02-26:
Hi,

I'm trying to connect an ESP32 (master) to a motor (slave) using MODBUS RTU and I found this promising project.

As I'm just starting to learn I hope for some help :-)

When connecting the ESP32 to the motor I use an RS485/max485 module like this one: http://www.bizkit.ru/en/2019/02/21/12563/

If I look at how this is to be connected to the ESP32 (also shown in the link above) it seems there is a connection from an ESP32 GPIO to DE and RE (which are shorted). And looking at example code (not using this project) it seems the GPIO has to be set accordingly depending on receiving or sending data to/from RS485.

I have looked through the examples but cannot see this implemented in any code. Therefore the question: How is that to be handled using this project? And is it possible to have a look at some "Arduino" working code (e.g. Master on MODBUS RTU reading from Input registers/writing holding registers)?

Thank you very much in advance.

emelianov, 2020-02-27:
Hello,
Data RX/TX control is supported. Just append corresponding control pin as second argument of .begin().

```c
#include <ModbusRTU.h>
#define RXTX_CONTROL_PIN 16

ModbusRTU mb;

bool cbWrite(Modbus::ResultCode event, uint16_t transactionId, void* data) {
  Serial.printf_P("Write result: 0x%02X\n", event);
  return true;
}

bool cbRead(Modbus::ResultCode event, uint16_t transactionId, void* data) {
  Serial.printf_P("Read result: 0x%02X\n", event);
  return true;
}

void setup() {
  Serial.begin(115200);
  Serial1.begin(9600, SERIAL_8N1, 17, 18);
  mb.begin(&Serial1, RXTX_CONTROL_PIN);
  mb.master();
}

uint16_t regs[20];
bool rd = true;

void loop() {
  if (!mb.slave()) { // Check if line is free
    if (rd) // Switch between read and write cycle
       mb.readIreg(1, 1, regs, 20, cbWrite);
    else
       mb.writeHreg(1, 21, regs, 20, cbWrite);
    rd = !rd;
  }
  mb.task();
  yield();
}
```

TelleRen2k, 2020-02-28:
Thank you. 

If what you have added is a working example - then I have to troubleshoot the RS485 interface as the call back function returns 0xE4 :-)

emelianov, 2020-02-28:
Code should work.

0xE4 - request timeout. (https://github.com/emelianov/modbus-esp8266/wiki/Callback:-Result-codes)
You need to check wiring and baudrate/stop bits matching target slave device settings.

TelleRen2k, 2020-03-04:
I finally got the request timeout to go away (had to replace the max485).

I'm using this code:

`#include <ModbusRTU.h>

#define RXD2 26
#define TXD2 27
#define TXPin 33
#define SLAVE_ID 1

ModbusRTU mb;

bool cbWrite(Modbus::ResultCode event, uint16_t transactionId, void* data) {
  Serial.printf_P("Write result: 0x%02X\n", event);
  return true;
}

bool cbRead(Modbus::ResultCode event, uint16_t transactionId, void* data) {
  Serial.printf_P("Read result: 0x%02X\n", event);
  return true;
}

void setup() {
  Serial.begin(115200);
  Serial2.begin(19200, SERIAL_8E1, RXD2, TXD2);
  mb.begin(&Serial2, TXPin);
  mb.master();
}

uint16_t result = 123;

void loop() {
  
  if (!mb.slave()) { 
    mb.readHreg(SLAVE_ID, 53666, &result, 1, cbRead);
    Serial.println("reading done...");
    Serial.println(result);
    Serial.println("-----");
  }
  mb.task();

  yield();
  delay(1000);
}

`

What happens is that I get 1 reading (one entry into the if(!mb.slave) statement), but the 'result' variable is not updated and the callback function is also not triggered.

Any ideas?

Thank you in advance.

emelianov, 2020-03-04:
It's not clear why callback is not fired.
Anyway you should decrease delay(1000) to, say, delay(100) as operation timeout is hardcoded to 1000ms.

TelleRen2k, 2020-03-04:
Changed delay to 100ms. Result is still the same. 

Any suggestions on how to find the problem? Thanks

emelianov, 2020-03-04:
What is the slave device? Is any documentation on Modbus configuration of them?

TelleRen2k, 2020-03-04:
Slave is a motor. I can access the motor using MODBUS from my laptop. Connection parameters are correct.

emelianov, 2020-03-04:
Try to play with register number: 53665, 53667.

If no success add incoming packet debug output to ModbusRTU.cpp (after line 154).
In case of no incoming data it's required to analize RS-485 dataflow.
```c
// Existing lines
for (uint8_t i=0 ; i < _len ; i++)
		_frame[i] = _port->read();   // read data + crc
	//_port->readBytes(_frame, _len);
// Line to add
for (uint8_t i = 0; i < _len; i++) Serial.printf("%02X ", _frame[i]);

// Rest of code
    u_int frameCrc = ((_frame[_len - 2] << 8) | _frame[_len - 1]); // Last two byts = crc
    _len = _len - 2;    // Decrease by CRC 2 bytes
```

TelleRen2k, 2020-03-04:
I have tried to read from several holding registers. Same issue.

Added the following code:

for (uint8_t i = 0; i < _len; i++) Serial.printf("%02X ", _frame[i]);
Serial.println("---");

And get this (notice the '---'):
CE 00 23 00 06 00 26 00 6E 00 03 00 00 00 27 00 4E 00 02 0F 00 00 4C 00 CE 00 03 00 44 07 26 00 6E 00 23 00 2C 00 6C 00 02 00 02 03 23 00 CE 00 C6 07 22 00 1C 07 04 00 6C 00 03 00 44 00 CE 00 6C 00 4E 00 07 3F 4E 00 EC 00 88 00 44 00 06 00 37 00 00 00 0C 00 26 00 CC 07 02 00 22 00 4E 00 06 00 04 00 03 00 4E 00 4E 00 07 00 46 00 23 00 00 00 23 00 07 00 27 00 6C 00 37 00 23 00 00 00 27 00 46 00 07 00 22 00 27 00 DC 00 23 00 27 03 CE 00 C4 00 03 00 22 00 22 00 EE 00 03 00 06 00 27 07 27 00 03 00 00 7B 4E 00 44 00 4C 00 00 00 00 2F 00 CE 00 06 00 22 00 00 00 EC 00 03 00 26 00 26 00 EC 00 03 00 02 00 6E 00 47 00 06 00 66 00 26 00 6E 00 07 00 4E 00 4E 00 06 00 03 00 01 00 DC 0F CE 00 33 00 46 00 06 03 CE 00 06 00 2E 00 46 00 4C 00 67 00 23 00 6E 00 EC 00 CC 00 ---
47 00 4E 00 27 00 46 00 00 24 00 26 00 EE 00 07 0F 23 00 4C 00 4E 00 DC 00 27 00 C4 00 67 03 6F 00 44 00 27 00 67 07 D8 00 03 00 02 00 27 00 6F 00 22 00 4C 00 CE 00 4E 00 00 00 00 03 4E 00 DC 00 22 00 07 00 27 00 EE 00 07 00 06 00 0C 00 6E 00 9C 00 0F 00 27 00 4C 03 27 00 23 00 0E 00 6C 00 E4 00 0C 03 24 00 27 03 DC 00 23 00 26 00 CC 00 E4 00 03 00 00 00 2E 00 6E 00 46 00 06 00 27 00 64 00 DC 00 06 00 0E 00 6F 00 4E 00 00 00 27 00 44 00 6E 00 CE 00 27 00 6E 07 8C 00 CC 00 23 00 23 00 26 00 EE 00 0C 00 03 00 23 00 66 00 07 00 02 00 6E 00 66 00 6F 00 03 00 23 00 4E 00 CE 00 00 04 00 01 00 6E 00 CE 00 07 07 4E 00 46 00 37 00 E4 00 07 00 02 00 EC 00 9C 00 00 00 02 00 37 00 03 00 17 03 06 00 27 00 EE 00 07 03 07 00 6E 00 66 00 EC 00 06 00 66 00 6E 00 44 00 00 ---
4E 00 EC 00 26 00 04 00 24 00 64 00 00 06 00 27 00 44 00 04 03 CE 00 00 03 00 02 00 6E 00 CC 00 02 07 23 00 06 00 76 00 00 07 17 07 46 00 4E 00 03 00 00 03 07 00 E4 00 44 00 17 00 00 00 4F 03 03 00 88 00 00 00 00 00 00 27 00 6E 00 00 00 26 00 02 00 6C 00 0F 00 02 00 6F 00 D8 00 CE 00 00 27 03 4C 00 DC 00 07 00 26 00 22 00 37 00 EC 07 0E 00 0E 00 44 00 00 00 07 00 0F 00 66 00 46 00 4C 00 02 00 00 00 6E 00 27 00 47 00 12 00 02 00 4E 00 03 00 07 00 07 00 27 00 4E 00 07 00 23 00 44 00 27 00 9C 00 00 02 00 27 00 4C 00 6C 00 00 00 00 00 37 00 37 00 7F 00 04 00 02 00 37 00 EC 00 00 00 00 27 00 4E 00 66 00 07 00 00 00 27 00 DC 00 4C 00 03 00 23 00 4C 03 CE 00 98 00 03 00 07 00 4E 00 07 00 23 00 06 00 22 00 23 00 D8 00 06 00 CC 00 67 00 8C 00 07 00 27 00 4C 00 66 ---
6F 00 DC 00 03 00 04 00 00 00 27 00 07 00 37 00 44 00 27 00 00 00 23 03 0F 00 27 00 44 00 03 00 00 00 6E 00 6E 00 DC 00 00 00 00 27 00 64 00 6E 00 06 00 27 00 47 00 22 00 66 00 27 00 37 07 6E 00 CC 00 EC 00 07 00 4E 6F 66 00 67 00 02 00 03 00 6E 2F EC 0F 02 07 23 00 4C 00 67 00 CE 00 07 03 04 00 22 00 36 00 9C 00 00 00 27 00 23 00 03 00 CE 00 27 00 37 00 44 00 44 00 03 00 03 00 27 00 DC 00 E4 00 00 03 00 00 00 06 00 CE 00 02 00 04 00 2E 00 6F 00 DC 00 27 00 00 00 4E 00 46 00 03 00 00 00 27 00 DC 00 44 00 06 00 27 00 4E 03 EE 00 E4 00 26 00 23 00 33 00 CE 00 CC 00 00 0E 00 2E 00 02 00 CE 00 3B 00 22 00 22 00 27 00 23 00 DC 00 07 00 22 00 47 00 44 00 00 06 00 00 00 44 00 64 00 CE 00 03 00 07 00 23 00 CC 00 4E 00 03 00 27 00 9C 00 CC 00 07 00 23 00 26 00 6E ---
00 00 27 00 6C 00 37 00 EE 00 0E 00 6C 00 64 00 CC 00 00 00 07 00 64 07 26 00 24 00 02 00 22 00 03 00 8C 00 44 00 03 00 06 00 27 00 6F 00 44 00 00 00 27 00 23 00 03 00 03 00 07 00 07 00 67 07 76 00 07 00 27 00 26 00 0E 00 EE 00 D8 00 00 00 06 00 11 00 42 00 DC 00 00 00 27 00 6C 00 46 00 CE 00 27 00 44 00 6E 00 64 00 44 00 17 00 46 00 CC 00 4E 00 64 00 00 04 00 27 00 4E 00 6E 00 2E 00 CC 00 00 07 00 03 00 CC 00 CE 00 03 02 03 00 00 00 4E 00 03 00 02 00 07 00 27 03 36 00 EE 00 02 00 27 00 37 00 6E 00 07 07 27 00 44 00 2E 00 CE 00 D8 00 0F 00 46 00 4E 00 27 00 6F 00 CC 00 03 4C 00 26 00 6F 00 47 00 EE 00 07 00 43 00 4C 00 24 00 4E 00 17 00 46 00 27 00 27 00 46 00 00 00 3D 00 6C 00 37 00 9C 00 00 00 00 27 00 6E 00 46 00 CE 00 02 00 37 00 27 03 EE 00 6F 00 03 ---
06 00 26 00 04 00 27 00 CE 07 CC 00 04 00 44 00 06 00 27 00 E4 00 84 00 00 04 00 23 00 6F 00 27 00 EC 00 04 00 07 00 6E 00 CE 00 07 03 17 00 23 00 26 00 27 00 03 00 00 00 26 00 67 00 37 00 23 00 07 00 00 00 47 00 46 00 00 00 0E 00 37 00 46 00 EC 00 07 00 00 00 6E 00 CC 00 46 00 D8 00 02 00 02 00 6E 00 44 00 4C 00 11 03 07 00 37 00 CC 00 6F 00 02 00 23 00 23 00 27 00 98 00 CE 00 07 00 26 00 37 00 FC 00 00 00 23 00 47 03 6F 00 2F 00 23 00 6E 00 23 00 CC 00 00 00 46 00 0F 00 24 00 4F 00 EE 00 23 00 4E 00 27 00 44 00 4C 00 27 00 27 00 6E 03 EC 00 EC 00 03 00 02 00 46 00 4C 00 46 03 02 00 27 00 EE 00 6E 00 02 00 8E 00 07 00 23 00 6E 7F CE 00 6C 00 00 17 00 02 00 37 00 6F 00 C6 00 00 00 00 27 0F 66 00 EC 00 00 00 27 00 37 00 DC 00 DC 00 3B 00 27 00 06 00 27 00 ---
19 00 6C 00 67 00 22 00 00 00 31 00 44 00 4E 00 9C 00 07 07 07 00 2E 00 CC 00 6E 00 02 00 00 00 00 00 66 00 4E 00 02 00 00 00 17 00 DC 00 EC 00 23 00 23 00 02 00 EE 00 EE 00 17 00 06 00 6E 00 DC 00 26 00 00 00 4E 00 23 00 6E 07 00 00 4E 00 2E 00 44 00 4E 00 27 00 46 00 6C 00 26 00 CC 00 6E 00 00 00 27 00 23 00 CE 00 CC 00 04 00 27 00 6C 00 4E 00 EE 00 00 00 00 23 00 4E 00 CC 00 03 00 00 00 02 00 EC 07 03 00 44 07 00 00 02 00 6E 00 DC 00 07 00 4E 00 6E 00 4E 00 CE 00 27 00 27 00 66 00 37 00 00 00 37 00 4E 00 EC 00 DC 00 23 00 2F 00 6C 00 C4 00 07 00 23 00 00 00 37 00 C4 00 4E 00 06 00 44 00 67 00 EE 00 00 EC 00 00 22 00 23 00 06 00 67 00 00 00 04 00 00 00 27  [... cut ...] 00 03 00 8E 00 C4 00 03 03 00 02 00 27 3B 6F 00 00 07 00 23 00 27 00 6E 00 00 00 27 00 4E 00 06 00 07 00 37 00 4E 00 00 00 07 00 44 00 4E 00 06 00 0C 00 33 00 4C 00 EC 03 23 00 27 00 03 00 03 00 27 00 2E 00 46 00 CE 00 00 00 4E 00 EE 00 88 00 22 00 37 00 6E 00 EE 00 27 00 46 00 6C 00 00 00 03 00 37 00 4E 00 07 00 06 00 27 00 6F 00 00 00 00 00 4E 00 4C 00 EE 00 02 00 6E 03 DC 00 23 07 07 01 27 00 EC 00 06 00 02 00 2E 00 6E 00 46 00 00 00 26 00 46 00 D8 00 06 00 27 00 6F 00 22 01 0C 00 ---
+++
+++

emelianov, 2020-03-04:
1. If ground wire is connected between master and slave devices? (I mean that should be 3 wire connection A, B and GND)
2. Basing on what you wrote before and debug output it's not looks like a reason but try 2400 and 9600 baudrates.
3. Examine sketch with slave device disconnected (no wiring at all connected to A, B and GND)

TelleRen2k, 2020-03-15:
I have tried to use a common ground but then I cannot transmit any data at all.

Then I have tried to switch the ESP32 with a PC with Simply Modbus Master installed. I have used the exact same cabling to the motor (no GND wire). And from the PC I can send and receive commands without any problems.

If I disconnect the wiring to the motor and run the sketch I get: "Read result: 0xE4 "

emelianov, 2020-03-16:
The questuion is: if motor is diconnected do you still getting incoming data (printed to Serial by debug code above) ?

TelleRen2k, 2020-03-18:
Short status update :-)

I got a cheap logic analyzer and I have started to look into what really happens. Update will follow.

TelleRen2k, 2020-03-19:
Thank you so much for your help. I finally got it working.

RS485 chip could only run on 5V, a GPIO on the ESP32 was "weird", had to throw out 2 RS485 drivers before I got one working. Logic level shifter was not working (all the time) - changed to simple voltage divider.

The logic analyzer saved the day (week) :-)

closed 2020-04-02 by emelianov