emelianov/modbus-esp8266 #34: modbus TCP/IP network

pptsk, 2020-03-01:
Hi,
I want try to build my own modbus TCP/IP network over wifi using ESP8266(as master and slaves).
I want to use 8 slaves with connected temperature sensors.
Master should read data from each slave every 300ms and print them on serial.
Can I do it with your library?
I cant find any example ...
thank you
pptsk

emelianov, 2020-03-02:
You can use bundelled examples as start point. Of'cause you need to rework them for your specific task. 
Dependin on your needs you can use multiple registers to store one floating point temperature value or store scaled integer value (for example 1854 for 18.54 deg.)

The problem is that ESP8266 is limited to 4 simultatious TCP connections. So to process 8 slaves you have to perform connect/disconnect for each query. Also if some of slaves become unavailable -- processing will stoped during TCP connection timeout time (which is 5 seconds).
Possible solution is to replace master device with ESP32.

Tomat7, 2020-03-02:
> I want to use 8 slaves with connected temperature sensors.
> Master should read data from each slave every 300ms and print them on serial.
So, 37.5 ms per full cycle of data transfer and processing. By Wi-Fi. By ESP8622. Hmmm...
And ~24 data set per second on serial.
What do you plan to do with this set of data? 
What temperature sensors you are using?

pptsk, 2020-03-02:
Hi all,
I messed two project here so sorry...
I made  "node" in the past,  where is used ESP8266 + MCP23008 + WizNet5500 ethernet module which was used for simple home automatization and now I would like use them for some home(farm) "industrial"  network.
-Each node(slave) will return only one byte ( MCP23008 has only 8 inputs )
-Node(slave) will be connected to external binnary counter ( 0-255 ).
-I would like read data every 300-500ms;
-I want to build simple web page on master and show received data there.

I am asking if its possible to do some like that with modbus TCP/IP protocol only before buying all needed parts or should I use another  method ?
thank you for any advice

emelianov, 2020-03-02:
The library can work with Ethernet using https://github.com/d-a-v/W5500lwIP only (Ethernet.h is not supported). Also take notes on slave count i wrote above.

pptsk, 2020-03-02:
Hi,
Thank you for fast reply!
Yes, I used the same ethernet library in my previous project and works fine.
I plan to use your master examle and put there IP address of all available nodes, and then request data in loop in 300ms period. No-answering(not-connected)  node will be skipped after cca 200ms, but they will be called in next loop again
Is this right way? Group communication is new for me...
Why is ESP32 better choice as master?
thank you!

emelianov, 2020-03-02:
If wireless network is stable 200ms timeout should be enough. But it seems to be shortest value possible.
ESP32 will allow to have 8 persistent connections.

natolaba, 2020-03-15:
Hello Guys !

I already used "ModbusIP_ESP8266.h" library from emilianov and was Sucessfully !
Thanks emilianov for this wonderfull job.
Now I want to use Arduino MEGA and a W5100 Ethernet Shield and uses the examples from ESP8266 to the new platform (Mega and W5100),  On Mega I have more I/O points and Analog Inputs.
I tried to change the code from ESP8266 based with the information on this post, but not sucessfully:

When I Compile using this library "W5100lwIP.h" I have some Error Messages as below:

Arduino: 1.8.12 (Windows 10), Board: "Arduino Mega or Mega 2560, ATmega2560 (Mega 2560)"
W5100_Reading_HoldingReg:38:10: fatal error: W5100lwIP.h: No such file or directory
 #include <W5100lwIP.h>
          ^~~~~~~~~~~~~
compilation terminated.
exit status 1
W5100lwIP.h: No such file or directory

Invalid library found in C:\Users\ ... \Arduino\libraries\W5100MacRaw-master: no headers files (.h) found in C:\Users\ ... \Documents\Arduino\libraries\W5100MacRaw-master

I already made the basic, that is copy the folder of "W5500lwIP-master" and "W5100MacRaw-master" to the Arduino IDE library folder ...

Could anyone help me ?
Thanks for the support.

emelianov, 2020-03-16:
I'm afraid it's not possible to use the library with Mega. W5500 and W5100 libraries you mentioned requires LwIP TCP stack implementation and WiFi class that wraps LwIP on ESP. Using of generic W5100 library for Arduino is also impossible without changing of ModbusIP library code.
ModbusRTU prabably will work.

natolaba, 2020-03-16:
Thanks emelianov for your feedback.
With you answer, now I forgot the MEGA Board and now I am thinking to use a ESP32 (ESP-WROOM-32) Board, where I have 18 Analog Inputs, with 12 Bits of resolution on this board. What your opinion about ? Your library "ModbusIP_ESP8266.cpp" and "ModbusIP_ESP8266.h" wil work with ESP32 ? Sorry my question, because I am new bee with ESP32 ... Thanks for support.

emelianov, 2020-03-16:
Yes, the library is working with ESP32 and ESP8266 both.
ESP32 is good enough for various applications.

natolaba, 2020-03-21:
Hello emelianov !
Now I am testing with ESP32 Board and is working ... but on Serial Monitor I see this messages:

1692: 0Modbus result: E4
Timeout
Modbus result: E4
Timeout
Modbus result: E4
Timeout
Modbus result: E4
Timeout
Modbus result: E4
Timeout

What´s meaning E4 ?

Thanks for help.

emelianov, 2020-03-22:
Hi,
E4 means transaction timeout. Refer [Callback: Result codes](https://github.com/emelianov/modbus-esp8266/wiki/Callback:-Result-codes) for error codes list.

closed 2020-04-02 by emelianov