emelianov/modbus-esp8266 #35: Feature Request: support non standard data types (int32, float, ...)

everslick, 2020-03-04:
Thanks for this great implementation of the MODBUS protocol. I found it to be as the most complete and clean library out there.

Would it be possible and in the spirit of the lib to support *non standard* data types like 32bit integers and float? there are quite a few devices and masters out there that make use of those, even if they are not strictly MODBUS.

Any opinions?

emelianov, 2020-03-04:
You're welcome Clemens.

It works out-of-box: https://github.com/emelianov/modbus-esp8266/issues/17.
 See no reason hiding this magic inside the library code.

```c
float value32;
//int32 value32;

mb.readHreg(ip, REG_NR, (*uint16_t)&value32, 2); // read two sequental registers to 32-bit variable
//value32 = (value32>>16) | (value32<<16); // Uncomment for words swap
```

everslick, 2020-03-04:
ah, ic! I assumed different data types would be transfered differntly on the wire, but they just consume multiple register indices. Thanks for making that clear for me!

closed 2020-03-04 by everslick