emelianov/modbus-esp8266 #36: Will miss message from master, if also receiving serial data on other Uart.

mstegen, 2020-03-06:
Hi, 
Thanks for your library.

I'm using the library set up as a slave on a ESP32.
For the modbus uart i'm using Uart1 at 9600bps.
In general everything seems to work, except when also at the same time Uart2 (115200 bps) receives a long message ~100ms. Then the modbus message from the master is missed. 
I verified this with a scope.

I 'fixed' it by changing the _t inter-frame delay to 10ms.
But of course this is not a very good solution.
It would be better if the receive function had better control over the uart, maybe with an interrupt?

Thanks!
-Michael

emelianov, 2020-03-06:
Hello Michael,
You are welcome.

Is your code is Arduino/ESP32 or it's RTOS with Uart1 (Modbus) and Uart2 processed within separate tasks?
I suspect you are using RTOS tasks. It looks like Modbus processing task interrupted by task switching for more than _t but less than Modbus frame transmititng time milliseconds. This causing the library stripping rest of frame.
Try to pin Modbus and Uart2 tasks to different cores.
In fact it looks like for receive at ESP32 should be `_t = portTICK_PERIOD_MS;`. Going to test and fix code.
Also there is a plan to implement native FreeRTOS support in next major release.

|If this project is helpful for your projects you can support it by a glass of beer|[image]|
|---|---|

mstegen, 2020-03-06:
Hi Alexander, 

I'm using Arduino/ESP32. I have tried to pin the Uart2 read task with xTaskCreatePinnedToCore to core 0, as the main loop uses core 1, but unfortunately this didn't fix the issue.
Also tried the same with the modbus task, same result.

Thanks, Michael

emelianov, 2020-03-16:
Michael, 
Can't test on real hardware but try 3.0.1 branch. I've added blocking of thread switching for both data send and receive. Hope this help to esacpe timeuots.
https://github.com/emelianov/modbus-esp8266/tree/3.0.1

mstegen, 2020-03-16:
Hello Alexander, 
Thanks for the update, i tested on the ESP32, but unfortunately, the result is the same.
Here's a video so you can see what's going on:
https://photos.app.goo.gl/H3LFRHygeTqaHDBw8

Top (yellow) trace is modbus data, short burst is received data from master, longer one the slave response.
The blue trace is burst of serial data received on uart 2.

emelianov, 2020-03-17:
Hello Michael,
Thanks for donate.
As for issue it looks like ESP-IDF uart buffering bug. I'll continue investigations.
At this point as workaround try to use uart**0** for burst and uart**2** for ModbusRTU.

emelianov, 2020-03-22:
Another solution that seems to be woking is to reinstall part driver for burst port. Something alike
```c
void setup() {
  uart_driver_delete(UART_NUM_0);
  uart_driver_install(UART_NUM_0, 256, 0, 0, NULL, 0);
  Serial.begin(115000);
  Serial1.begin(9600, SERIAL_8N1, 19, 18);
```

mstegen, 2020-03-24:
Thanks, I tried both methods. 
Swapping the uarts, so that uart0 is used for burst, and uart2 for Modbus RTU did not fix the issue for me.

When using the ESP-IDF uart_driver_install code, together with Serial.begin(115200) the receive function doesn't seem to work anymore.

I therefore tried changing all uart2 code to ESP-IDF functions, but as my receive buffer is 2Kb, the uart_read_bytes function blocks until it has received the complete burst. So now the modbus response is delayed by 100ms ;-)

I think i'll for now stick to the small _t timeout change i made to your code..

emelianov, 2020-06-18:
Hello Michael,
Just tested latest library code against arduino-esp32 source from master branch.
It seems to me that data lost has gone.

closed 2020-07-13 by emelianov