emelianov/modbus-esp8266 #38: Newbie help request.

OfferLa, 2020-03-18:
I'm looking for a good TCP Modbus controller for a long time, want now to use this library on my NodeMCU board.
Regularly, I use the Arduino IDE to program it, and the Library manager to add needed libraries to my environment.

I've tried to find this library in my manager without success.

2 things:
1) May you please add your great program to the public Arduino library repository so it will be available there?
2) Meanwhile, when it's not there, how can I install and use it in my environment?

Many thanks!

emelianov, 2020-03-22:
Hello,
Thanks for donation ;).
1. I'm afraid the library will not be placed to Arduino repo as it's designed for very little set of Arduino-compatible devices.
2. [As far as i understand you've installed the library but just in case] Just extract source code of latest release to <home_directory>\Arduino\libraries and restart IDE.

OfferLa, 2020-04-11:
Dear Alexander.

 

I hope you get this email. Anyway, I'll appreciate also a direct email address of you.

 

If you have some minutes for me – I'm trying to achieve a simple read from Modbus, and have some problems.

I'll really appreciate if you'll take some minutes to walk me through this little process.

 

Many thanks!

Be Well!

Offer.

emelianov, 2020-04-12:
Hello,
Please share the code and discription of the problem.

OfferLa, 2020-04-12:
This is the simple function I have to write one value to one holding register, and it works for me:

 

void ModbusWriteOneReg(int Where, int What) {

  Serial.println("Entered Write Function.");

 

   while (!(mb.isConnected(remote))) {

    delay(250);

    Serial.print(".");

    mb.connect(remote);           // Try to connect if no connection

  }

 

  mb.writeHreg(remote, Where, What);

  mb.task();

  delay(100);

}

 

For some reason, I wasn't able the make an equivalent read function to read one integer value from one holding register.

 

Would you please show me how?

 

Many thanks!

OfferLa, 2020-04-13:
Still waiting..

emelianov, 2020-04-14:
Hello,
The library is designed to run assync. That is no calls returns data or results immediate (except immediate errors in case of unexpected connection state or out-of-resources). Take a look at [Master example](https://github.com/emelianov/modbus-esp8266/tree/master/examples/MasterSimpleRead).
Sure you can agregate loop and block execution till query ended. Somethig alike (not tested code):
```c
ModbusIP mb;
uint16_t readHregBlocking(IPAddress remote, uint8_t hreg) {
  uint16_t res = 0;
  uint16_t id = mb.readHreg(remote, hreg, &res);
  while(mb.isTransaction(id)) // For ModbusIP
//while(mb.slave()) // For ModbusRTU
    mb.task();
  return res;
}
```
But i personally suggest to design your sketch in non-blocking way.

OfferLa, 2020-04-16:
In non-blocking mode - how can I control when to query the device and when I get the needed data?

emelianov, 2020-04-17:
You need to [use callbacks](https://github.com/emelianov/modbus-esp8266/blob/master/examples/MultipleHRegDebug/MultipleHRegDebug.ino).

closed 2020-05-06 by emelianov

emelianov, 2020-05-21:
@OfferLa, 
> May you please add your great program to the public Arduino library repository so it will be available there?

Done.