emelianov/modbus-esp8266 #39: [documentation improvement] Supplying multiple registers from callback when running as slave

marcovannoord, 2020-03-31:
I couldn't really find in the documentation on how to use the callbacks in combination to using a multiple of registers, and it took me some time before i got this working. Maybe it is an idea to add some example to the documentation?

For reference, this is how i did it:
```   
// To define something that has multiple registers, like an ip address:
mbIPSlave.addIreg(1000, 0, 4);
// Adding the callback with 4 registers, will mean that the same callback will be called once for every register.
mbIPSlave.onGetIreg(1000, cbReadWiFiIP, 4); 
```
And the callback itself: 
```
uint16_t cbReadWiFiIP(TRegister *reg, uint16_t val)
{

    switch (reg->address.address)
    {
    case MB_ADDR_IP:
        // Get the first octet of the local ip address (uint8_t), and cast it to uint16_t
        val = static_cast<uint16_t>(WiFi.localIP()[0]);
        break;
    case MB_ADDR_IP + 1:
        val = static_cast<uint16_t>(WiFi.localIP()[1]);
        break;
    case MB_ADDR_IP + 2:
        val = static_cast<uint16_t>(WiFi.localIP()[2]);
        break;
    case MB_ADDR_IP + 3:
        val = static_cast<uint16_t>(WiFi.localIP()[3]);
        break;
    default:
        break;
    }
    return val;
}
```

emelianov, 2020-04-02:
Hello Marco,
That's make sence.

emelianov, 2020-06-17:
https://github.com/emelianov/modbus-esp8266/blob/master/examples/IP-server-MassOperations/IP-server-MassOperations.ino

closed 2020-07-13 by emelianov

marcovannoord, 2020-07-13:
Looks great, thanks for adding!