emelianov/modbus-esp8266 #41: Using TCP and RTU with common Registers

andy-freund, 2020-05-11:
First of all I find this tool extremely helpful and thank you very much for providing it.

I am currently working on a project where I would like to enable access to Modbus registers via Modbus TCP as well as via RTU simultaneously (parallel operation). This allows a flexible and comfortable use.

My question:
To save resources I would like to create the Modbus registers only once (addHreg, addCoil, addIsts, addIreg). Is that possible? If so, what do I have to consider?

emelianov, 2020-05-11:
Hello Andreas,
Thanks.

That's default behaviour. I mean registers and callbacks are shared for all ModbusIP and ModbusRTU instances.

closed 2020-05-21 by emelianov

percramer, 2020-05-21:
Hi,

so I am testing this now. But should I just create an RTU and IP modbus object and do all init code twice? So Addreg etc once for the ip object and once for the RTU?

emelianov, 2020-05-21:
All instances of ModbusIP and ModbusRTU are sharing common registers and callbacks set. So you need to add register only once using any instance of ModbusXXX and this register become available via any other instances of ModbusXXX.
In other words registers are global.