emelianov/modbus-esp8266 #43: Sensor float value

percramer, 2020-05-20:
Hi,

I wonder if you could give me some advice/help. I am a C# programmer by day (and work with modbus quite often). I can't seem to get float values to work. And this shouldn't be to hard probably... The sensor I am reading gives me a float value as sensor reading and I want to present that as 2 registers (running my code as modbus slave). But I can't seem to figure it out somehow. Do I first need to declare 2 registers separately for example?

emelianov, 2020-05-21:
Hello,
That's right. Float values typically stored within two sequental Modbus registers.
Depending on endinnes of devices byte and word orded should be changed on receiver side. Somthing like:
```c
float value32;
mb.readHreg(ip, REG_NR, (*uint16_t)&value32, 2); // read two sequental registers to 32-bit variable
value32 = (value32>>16) | (value32<<16); // swap words order
```

percramer, 2020-05-21:
Hi, 

thank you for the quick response. Sorry I wasn’t clear I think. My device is the slave so it will be read from me using Modbus. Should I manually create 2 internal registers and split the float? Or is there a function in the library that takes care of that? Great library by the way!

emelianov, 2020-05-21:
No special function there. Just create two registers and fill them by high and low words of the float.
Thanks.

percramer, 2020-05-21:
Ok thanks!

closed 2020-05-21 by percramer