emelianov/modbus-esp8266 #44: ESP32 as a slave But it had no response

CHENCY1982, 2020-05-29:
hi,
i  studied some days ,ESP32 as a master ,the sysem was good,use modbus slave can see the communication result.
but now i use the ESP32 as a slave ,use modbus poll as a master ,it had faults,timeout error.

`#include <ModbusRTU.h>
#include <HardwareSerial.h>
#define REGN 10
#define SLAVE_ID 1
#define BAUD 9600

ModbusRTU mb;
HardwareSerial mySerial1(1);
void setup() {
  //Serial.begin(9600, SERIAL_8N1);
  mySerial1.begin(BAUD,SERIAL_8N1,5,18);
#if defined(ESP32) || defined(ESP8266)
  mb.begin(&mySerial1,16);//16 is the control pin
#else
  mb.begin(&mySerial1,16);
  mb.setBaudrate(BAUD);
#endif
  mb.slave(SLAVE_ID);
  mb.addHreg(REGN);
  mb.Hreg(REGN, 100);

}

void loop() {
  mb.task();
  yield();
}`
thanks!

emelianov, 2020-05-29:
Hello.
Try to drop `#include <HardwareSerial.h>` and `HardwareSerial mySerial1(1);` use predefined `Serial1`:

```c
Serial1.begin(BAUD,SERIAL_8N1,5,18);
mb.begin(&Serial1,16);
```

CHENCY1982, 2020-05-29:
> Hello.
> Try to drop `#include <HardwareSerial.h>` and `HardwareSerial mySerial1(1);` use predefined `Serial1`:
> 
> ```c
> Serial1.begin(BAUD,SERIAL_8N1,5,18);
> mb.begin(&Serial1,16);
> ```

now I have change the code.it also did not have  response .now I Connected to pin 16 a led , when ESP32 as a master ,the led would light on and light off ,but ESP32 as a slave ,the led always light off,so I think "ModbusRTU::rawSend" may be not work.
The following code is what I am using  

#include <ModbusRTU.h>
//#define ESP32  is it must?
#define REGN 10
#define SLAVE_ID 1

ModbusRTU mb;

void setup() {
 
  Serial1.begin(9600,SERIAL_8N1,5,18);
  mb.begin(&Serial1,16);
  mb.slave(SLAVE_ID);
  mb.addHreg(REGN);
  mb.Hreg(REGN, 100);
}

void loop() {
  mb.task();
  yield();
 
}
thanks!

emelianov, 2020-05-30:
LED you mentioned is not flash just because no send action on slave side. That's may be because of broken incoming packet. Probably some problem with wiring. Take a look sample wiring below: 
[image]

CHENCY1982, 2020-06-08:
> LED you mentioned is not flash just because no send action on slave side. That's may be because of broken incoming packet. Probably some problem with wiring. Take a look sample wiring below:
> [image]

Now I soldered the  circuit boards (TTL TO RS485),now it's good now .thank you very much!

emelianov, 2020-06-08:
Glad to hear the problem has resolved.

closed 2020-06-08 by emelianov