emelianov/modbus-esp8266 #45: MODBUS RTU Master receive \0

TelleRen2k, 2020-06-05:
Hi,

I try to read some info (fan name) from a modbus rtu slave in the setup() of a sketch and I'm doing this to do some testing:

uint16_t fanType[6] = {0, 0, 0, 0, 0, 0};
char fanType_hex[2];
String fanTypeName = "";

Serial2.begin(19200, SERIAL_8E1, RXD, TXD);
mb.begin(&Serial2, RXTXEnablePin);
mb.master();

  if (!mb.slave()) { 
    mb.readHreg(SLAVE_ID, FanType, fanType, 6, NULL);
// I think I need to wait here as this is done in setup()
    while (mb.slave()) mb.task();

    for (int8_t i = 0; i <= 5; i++) {
      Serial.println(fanType[i], HEX);
      fanType[i] = (fanType[i]>>8 | fanType[i]<<8);
      Serial.println((fanType[i]>>8) &0xFF);
      Serial.println(fanType[i] & 0xFF);
      sprintf(fanType_hex, "%02X", fanType[i]); 
      Serial.println(fanType_hex);
      fanTypeName += fanType_hex;
    }
  Serial.println(fanTypeName);

I receive 6 uint16_t values as expected, but one of the values is "missing".

This is the result:

334B
75
51
4B33
3200
0
50
32
3035
53
48
3530
5252
82
82
5252
3130
48
49
3031
0
0
0
00
4B333235305252303100

The second uint16_t is: 3200 (00 = \0) but if I read it with another MODBUS RTU controller I can see the value is really 3247 ("2G") which is part of the model name of the slave.

Any insights into why this can be?

Thank you in advance.

emelianov, 2020-06-08:
Hello,
It's not clear how the situation can take place.
For debug please take code of the library from 3.0.1 branch.
Code from the branch will dump incoming ModbusRTU packets to Serial. Please send this dump for analyses.

TelleRen2k, 2020-06-12:
Thank you for your feedback.

This is output with 3.01.

334B
75
51
4B33
3200
0
50
32
3035
53
48
3530
5252
82
82
5252
3130
48
49
3031
0
0
0
00
4B3332353052523031004B3332353052523031004B3332353052523031004B3332353052523031004B3332353052523031004B3332353052523031004B3332353052523031004B3332353052523031004B3332353052523031004B3332353052523031004B3332353052523031004B3332353052523031004B3332353052523031004B3332353052523031004B3332353052523031004B3332353052523031004B3332353052523031004B3332353052523031004B3332353052523031004B3332353052523031004B3332353052523031004B3332353052523031004B3332353052523031004B3332353052523031004B3332353052523031004B3332353052523031004B3332353052523031004B3332353052523031004B3332353052523031004B3332353052523031004B3332353052523031004B3332353052523031004B3332353052523031004B3332353052523031004B3332353052523031004B3332353052523031004B3332353052523031004B3332353052523031004B3332353052523031004B3332353052523031004B3332353052523031004B3332353052523031004B3332353052523031004B3332353052523031004B3332353052523031004B3332353052523031004B3332353052523031004B3332353052523031004B3332353052523031004B3332353052523031004B3332353052523031004B3332353052523031004B3332353052523031004B3332353052523031004B3332353052523031004B3332353052523031004B3332353052523031004B3332353052523031004B3332353052523031004B3332353052523031004B3332353052523031004B3332353052523031004B3332353052523031004B3332353052523031004B3332353052523031004B3332353052523031004B3332353052523031004B3332353052523031004B3332353052523031004B3332353052523031004B3332353052523031004B3332353052523031004B3332353052523031004B3332353052523031004B3332353052523031004B3332353052523031004B3332353052523031004B3332353052523031004B3332353052523031004B3332353052523031004B3332353052523031004B3332353052523031004B3332353052523031004B3332353052523031004B3332353052523031004B3332353052523031004B3332353052523031004B3332353052523031004B3332353052523031004B3332353052523031004B3332353052523031004B3332353052523031004B3332353052523031004B3332353052523031004B3332353052523031004B3332353052523031004B3332353052523031004B3332353052523031004B3332353052523031004B3332353052523031004B3332353052523031004B3332353052523031004B3332353052523031004B3332353052523031004B3332353052523031004B3332353052523031004B3332353052523031004B3332353052523031004B3332353052523031004B3332353052523031004B3332353052523031004B3332353052523031004B3332353052523031004B3332353052523031004B3332353052523031004B3332353052523031004B3332353052523031004B3332353052523031004B3332353052523031004B3332353052523031004B3332353052523031004B3332353052523031004B3332353052523031004B3332353052523031004B3332353052523031004B3332353052523031004B3332353052523031004B3332353052523031004B3332353052523031004B3332353052523031004B3332353052523031004B3332353052523031004B3332353052523031004B3332353052523031004B3332353052523031004B3332353052523031004B3332353052523031004B3332353052523031004B3332353052523031004B3332353052523031004B3332353052523031004B3332353052523031004B3332353052523031004B3332353052523031004B3332353052523031004B3332353052523031004B3332353052523031004B3332353052523031004B3332353052523031004B3332353052523031004B3332353052523031004B3332353052523031004B3332353052523031004B3332353052523031004B3332353052523031004B3332353052523031004B3332353052523031004B3332353052523031004B3332353052523031004B3332353052523031004B3332353052523031004B3332353052523031004B3332353052523031004B3332353052523031004B3332353052523031004B3332353052523031004B3332353052523031004B3332353052523031004B3332353052523031004B3332353052523031004B3332353052523031004B3332353052523031004B3332353052523031004B3332353052523031004B3332353052523031004B3332353052523031004B3332353052523031004B3332353052523031004B3332353052523031004B3332353052523031004B3332353052523031004B3332353052523031004B333235305252303100
03 0A 33 4B 32 47 30 35 52 52 31 30 0E 65

TelleRen2k, 2020-06-13:
I have created this test sketch:

```
#include <ModbusRTU.h>
#define SLAVE_ID 1
#define RXD                   26
#define TXD                   23
#define RXTXEnablePin         27
#define FanTypeAddress     53669   // D1A5

ModbusRTU mb;
unsigned long lastPollFanTime = 0;
uint16_t fanType[6] = {0, 0, 0, 0, 0, 0};
char fanType_hex[2];
String fanTypeName = "";

uint16_t res = 0;

void readFanType() {
  if (!mb.slave()) { 
    mb.readHreg(SLAVE_ID, FanTypeAddress, (uint16_t*)fanType, 6, NULL);  // Works fine without cast (uint16_t*)
    // Wait until transaction is complete
    while (mb.slave()) mb.task();

    // Swap bytes in each 16 bit integer and place in fanType_hex[]
    for (int8_t i = 0; i <= 5; i++) {
      Serial.print(i); Serial.print(": ");
      Serial.println(fanType[i]);
      Serial.println(fanType[i], HEX);
      fanType[i] = (fanType[i]>>8 | fanType[i]<<8);
      //Serial.println((fanType[i]>>8) &0xFF);
      //Serial.println(fanType[i] & 0xFF);
      sprintf(fanType_hex, "%02X", fanType[i]); 
//      Serial.println(fanType_hex);
      fanTypeName += fanType_hex;
    }
    Serial.println(fanTypeName);
  } // end if
}

void setup() {
  // Seriel Monitor
  Serial.begin(115200);

  Serial2.begin(19200, SERIAL_8E1, RXD, TXD);
  mb.begin(&Serial2, RXTXEnablePin);
  mb.master();

  readFanType();
}

void loop() {

  if ((millis() - lastPollFanTime) > 20000) {
    //readFanPower();
    lastPollFanTime  = millis();
  }
  mb.task();  
}
```

It generates this output:

```
03 0C 33 4B 32 47 30 35 52 52 31 30 32 48 D8 DB 
0: 13131
334B
1: 12800
3200
2: 12341
3035
3: 21074
5252
4: 12592
3130
5: 12872
3248
4B33323530525230314832
```

The debug output is correct: 03 0C 33 4B 32 47 30 35 52 52 31 30 32 48 D8 DB  

The second integer value is 12800 but it should be 12871?

How can this be?

emelianov, 2020-06-14:
See no errors in code except `char fanType_hex[2];` -- must be `char fanType_hex[3];` to store `/0` string terminator.
Probably this out of bounds resulting the issue.

TelleRen2k, 2020-06-15:
Ha, thank you very much for the review. Strange side effect for sure :-)

closed 2020-06-25 by emelianov