emelianov/modbus-esp8266 #46: I must be missing a simple thing here...

AutoDoc1955, 2020-06-08:
First of all, thank you for your work in putting this library together.  I'm a controls engineer and PLC programmer and am on ly over here in C++ Land when I have to be (like now).  I (using an ESP8266 as the master/client) am trying to understand how to read multiple, contiguous holding registers from a PLC.  I started with your MasterSimpleRead example and I am able to read any single register, but I cannot understand how to do multiples.  If I tack on another input parameter onto the "mb.readHReg" call at line 49 and set it to 2 for example I compile OK and run but, of course, only one register value comes back.  I'm thinking that my syntax is OK, but that I'm using an incorrect variable type to put the 2 registers of data into.  Should I be using an array of 16 bit words instead of a single word, and if so, how?  My apologies for asking a question that might be an obvious thing to a C++ guy, but I'm stumped here today.

emelianov, 2020-06-09:
Hello,
For multiple registers it should be something like
```c
...
uint16_t res[2];
...
void loop() {
  if (mb.isConnected(remote)) {   // Check if connection to Modbus Slave is established
    mb.readHreg(remote, REG, &res, 2);  // Initiate Read Hreg from Modbus Slave
  } else {
...
```

AutoDoc1955, 2020-06-09:
Thanks for your quick reply.

Yes, that's what I thought as well, but it won't compile...
"no matching function for call to 'ModbusIP::readHreg(IPAddress&, const int&, uint16_t (*)[2], int)'"

Any other ideas?

emelianov, 2020-06-09:
It's not  clear what compiler you are using. Anyway try `mb.readHreg(remote, REG, (uint16_t*)res, 2);` notation.

AutoDoc1955, 2020-06-09:
That did it!  Thanks so much.  Again, I'm not really a C++ guy by nature.  Could you explain the use of the parentheses around the uint16_t* data type?

BTW, here's how I'm compiling: Arduino IDE 1.8.10 with ESP8266 hardware library 2.6.3.

emelianov, 2020-06-10:
Indeed third parameter of readHreg is pointer to uint16_t that is declared as uint16_t* - pointer to one or multiple uint16_t values. (uint16_t*)res overrides res type to be uint16_t* to avoid warnings or error from compiler.
It's strange that Arduino generates error for code without (uint16_t*) override - as it was built without errors in my environment.

AutoDoc1955, 2020-06-10:
We must chalk it up as another mystery in the world of software.  Thanks again for you help.

closed 2020-06-25 by emelianov