emelianov/modbus-esp8266 #47: Not obvious examples.

zekageri, 2020-06-19:
Hi! First of all thanks for this amasing lib. I don't have the luck to try it out because some minor things.

I have a sample sketch:

```
#include <ModbusRTU.h>
ModbusRTU mb;

bool cbWrite(Modbus::ResultCode event, uint16_t transactionId, void* data) {
  String Cucc = String(event);
  Cucc += "id: ";
  Cucc += String(transactionId);
  //Cucc += "Data: ";
  //Cucc += String(data);
  Send_Async(Cucc,"~ModBus");
  return true;
}

void Pupu(){
  Serial.begin(115200, SERIAL_8N1);
  mb.begin(&Serial,5);
  mb.master();
}

long proba = millis();
void Proba(){
  if(millis() - proba >= 1000){
    proba = millis();
    //mb.readHreg(11,0,&Start_In_Reg_Address,Reg_Count,cbWrite);
    if (!mb.slave()) {
      mb.writeHreg(11,Start_In_Reg_Address,255,1,cbWrite);
    }
  }
  mb.task();
}

void setup(){
  Pupu();
}

void loop(){
  Proba();
}
```

**I want to understand the multi reads and multi writes.
So i'am using esp32 and in my understanding the writing multi reg needs an address, a start reg address, a value to write to the regs and a register count.** 
### This sketch gives me an error:

`src\main.cpp:28:57: error: invalid conversion from 'int' to 'uint16_t* {aka short unsigned int*}' [-fpermissive]`

_Would you be so kind and help me in this situation?_

Oh and i also don't understand the void* data in the callback. O_o

emelianov, 2020-06-20:
Hello,
To write multiple registers you have to declare array of values to write. Something like:
```c
...
uint16_t regs = {1, 2, 3};
...
mb.writeHreg(11,Start_In_Reg_Address,regs,3,cbWrite);
```
Only single register write is possible using constant value (note register count is not used in this case):
```c
mb.writeHreg(11,Start_In_Reg_Address,255,cbWrite);
```

`void* data` is reserved and not used.

`uint16_t transactionId` is used in case of ModbusIP only.

zekageri, 2020-06-20:
### So if i want the whole data in the callback what should i write?

**If i declare the cbWrite like this:**

```
bool cbWrite(Modbus::ResultCode event, uint16_t* data) {
  String Cucc = String(event);
  Cucc += "id: ";
  Send_Async(Cucc,"~ModBus");
  return true;
}
```

_I get an error again. The only thing i want in the callback is the returned data and the address._

zekageri, 2020-06-20:
**Oh and the lib is handling the third transmit pin like this?**

`mb.begin(&Serial,5);`

**So if i want to write, does the lib raise the pin 5 to high for me or do i have to do it myself?**

zekageri, 2020-06-20:
Sorry everything is okay now.

closed 2020-06-20 by zekageri