emelianov/modbus-esp8266 #48: Question about reading slaves in a for loop!

zekageri, 2020-06-20:
### I'am using an esp32. At startup i want to read all the slaves on the bus.
My max  slave count is 128, starting address is 10.

_I want all my existing slaves in an array to be able to loop on the existing slaves at runtime._ 
**I created this code:**

```
#define MAX_EXPANDERS 128
boolean First_Read_Mode = false;
int Existing_Expanders[MAX_EXPANDERS];
String Ascii_Data = "";
int Exp_Arr_Counter = 0;
uint16_t regs[44];

const char ASCIIARR[] = {
    '0','1','2','3','4','5',
    '6','7','8','9',':',';',
    'A','B','C','D','E','F',
    'G','H','I','J','K','L',
    'M','N','O','P','Q','R',
    'S','T','U','V','W','X',
    'Y','Z','a','b','c','d',
    'e','f','g','h','i','j',
    'k','l','m','n','o','p',
    'q','r','s','t','u','v',
    'w','x','y','z',' ',
  };
const int ASCIINUM[] = {
    48,49,50,51,52,53,
    54,55,56,57,58,59,
    65,66,67,68,69,70,
    71,72,73,74,75,76,
    77,78,79,80,81,82,
    83,84,85,86,87,88,
    89,90,97,98,99,100,
    101,102,103,104,105,106,
    107,108,109,110,111,112,
    113,114,115,116,117,118,
    119,120,121,122,32,
  };

static inline char Get_ASCII(int Data){
    char Found_ASCII;boolean Found = false;
    for(byte i = 0; i < (sizeof(ASCIINUM) / sizeof(ASCIINUM[0])); i++){
        if(ASCIINUM[i] == Data){
        Found_ASCII = ASCIIARR[i];
        Found = true;
        break;
        }
    }
    if(Found){
        return Found_ASCII;
    }else{
        return '-';
    }
}

bool cbRead(Modbus::ResultCode event, uint16_t transactionId, void* data) {
    if(First_Read_Mode){
        for(int i = 0; i < 44;i++){
            if(i == 0){
                Existing_Expanders[Exp_Arr_Counter] = regs[i];
                Exp_Arr_Counter++;
            }else{
                Ascii_Data += Get_ASCII(regs[i]);
            }
        }
    }
  return true;
}

static const inline void Sync_ModBus_Setup(){
    Serial.begin(115200, SERIAL_8N1);
    mb.begin(&Serial,5);
    mb.master();
}

static const inline void Read_All_Expanders(){
    First_Read_Mode = true;
    uint8_t Start_Address = 10;
        for(int i = 0; i < MAX_EXPANDERS;i++){
            mb.readHreg(Start_Address,0x4064,regs,0x2c,cbRead);
            Start_Address++;
            vTaskDelay(10);
        }
    First_Read_Mode = false;
    Exp_Arr_Counter = 0;
}

void ModBus_Task( void * parameter ){
Sync_ModBus_Setup();
Read_All_Expanders();
  for ever{
    Read_Existing_Expanders();
    mb.task();
    vTaskDelay(1);
  }
}
```

**As you can see, at startup the program wants to read all the slaves until it reaches 128 ( max slaves ). When in first readd mode inside the cbRead callback i try to separate the address from the rest of the data. The first data is the address.
The problem is that i don't get any data.** 

### If i define the existing modules like this: 

`int Existing_Expanders[MAX_EXPANDERS] = {11,14};  // The two slaves currently on the BUS`

_The reading is working fine._

**Is it not recommended to read the slaves inside a for loop?**

zekageri, 2020-06-20:
### Okay i see the problem now.

**The code should be like this:**

```

bool cbReadInfo(Modbus::ResultCode event, uint16_t transactionId, void* data) {
  return true;
}

static const inline void Get_Infos(){
    for(int i = 0; i < 44;i++){
            if(i == 0){
                Ascii_Data += Info_regs[i];
                //Existing_Expanders[Exp_Arr_Counter] = Info_regs[i];
                //Exp_Arr_Counter++;
            }else{
                Ascii_Data += Get_ASCII(Info_regs[i]);
            }
        }
        Ascii_Data += "\n";
}

static const inline void Read_All_Expanders(){
    uint8_t Start_Address = 10;
        for(int i = 0; i < MAX_EXPANDERS;i++){
            if (!mb.slave()) {
                mb.readHreg(Start_Address,0x4064,Info_regs,0x2c,cbReadInfo);
                Start_Address++;
                while (mb.slave()){
                    mb.task();
                }
                Get_Infos();
            }
        }
    Exp_Arr_Counter = 0;
}
```
**But how can i get the address of the slave from the response?**

emelianov, 2020-06-20:
Blocking 
```c
while (mb.slave()){
                    mb.task();
                }
```
is Ok if you are using dedicated thread for Modbus processing. Just note that the library is not thread-save at this point.

> **But how can i get the address of the slave from the response?**

If you mean the way to determine SlaveID from callback you should use `mb.eventSource()` to get it.

zekageri, 2020-06-20:
I'am running a continous read from the slaves in a separate task.
To get the slaves input pin states. ( these are I/O Expanders )
Because i want to get the PIN changes on the different expanders as fast as possible.

But on the other hand, in a separate task, i want to write to one of the expanders while the read is happening on the other.
This is a little bit tricky for me because the bus is busy while i read, and i can't write meanwhile. 

So i wanted to block the read while the write is happening but with not much luck. :/

zekageri, 2020-06-20:
The continous read is looks like this:

```
static const inline void Read_Existing_Expanders(){
    if(!Main_is_Writing){
    if (!mb.slave()) {
        mb.readHreg(Existing_Expanders[Exp_Arr_Counter],0x4000,Slave_Input_Reg,0x0001,cbRead);
        while (mb.slave()){mb.task();}
        Exp_Arr_Counter++;
        if(Existing_Expanders[Exp_Arr_Counter] == 0){Exp_Arr_Counter = 0;}
    }
    }
}
```

It's running in its own task.

Meanwhile inside an other task, if i get an interrupt from somewhere else ( its an async server callback usually or one of the slave input change pins) i want to write to one of the slaves outputs like this:

```
uint16_t PINS[1] = {0000000000000000};
bool cbWriteSmart(Modbus::ResultCode event, uint16_t transactionId, void* data) {
    return true;
}

static const inline void Write_To_Expand(uint16_t Address,int PIN){
    if(bitRead(PINS[0],PIN) == 0){bitSet(PINS[0],PIN);}
    else{bitClear(PINS[0],PIN);}
    Main_is_Writing = true;
    if (!mb.slave()) {
        mb.writeHreg(Address,0x4032,PINS,1,cbWriteSmart);
        while (mb.slave()){mb.task();}
        Main_is_Writing = false;
    }
}

void Smart_Switch_Run( void * parameter ){
  for ever{
    if(xQueue == NULL){return;}
    struct QueueSmart PINProp;
    xQueueReceive(xQueue, &PINProp, portMAX_DELAY);
    Write_To_Expand(PINProp.Address,PINProp.Pin);
  }
}
```
I tried with a boolean variable. If the Main_is_Writing then the read should stop. But it is working just half of the time for some reason. :/

zekageri, 2020-06-20:
### Now i tried to suspend the reader task while i write in another task. Like this:

```
static const inline void Write_To_Expand(uint16_t Address,int PIN){
    if(bitRead(PINS[0],PIN) == 0){
        bitSet(PINS[0],PIN);
    }else{
        bitClear(PINS[0],PIN);
    }
    vTaskSuspend(ModBus_Task_Handle);
    vTaskDelay(50);
    if (!mb.slave()) {
        mb.writeHreg(Address,0x4032,PINS,1,cbWriteSmart);
        while (mb.slave()){mb.task();}
    }
    vTaskDelay(50);
    vTaskResume(ModBus_Task_Handle);
}
```

**This solution is a bit more stable than the other ones but the writeHreg sometimes not writing the value. 
Is it possible?**

emelianov, 2020-06-20:
You need to use semaphore not variable flagging for arbitrating access to Modbus read/write operations. Easy way is just take semaphore instead of `Main_is_Writing = true;` and give them back instead of `Main_is_Writing = false;` in both read and write tasks.

emelianov, 2020-06-20:
Just random thought: if you lost slave device send/receive process will stale for 1 second (default timeout value) each time the device is queried.

zekageri, 2020-06-21:
### Thank you for your help and understanding. 
In the end, i could managed to work this by running the two tasks in a separate core.

**My continous reader task is running on core 1:**
`xTaskCreate(ModBus_Task,"ModBus_Setup",7000,NULL,1,&ModBus_Task_Handle);`
**And the Switch task ( wich is switching the expanders pins ) is running on core 0:**
`xTaskCreatePinnedToCore(Smart_Switch_Run,"Smart_Switch_Run",10000,NULL,0,&Smart_Task_Handle,0);`

That way, i can use
`vTaskSuspend(ModBus_Task_Handle);`
while writing to one of the expanders and after that i can
`vTaskResume(ModBus_Task_Handle);`

### The only thing is that i have to wait for like 50 millis before writing after i suspended the task, and like 50millis after write before resuming the task.

**Like this:**

```
bool cbWriteSmart(Modbus::ResultCode event, uint16_t transactionId, void* data) {
    if(event == mb.EX_SUCCESS){
        return true;
    }else{
        return false;
    }
}

static const inline void Write_To_Expand(uint16_t Address,int PIN,String ID){
    vTaskSuspend(ModBus_Task_Handle);
    vTaskDelay(50);
    if (!mb.slave()) {
        if(bitRead(Saved_PINS[0],PIN) == 0){
          bitSet(PINS[0],PIN);
        }else{
          bitClear(PINS[0],PIN);}
        if(mb.writeHreg(Address,0x4032,PINS,1,cbWriteSmart)){
            Saved_PINS[0] = PINS[0];Send_Async(ID,";Gomb");
        }
        while (mb.slave()){mb.task();}
    }
    vTaskDelay(50);
    vTaskResume(ModBus_Task_Handle);
}
```

### Maybe with semaphore, i can manage it without delay. I will try it out!

zekageri, 2020-06-21:
> Just random thought: if you lost slave device send/receive process will stale for 1 second (default timeout value) each time the device is queried.

I Edited this to 15millisec timeout. This is the maximum the program should wait for me! Thanks.
```
#define MB_MAX_TIME 10
#define MODBUSRTU_TIMEOUT 15
```

zekageri, 2020-06-23:
Semaphore version is working properly. 
Thank you for the help!

closed 2020-06-23 by zekageri