emelianov/modbus-esp8266 #51: Help getting started with two ESP8266

whnr, 2020-06-27:
Hi!

I need some help getting started. I initially was naive and just wanted to connect an esp to a sensor and read it. That did not work so I tried connection to ESP8266s together. Electrically the signals get transmitted from through two MAX3485 and the Tx signal from the master makes it to the slave just fine.

I used a slightly modified code from the examples. Please help me to understand why they are not communicating. The only error code I get is `0xE4`.

### Master

``` cpp
#include <Arduino.h>

#include <SoftwareSerial.h>    // https://github.com/PaulStoffregen/SoftwareSerial
#include <ModbusRTU.h>
#include <ESP8266WiFi.h>

#define MODBUSRTU_DEBUG

int incomingByte = 0;

#define RX        D5    // Soft Serial RS485 Receive pin
#define TX        D4   // Soft Serial RS485 Transmit pin
#define RTS       D7    // RS485 Direction control

SoftwareSerial RS485(RX, TX);
ModbusRTU mb;

bool cbWrite(Modbus::ResultCode event, uint16_t transactionId, void* data) {
  Serial.printf_P("Request result: 0x%02X, Mem: %d\n", event, ESP.getFreeHeap());
  return true;
}

void setup() {
//  pinMode(RTS, OUTPUT);  
  Serial.begin(115200);
  RS485.begin(300, SWSERIAL_8N2);
  Serial.println("Booting");
  delay(4000);
  
  mb.begin(&RS485, RTS);
  mb.setBaudrate(300);
  mb.master();
}

uint16_t test;

void loop() {
    if (!mb.slave()) {
      mb.readHreg(1, 10, &test, 1, cbWrite);
      Serial.println("readHreg");
    }
    mb.task();
    yield();
    Serial.print("Received: ");
    Serial.println(test);
    delay(2000);
}
```

The only output i get from this code is:
```
readHreg
Received: 0
Request result: 0xE4, Mem: 48552
Received: 0
```

### Slave

``` cpp
/*
  ModbusRTU ESP8266/ESP32
  Simple slave example

  (c)2019 Alexander Emelianov (a.m.emelianov@gmail.com)
  https://github.com/emelianov/modbus-esp8266

  modified 13 May 2020
  by brainelectronics

  This code is licensed under the BSD New License. See LICENSE.txt for more info.
*/

#include <Arduino.h>

#include <SoftwareSerial.h>    // https://github.com/PaulStoffregen/SoftwareSerial
#include <ModbusRTU.h>
#include <ESP8266WiFi.h>

#define MODBUSRTU_DEBUG
#define REGN 10
#define SLAVE_ID 1

#define RX        D4    // Soft Serial RS485 Receive pin
#define TX        D1   // Soft Serial RS485 Transmit pin
#define RTS       D7    // RS485 Direction control

SoftwareSerial RS485(RX, TX);
ModbusRTU mb;

void setup() {
  Serial.begin(115200);
  RS485.begin(300, SWSERIAL_8N2);
  Serial.println("Booting");
  
  mb.begin(&RS485, RTS);
  mb.setBaudrate(300);
  mb.slave(SLAVE_ID);
  mb.addHreg(REGN);
  mb.Hreg(REGN, 100);
}

void loop() {
  mb.task();
  yield();
}
```

emelianov, 2020-06-28:
Hello.
In your implementation it may be looks like
```c
void loop() {
    if (!mb.slave()) {
      mb.readHreg(1, 10, &test, 1, cbWrite);
      Serial.println("readHreg");
    }
    while (mb.slave()) {
      mb.task();
      yield();  // Not needed in this particular case
    }
    Serial.print("Received: ");
    Serial.println(test);
    delay(2000);
}
```
Indeed time between .task() calls should not be longer than timeout value (that is 1 Second by default).
Also note that .readHreg() not fetches value but just sending request to slave. Response processed by .task() as soon as data arrive.

whnr, 2020-06-28:
Thank you @emelianov. It is still timing out. Can I somehow `Serial.print` everything that is being received (and successfully decoded) over Modbus on the slave? I will try to dive deeper into `task()` and maybe come up with a few comments and documentation.

Sidenote: I increased the baudrate again from 300 to 19200 (which did not change anything).

whnr, 2020-06-28:
I did some more debugging and it seems to be something on the hardware side. I skipped the RS485 transceivers and directly connected Rx/Tx crossed over and it works! Now I have to dive deeper into why it did not work.

Maybe it is a timing enable issue on the ready to send pin?

TODO: I will update this post for future reference once I found the reason.

closed 2020-06-29 by whnr