emelianov/modbus-esp8266 #53: How to add Line control for ESP32 modbus RTU slave?

heroseller17, 2020-07-03:


emelianov, 2020-07-03:
Hello,
Data flow control pin should be specified in .begin() as below:
```c
 mb.begin(&Serial, REDE_PIN);
```
Complete code for basic slave device is something like:
```c
#include <ModbusRTU.h>
#define SLAVE_ID 1
#define RX_PIN 17
#define TX_PIN 18
#define REDE_PIN 5

ModbusRTU mb;

void setup() {
  Serial1.begin(9600, SERIAL_8N1, RX_PIN, TX_PIN);
  mb.begin(&Serial1, REDE_PIN);
  mb.slave(SLAVE_ID);
  mb.addCoil(Sensor1_COIL_addr);
  mb.addCoil(Sensor4_COIL_addr);
  mb.addCoil(Valve3_COIL_addr);
  mb.addCoil(Motor_COIL_addr);
}

void loop() {
  mb.task();
  delay(10);
}
```

Note to power ADM485 for 3.3V not 5V.

heroseller17, 2020-07-04:
Dear emelianov ,
Thanks for your response.
actually this is my setup,i use ESP32 Serial2 for rs485 communication :
here is the code:
`#define RX_PIN 16
#define TX_PIN 17
#define REDE_PIN 2
 `
i get response without RS485 but when i use RS485 it doesn't work.
i use modpoll in my laptop .
what is the problem?

emelianov, 2020-07-07:
I suspect wiring or bitrate problem. Try to use 9600.

closed 2020-07-17 by emelianov

Voha888, 2021-07-27:
But how to set or get values of this Coils? Where stored its values? ThankYou

> Hello,
> Data flow control pin should be specified in .begin() as below:
>
> ```c
>  mb.begin(&Serial, REDE_PIN);
> ```
>
> Complete code for basic slave device is something like:
>
> ```c
> #include <ModbusRTU.h> [...]

emelianov, 2021-07-27:
@Voha888 

> But how to set or get values of this Coils? Where stored its values? I not see any array or bool variables. Thank You

```c
bool coil = mb.Coil(Sensor1_COIL_addr); // Get current coil value
mb.Coil(Sensor2_COIL_addr, true); // Set coil value
```

Value are stored internally by the library code.

Voha888, 2021-07-27:
Thank You very match! 
When I am at the computer, I will make examples with this code and add them to your repository.

One small question.  Should the master also store data inside the library?  Or is it necessary for the master to create an external array?  Thank you!

emelianov, 2021-07-27:
@Voha888 

Master stores received registers values (data) to user variable (or array) that was specified not internally.

Voha888, 2021-07-29:
Thank You!
I want the master to be able to write and read Coils.
But if I write like this, then only reading Coils works:
```
void loop () {
if (! mb.slave ()) {
mb.writeCoil (SLAVE_ID, OFFSET_OUTPUTS, output_coils, NUM_OUTPUTS, cbWrite);
mb.readCoil (SLAVE_ID, OFFSET_INPUTS, input_coils, NUM_INPUTS, cbWrite);
  }
mb.task ();
 }
}
```
and like this, then only reading Coils works:
```
void loop () {
if (! mb.slave ()) {
mb.writeCoil (SLAVE_ID, OFFSET_OUTPUTS, output_coils, NUM_OUTPUTS, cbWrite);
mb.task ();
mb.readCoil (SLAVE_ID, OFFSET_INPUTS, input_coils, NUM_INPUTS, cbWrite);
mb.task ();
  }
 }
}
```
The option in which both reading and writing works is as follows - in one cycle either read or write Coils:
```
void loop () {
 if (! mb.slave ()) {
    if (need_update_output_coils) {
mb.writeCoil (SLAVE_ID, OFFSET_OUTPUTS, output_coils, NUM_OUTPUTS, cbWrite);
need_update_output_coils = 0;
  } else {
mb.readCoil (SLAVE_ID, OFFSET_INPUTS, input_coils, NUM_INPUTS, cbWrite);
  }
 }
  mb.task ();
}
```
Can several operations be done in one cycle? For example, call read and write Colis

emelianov, 2021-07-29:
`read*`/`write*` functions just sends request to slave and returns. According to Modbus specification next request can be send only after result is returned by slave device (or timeout reached). So then you calling `readCoil` immediate after `writeCoil` nothing happens as library state is `waiting for response` at this point.
Simplest (but not the best) way is to add
```c
while (ms.slave()) {
  mb.task();
  yield();
}
```
after `read*`/`write*` to make sure the operation is complete.

Voha888, 2021-07-29:
Thank you! I will also consult with my friend who has experience with modbus how best to implement everything. Thank you very much!