emelianov/modbus-esp8266 #55: RTU on ESP32 not working reliable if mb.task is called too often

DanielWeigl, 2020-07-20:
If I call the `mb.task()` function in the `void loop()` on every loop (and have nearly no other code in the loop) the receiver often fails (but not always).

To better debug it, i added some `Serial.print`s into the `mb.task` function:
```
void ModbusRTU::task() {
	#ifdef ESP32
	if (_len == 0)
		// portENTER_CRITICAL(&mux);
	#endif
    if (_port->available() > _len)	{
        _len = _port->available();
        t = millis();
        Serial.print("L");
        Serial.println(_len);
		return;
    }
    if (_len != 0 && millis() - t < _t) { // Wait data whitespace if there is data
        Serial.print("W.");
        Serial.print(_len);
        Serial.print("D.");
        Serial.println(millis() - t);
        return;
    }

	#ifdef ESP32
    // portEXIT_CRITICAL(&mux);
 	#endif
    if (isMaster) cleanup();
	if (_len == 0)
		return;

    Serial.print("OK");
    Serial.println(_len);
...
```

I then get this output:
```
L1
W.1D.0
W.1D.0
W.1D.0
W.1D.0
W.1D.1
W.1D.1
W.1D.1
W.1D.1
W.1D.1
W.1D.1
W.1D.1
W.1D.1
W.1D.1
W.1D.1
W.1D.1
W.1D.1
W.1D.9
W.1D.17
OK1

L12
W.12D.11
OK12
```

Which means, the `mb.task` for some reason only gets one byte in ` _port->available()` and then waits until the timeout happens, then tries to parse this one byte, but discards it, because its not valid.
After that it gets the reminding data (12byte) but also discards it, because the 1byte from the first iteration is missing.

If I add a `delay(100)` in the main-loop it works better, but also not 100% of the time.

ESP32, with platform.io
```
[env]
platform = espressif32
board = nodemcu-32s
framework = arduino
lib_deps =
    modbus-esp8266
```
The modbus is on Serial2:
```
   // Setup ModBus
    Serial2.begin(9600, SERIAL_8N1, MODBUS_RXD, MODBUS_TXD);
    mb.begin(&Serial2, -1);
    mb.slave(SLAVE_ID);
```

```
void loop() {
    ArduinoOTA.handle();
    mb.task();
    // otherwise mb.task makes problems?
//        delay(150);
}
```
Any ideas what could be wrong?

DanielWeigl, 2020-07-20:
Nb: For testing I already tried to disable the locking in `mb.task()` (as seen in the snippet above, the commented `// portENTER_CRITICAL(&mux);`), because I was afraid this might "dead-lock" ESP32 internals from updating the `Serial2` state, but this does not change the behavior at all:

emelianov, 2020-07-20:
Hello Daniel,
1. You'd better use last release of the library (part of code mentioned in your first post was refactored).
2. Try to use ESP32 Arduino from master branch not 1.04 (current last release that has some problems with UART). Code from master branch eliminates trobles in my tests.

DanielWeigl, 2020-07-20:
Hi, thank you for your fast feedback, 

ad1)  Im using the latest modbus lib:
```
Dependency Graph
|-- <modbus-esp8266> 3.0.1
|   |-- <WiFi> 1.0
```
ad2) Im not completely sure whats the best way to compile against master of the android ESP32 stack, but I tried to simply replace the content of the folder ` ~/.platformio/packages/framework-arduinoespressif32` with the current development state via `git clone https://github.com/espressif/arduino-esp32.git`. It seems to use this code (i see the files in the compile log), but it does not change the above described behavior.

emelianov, 2020-07-20:
Code (provided in your first message)
```c
void ModbusRTU::task() {
	#ifdef ESP32
	if (_len == 0)
		// portENTER_CRITICAL(&mux);
	#endif
```
is not from 3.0.1 release nor current master branch. It's outdated buggy variant of task() implementation.

Also i'm unable to reproduce the problem when building sketch from Arduino IDE:
```c
void loop() {
  mb.task();
}
```
running on ESP32 processing requests correctly.

Can you provide any additional info/code to make me able to reproduce the issue?

DanielWeigl, 2020-07-20:
Hm, thats strange.

I just deleted the projects' `.pio` directory to clear the local cache and fetched all libs again:
```
HARDWARE: ESP32 240MHz, 320KB RAM, 4MB Flash
DEBUG: Current (esp-prog) External (esp-prog, iot-bus-jtag, jlink, minimodule, olimex-arm-usb-ocd, olimex-arm-usb-ocd-h, olimex-arm-usb-tiny-h, olimex-jtag-tiny, tumpa)
PACKAGES: 
 - framework-arduinoespressif32 4d98cea 
 - tool-esptoolpy 1.20600.0 (2.6.0) 
 - toolchain-xtensa32 2.50200.80 (5.2.0)
LDF: Library Dependency Finder -> http://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Looking for modbus-esp8266 library in registry
Found: https://platformio.org/lib/show/7364/modbus-esp8266
LibraryManager: Installing id=7364
modbus-esp8266 @ 3.0.1 has been successfully installed!
Looking for Adafruit NeoPixel library in registry
Found: https://platformio.org/lib/show/28/Adafruit NeoPixel
LibraryManager: Installing id=28
Adafruit NeoPixel @ 1.5.0 has been successfully installed!
Found 42 compatible libraries
Scanning dependencies...
Dependency Graph
|-- <DallasTemperature> 3.8.1
|   |-- <OneWire> 2.3.5
|-- <ArduinoOTA> 1.0
|   |-- <ESPmDNS> 1.0
|   |   |-- <WiFi> 1.0
|   |   |   |-- <SimpleBLE> 1.0
|   |-- <Update> 1.0
|   |-- <WiFi> 1.0
|   |   |-- <SimpleBLE> 1.0
|-- <modbus-esp8266> 3.0.1
|   |-- <WiFi> 1.0
|   |   |-- <SimpleBLE> 1.0
|-- <Adafruit NeoPixel> 1.5.0
|-- <OneWire> 2.3.5
|-- <WiFi> 1.0
|   |-- <SimpleBLE> 1.0
Building in release mode
```
So, it downloads the "3.0.1" from https://platformio.org/lib/show/7364/modbus-esp8266

But then for comparison I checked the downloaded lib against the code from this git repo and it had diffs all over it, even so, both say they are "3.0.1":
[image]

(left is git version, right the version from platform.io)

So there might be an corrupted upload of something.

Ill try to compile against your version from this repo and see if it makes a difference

DanielWeigl, 2020-07-20:
Yes, seems to work now 100% reliable.

I changed the `platformio.ini` to

```
[env]
platform = espressif32
platform_packages =
    framework-arduinoespressif32 @ https://github.com/espressif/arduino-esp32.git
board = nodemcu-32s
framework = arduino
lib_deps =
    git+https://github.com/emelianov/modbus-esp8266.git
    ...
```

Compiled and uploaded and all of the tests i made now worked flawlessly - Thanks for your help!

Maybe check again if the released version on platform.io is correct or what went wrong there and maybe make a new release version, so there are now mixups.

(thx btw, for maintaining this library anyway - is really nice to use!)

emelianov, 2020-07-21:
> (thx btw, for maintaining this library anyway - is really nice to use!)

You are welcome.

Thank you for pointing to Platformio version inconsenency. I think it's because of the way they using for version control. I'll plan minor release just to become in-sync next days.

closed 2020-07-21 by DanielWeigl

hrushikesh1024, 2020-11-17:
@DanielWeigl 
`platform = espressif32
platform_packages =
 framework-arduinoespressif32 @ https://github.com/espressif/arduino-esp32.git
board = nodemcu-32s
framework = arduino
lib_deps =
    git+https://github.com/emelianov/modbus-esp8266.git
    ...`

I used this in `platformio.ini` but getting the following error while compiling.

`Traceback (most recent call last):
  File "/Users/hrushi/.platformio/packages/tool-esptoolpy@1.20600.0/esptool.py", line 57, in <module>
    import serial.tools.list_ports as list_ports
  File "/Users/hrushi/.platformio/penv/lib/python3.8/site-packages/serial/tools/list_ports.py", line 29, in <module>
    from serial.tools.list_ports_posix import comports
  File "/Users/hrushi/.platformio/penv/lib/python3.8/site-packages/serial/tools/list_ports_posix.py", line 31, in <module>
    from serial.tools.list_ports_osx import comports
  File "/Users/hrushi/.platformio/penv/lib/python3.8/site-packages/serial/tools/list_ports_osx.py", line 32, in <module>
    kIOMasterPortDefault = ctypes.c_void_p.in_dll(iokit, "kIOMasterPortDefault")
ValueError: dlsym(RTLD_DEFAULT, kIOMasterPortDefault): symbol not found
*** [.pio/build/esp32dev/firmware.bin] Error 1`

How to resolve this?

emelianov, 2020-11-17:
Hello,

It’s about ESP-IDF and macOS BigSur compatibility.

Take a look this thread:
https://community.platformio.org/t/macos-big-sur-dlsym-rtld-default-kiomasterportdefault-symbol-not-found/17129/16

hrushikesh1024, 2020-11-17:
@emelianov 
Thank you !
I will try this tomorrow and see if it works..

hrushikesh1024, 2020-11-21:
@emelianov 

It solved the previous error but now it is showing the warnings, as attached below, and the even if the program is getting compiled the Modbus is not working on the ESP device after I flash it.

But the same program when I compile and flash it using Arduino software it works fine.
[image]

emelianov, 2020-11-22:
It's hard to say what wrong with code build as internally Plaformio and Arduino uses the same tools to create ESP firmware.
(Warnings you received has no affect on binary code)