emelianov/modbus-esp8266 #56: ESP32S not working with modbus TCP Server

rayoffleon, 2020-08-06:
Hi, im trying to make TCP server to work with a client in Java, i have tested that client with the  EasyModbusTCP simulator,
my device is a ESP32S (NodeMCU-32S) the server starts correctly, i am debuggin with the callback fuctions and the
connection from the client is arriving, but the read on the register is not working.

The test is based on the Test Holding Register example, i just changed it to Input Register.

Regards

`
#ifdef ESP8266
#include <ESP8266WiFi.h>
#else //ESP32
#include <WiFi.h>
#endif
#include <ModbusIP_ESP8266.h>

//ModbusIP object
ModbusIP mb;

bool cbConn(IPAddress ip) {
  Serial.println(ip);
  return true;
}

uint16_t cbMod(TRegister* reg, uint16_t val){

  Serial.println("Val: "+val);
  return true;
  }

void setup() {
  Serial.begin(115200);

  WiFi.begin("anynet", "1245787845");

  while (WiFi.status() != WL_CONNECTED) {
    delay(500);
    Serial.print(".");
  }

  Serial.println("");
  Serial.println("WiFi connected");
  Serial.println("IP address: ");
  Serial.println(WiFi.localIP());

  mb.onConnect(cbConn);
  mb.onGetIreg(0,cbMod);
  mb.onGetIreg(1,cbMod);
  mb.onGetIreg(2,cbMod);
  mb.server();
  mb.addIreg(1, 0);
  mb.addIreg(2, 17151);

}

void loop() {
  //Call once inside loop() - all magic here
  mb.task();
  delay(10);
}`

emelianov, 2020-08-06:
Hello,
1. Callback should return value to respond. Just `val` for passthrough.
```c
uint16_t cbMod(TRegister* reg, uint16_t val){

Serial.println("Val: "+val);
return val;
}
```
2. What do you mean under ` read on the register is not working.` ?
2.1. If debug message from `cbMod()` is printed during request?
2.2. What is error code on `EasyModbusTCP` side?
2.3. What Ireg number you are requesting from simulator?

rayoffleon, 2020-08-06:
The callback cbMod is not printing, i have changed it to:

`uint16_t cbMod(TRegister* reg, uint16_t val){
Serial.println("Val: "+val);
return val;
}`

It is working now, i think the ESP32 was (and still does) freezing because of the several connections,
looking at the serial plotter i see that after 80 to 90 connections the ESP32S freezes.

From my client i call this code in java:

`
float devuelveValor(int register,int longitud) {
        float dato = 0;

        ModbusClient modbusClient = new ModbusClient("192.168.1.138", 502);
        try {
            modbusClient.Connect();

            dato = ModbusClient.ConvertRegistersToFloat(modbusClient.ReadInputRegisters(register, longitud), ModbusClient.RegisterOrder.LowHigh);
            System.out.println(String.format("%.2f", dato));

          modbusClient.Disconnect();

        } catch (Exception e) {
            System.out.println(e);
        }

        return dato;
    }`

With each field i call this method once, and i have 9 fields refreshed every 2 seconds, i must say, that i was consulting some register that i wasn't assigned from the ESP32S, i think it could be the problem, anyway the server EasyModbusTCP doesn't freeze, with the same code in the client.

To make it work i now just calling the method for one field (2 registers that are "registered" on the ESP32S).

My client web looks like:

[image]

The EasyModbusTCP server looks like:
[image]

I will continue to make tests, to improve the application. Sorry for my bad English.

Thanks.

emelianov, 2020-08-06:
Connection count supported by ESP is limited. Also even after connection closed on client side server side may keep it open (or semi-open) for a while.
The library itself limited to 4 connections per Modbus instance.
In short i suggest to keep connection persistent or at least combine field requests to single session.
```java
try {
        modbusClient.Connect();

        dato1 = ModbusClient.ConvertRegistersToFloat(modbusClient.ReadInputRegisters(register1, 2), ModbusClient.RegisterOrder.LowHigh);
        System.out.println(String.format("%.2f", dato1));
        dato1 = ModbusClient.ConvertRegistersToFloat(modbusClient.ReadInputRegisters(register2, 2), ModbusClient.RegisterOrder.LowHigh);
        System.out.println(String.format("%.2f", dato2));
//... etc
      modbusClient.Disconnect();

    } catch (Exception e) {
        System.out.println(e);
    }
```

> Sorry for my bad English.

It's ok. I'm not native speaker too :).

closed 2020-08-16 by emelianov