emelianov/modbus-esp8266 #57: Problem reading multiple non-sequential addresses RTU

maiquimascarello, 2020-08-06:
Hello, first congratulations on the work!
I am communicating an ESP8266 (NodeMCU) via RTU with a device that is running on RS485 or RS232.
When I send  read multiple regs in a single request sequential everything goes perfectly, as in the code:

`uint16_t res[100];

void loop() {
  if (!mb.slave()) {
    mb.readHreg(10, 50, &res[0], 100, cbWrite);
    for (int i = 0; i < dados; i++) {
      Serial.println(res[i]);
      res[i] = 0;
    }
  }
    mb.task();
    yield();
}

But, I have problems when I want to read at specific addresses, how can I read single regs?
Here is an example, I want to read the regs at addresses 50, 100 and 300 of the same slave

uint16_t res[1];
uint16_t esc[1];
uint16_t tes[1];

 void loop() {
    if (!mb.slave()) {
      mb.readHreg(10, 50, &res[0], 1, cbWrite);
      Serial.println(res[0]);
      mb.readHreg(10, 100, &esc[0], 1, cbWrite);
      Serial.println(esc[0]);
      mb.readHreg(10, 300, &tes[0], 1, cbWrite);
      Serial.println(tes[0]);
    }
    mb.task();
    yield();
}`

Thank you very much!

emelianov, 2020-08-06:
Hello,
New request can't be sent unlit got responce (or timeout) for pervious one. That is because of ModbusRTU nature. The library itself doesn't performing any request queueing for RTU.
So you need to implement something like
```c
uint8_t cReg = 0;
void loop() {
  if (!mb.slave()) {
    switch(cReg++) {
    case 0:
      mb.readHreg(10, 50, &res[0], 1, cbWrite);
      Serial.println(res[0]);
    break;
    case 1:
      mb.readHreg(10, 100, &esc[0], 1, cbWrite);
      Serial.println(esc[0]);
    break;
    case 3:
      mb.readHreg(10, 300, &tes[0], 1, cbWrite);
      Serial.println(tes[0]);
    break;
    default:
       cReg = 0;
    }
  }
  mb.task();
  yield();
}
```

maiquimascarello, 2020-08-06:
Thank you very much, I'm going to study a better way : D 
Is this also for writing? 
Could you show a simple example of writing?

emelianov, 2020-08-06:
For write code is preaty the same.
(Ops! Fixed mistipe in code above)

maiquimascarello, 2020-08-06:
Thank you very much!

closed 2020-08-06 by maiquimascarello