emelianov/modbus-esp8266 #58: Question about multi register reading. ( MODBUS TCP )

zekageri, 2020-08-13:
### Hi! I have a question about mb.readHreg.

**How can i read multiple registers?
Do i have to declare somewhere the register addresses?**

### If i try this way:

```
#define Input_Reg    0x4000
#define OutPut_Reg    0x4032
uint16_t EXP_Input_Buffer[2];

bool SaveInput(Modbus::ResultCode event, uint16_t transactionId, void* data) {
    if(event == mb.EX_SUCCESS){
        Expander_Info[ADDRESS_COUNTER]->Inputs[0]     = EXP_Input_Buffer[0];
        Expander_Info[ADDRESS_COUNTER]->OutPuts[0]  = EXP_Input_Buffer[1];
        return true;
    }else{
        return false;
    }
}

static const inline void Read_Info_Registers(){
    if (!mb.slave()) {
        if(Expander_Info[ADDRESS_COUNTER] != 0){
            mb.readHreg(Expander_Info[ADDRESS_COUNTER]->Address,Input_Reg,EXP_Input_Buffer,2,SaveInput);
            while (mb.slave()){mb.task();}
        }
        ADDRESS_COUNTER++;
        if(ADDRESS_COUNTER == MAX_EXPANDER){ADDRESS_COUNTER = 0;}
    }
}
```

I don't get anything inside `Expander_Info[ADDRESS_COUNTER]->OutPuts[0]`. :(

emelianov, 2020-08-14:
As far as i can see code you provided reads two sequental Holding registers Input_Reg (0x4000) and Input_Reg+1 (0x4001).
So Expander_Info[ADDRESS_COUNTER]->OutPuts[0] will filled with value of Hreg 0x4001 whatever it is.

If you want to get mentioned in you code OutPut_Reg (0x4032) simpliest way is something like
```c
static const inline void Read_Info_Registers(){
    if (!mb.slave()) {
        if(Expander_Info[ADDRESS_COUNTER] != 0){
            mb.readHreg(Expander_Info[ADDRESS_COUNTER]->Address,Input_Reg,&EXP_Input_Buffer[0],1,SaveInput);
            while (mb.slave()){ mb.task(); yield(); }
            mb.readHreg(Expander_Info[ADDRESS_COUNTER]->Address,OutPut_Reg,&EXP_Input_Buffer[1],1,SaveInput);
            while (mb.slave()){ mb.task(); yield(); }
        }
        ADDRESS_COUNTER++;
        if(ADDRESS_COUNTER == MAX_EXPANDER){ADDRESS_COUNTER = 0;}
    }
}
```

zekageri, 2020-08-14:
> As far as i can see code you provided reads two sequental Holding registers Input_Reg (0x4000) and Input_Reg+1 (0x4001).
> So Expander_Info[ADDRESS_COUNTER]->OutPuts[0] will filled with value of Hreg [...]

**Thank you for your example. The thing is that i tried this method. It works but not well enough. The reading becomes slow. I need the values as fast as possible. The slave device holding its Inputs in the reg 0x4000 and its outputs in reg 0x4032.** 

### No way that i can read these two registers in one call?

emelianov, 2020-08-14:
Unfortunatly for ModbusRTU is possible to combine only multiple sequental registers of the same type to one query. You may try to request 32 registrs startting from 0x4000 (certanly result and speed will be dependent on target slave).
Also to sppeup data exchange it's possible to increase baudrate (if this is supported by your target slave device).

zekageri, 2020-08-14:
> Unfortunatly for ModbusRTU is possible to combine only multiple sequental registers of the same type to one query. You may try to request 32 registrs startting from 0x4000 (certanly result and speed [...]

**Thanks for the response. My slave devices has 115200 baud rate for now ( willing to increase in the future ) and max 10ms timeout. I need almost real time response now. The responses from the salves are near 5ms now if everything goes ok.
But i'am monitoring user interaction on these.** 

### Nevermind, thanks for your help i will figure out somethig!

closed 2020-08-14 by zekageri