emelianov/modbus-esp8266 #59: About MODBUS error code

lnsane-G, 2020-08-16:
Hello, sir. I need some help.
My device is esp32 and uses the 3.0.1 library.
In the actual test, the library reported an error: MODBUS result: E4
I really can't find the reason and ask for your help.
Here is a screenshot of my test code and software.

`
bool cb(Modbus::ResultCode event, uint16_t transactionId, void* data) { // Modbus Transaction callback
  if (event != Modbus::EX_SUCCESS)                  // If transaction got an error
    Serial.printf("Modbus result: %02X\n", event);  // Display Modbus error code
  // if (event == Modbus::EX_TIMEOUT) {    // If Transaction timeout took place
  //   mb.disconnect(remote);              // Close connection to slave and
  //   mb.dropTransactions();              // Cancel all waiting transactions
  // }
  return true;
}

void setup() {
  Serial.begin(9600);
  WiFi.begin("BYJD","BYJD123456");
  
  while (WiFi.status() != WL_CONNECTED) {
    delay(500);
    Serial.print(".");
  }
  Serial.println("");
  Serial.println("WiFi connected");  
  Serial.println("IP address: ");
  Serial.println(WiFi.localIP());
  mb.client();
}

uint16_t v[1]={true};
void loop() {
    if (!mb.isConnected(remote)) {   // Check if connection to Modbus Slave is established
        mb.connect(remote);           // Try to connect if no connection
        Serial.print(".");
    }
    mb.readHreg(remote,0,v,1,cb,1);
    //Serial1.println(v[0]);
    //mb.readCoil(remote,0,v,1,cb,1);
    mb.task(); // Modbus task
    delay(1000); // Pushing interval
}
`
[image]

lnsane-G, 2020-08-16:
[image]

emelianov, 2020-08-16:
Hi,
For the first decrease `delay(1000);` to say `delay(500);`

lnsane-G, 2020-08-16:
I don't seem to understand

lnsane-G, 2020-08-16:
> 嗨，
> 第一次减少`delay(1000);`说`delay(500);`

I don't seem to understand

emelianov, 2020-08-16:
Replace `delay(1000);' with `delay(500);` in your code.
```c
void loop() {
if (!mb.isConnected(remote)) { // Check if connection to Modbus Slave is established
mb.connect(remote); // Try to connect if no connection
Serial.print(".");
}
mb.readHreg(remote,0,v,1,cb,1);
//Serial1.println(v[0]);
//mb.readCoil(remote,0,v,1,cb,1);
mb.task(); // Modbus task
delay(500); // Pushing interval
}
```

lnsane-G, 2020-08-16:
> Replace`delay(1000);`with`delay(500);`in your code.
>
> ```c
> void loop() {
> if (!mb.isConnected(remote)) { //Check if connection to Modbus Slave is established
> mb.connect(remote); //Try to connect if no [...]
Thank you, but the result is the same
[image]

emelianov, 2020-08-16:
E4 is Timeout error. For TCP possible reason is communication error or firewall blocking.

lnsane-G, 2020-08-16:
> E4 is Timeout error. For TCP possible reason is communication error or firewall blocking.
I changed the router and it still didn't work. But Android Modbus TCP debugging tool can be successfully accessed.
[image]

emelianov, 2020-08-16:
It was my fault. Fixed in [3.0.2](https://github.com/emelianov/modbus-esp8266/releases/tag/3.0.2)

lnsane-G, 2020-08-16:
> It was my fault. Fixed in [3.0.2](https://github.com/emelianov/modbus-esp8266/releases/tag/3.0.2)

Thank you very much. I'll test it!

lnsane-G, 2020-08-16:
Test results are successful, thank you!
[image]

closed 2020-08-17 by emelianov