emelianov/modbus-esp8266 #62: Modbus RTU - Serial Direction Pin

gonzabrusco, 2020-09-04:
Hello! First I want to congratulate you on such a great work you are doing. 

Second I would like to make a request. When you are using Modbus RTU most of the times you use RS-485. And for using RS485 you need a transceiver. 

The thing is, the transceivers need an extra line that tells them the dirección of the communication. When you are "hearing" for new data, the transceivers output must be in high-Z mode. So you must set this pin accordingly. When you want to send data, you must change the state of this pin. In the IC I'm using is HIGH to read, and LOW to write. But lots of other ICs use inverted logic. 

What I would like you to implement is:

Add a method to tell MODBUS RTU that for writing or reading it must change the status of a pin. The pin number must be given on calling this method. Also, on calling this method you should also specify the logic (0= RECEIVING 1=SENDING or 1= RECEIVING 0=SENDING). You could make some define and specify them as NORMAL_LOGIC and INVERTED_LOGIC? 

Finally, some transceivers have automatic direction recognition. So for those transceivers this pin is not needed. So if you don't call this method when creating de modbus rtu object should be Ok too. 

That's all my friend. This would make your library really useful when using modbus RTU.

gonzabrusco, 2020-09-04:
For example: https://www.ti.com/product/ISO1430

The DE and RE pins normally are connected together and this way you control the direction of the communication with just one pin.

gonzabrusco, 2020-09-04:
SORRY. I just saw you already implemented this.

bool begin(HardwareSerial* port, int16_t txPin=-1);

Could you add a way of changing the logic? Somethins the txPin is inverted logic (like in this IC TD331S485H-E) and you need to write 0 on txPin to send data and 1 to receive.

emelianov, 2020-09-04:
Hello,
It makes sense. 
```c
bool begin(SoftwareSerial* port, int16_t txPin=-1, bool direct=true);
bool begin(HardwareSerial* port, int16_t txPin=-1, bool direct=true);
```
Changes have pushed to master branch. For inverse logic use
```c
mb.begin(&S, TX_PIN, false);
```

gonzabrusco, 2020-09-04:
Wow you are FAST. Thanks! I will try it during the weeking. Thanks again!!

gonzabrusco, 2020-09-13:
Ok. I tried modbusRTU as slave and as a master, using an inverted logic RS485 tranceiver. Everything works as expected. Thanks! Closing this now...

closed 2020-09-13 by gonzabrusco

fglaria, 2021-08-11:
> SORRY. I just saw you already implemented this.
> 
> ```
> bool begin(HardwareSerial* port, int16_t txPin=-1);
> ```
> 
> Could you add a way of changing the logic? Somethins the txPin is inverted logic (like in this IC TD331S485H-E) and you need to write 0 on txPin to send data and 1 to receive.

Is there a way to add this to `begin` call with Stream?

```c
bool ModbusRTU::begin(Stream* port) {
    _port = port;
    _t = 2;
    return true;
}
```