emelianov/modbus-esp8266 #63: modbus signal randomly go down to become zero

methaqali, 2020-09-22:
Hi @emelianov, 

I have problem with modbus signal randomly go down to become zero and return back to normal value, this issue continuously happen.

Thank in advance for your help.

emelianov, 2020-09-28:
Hello,
What do you mean under "modbus signal randomly go down to become zero and return back to normal value"
Could you explain? Is that about RX/TX control or something else? Probably you can provide some related code example?

methaqali, 2020-09-28:
I have using ESP8266 and connected by modbas t PLC use point to point(p-p) protocol

emelianov, 2020-09-29:
That's no enoght information to say something special. Can you share ESP code?

methaqali, 2020-09-29:
/*
Modbus-Arduino Example - Test Holding Register (Modbus IP ESP8266)
Configure Holding Register (offset 100) with initial value 0xABCD
You can get or set this holding register
Original library
Copyright by André Sarmento Barbosa
http://github.com/andresarmento/modbus-arduino

Current version
(c)2017 Alexander Emelianov (a.m.emelianov@gmail.com)
https://github.com/emelianov/modbus-esp8266
*/

/* Its suceeded to red distance in serial port and read value in plc but dont forget to set the Modbus address to the address of test regester if test regester equal to 100 in Hexa =64 
then sum 64 with 40000 = 40064*/ 
#ifdef ESP8266
#include <ESP8266WiFi.h>
#else //ESP32
#include <WiFi.h>
#endif
#include <ModbusIP_ESP8266.h>
const int trigPin = 5;    // pin no. 5 is D1 ESP8266
const int echoPin = 4;    // pin no. 4 is D2 ESP8266
int dist;
int dist2;
long duration;
// Modbus Registers Offsets
const int TEST_HREG = 100;

//ModbusIP object
ModbusIP mb;
IPAddress staticIP224_60(192,168,2,90);
IPAddress gateway224_60(192,168,2,2);
IPAddress subnet224_60(255,255,255,0);
void setup() {
#ifdef ESP8266
Serial.begin(9600);
#else
Serial.begin(115200);
#endif
pinMode(trigPin, OUTPUT); 
pinMode(echoPin, INPUT);
WiFi.config(staticIP224_60, gateway224_60, subnet224_60);
WiFi.begin("xxxxx", "xxxxxx");
while (WiFi.status() != WL_CONNECTED) {
delay(500);
Serial.print(".");
}

Serial.println("");
Serial.println("WiFi connected");  
Serial.println("IP address: ");
Serial.println(WiFi.localIP());

mb.slave();
mb.addHreg(TEST_HREG, 0x64);

}

void loop() {
 //Call once inside loop() - all magic here
 mb.task();
  // Clears the trigPin
  digitalWrite(trigPin, LOW);
  delayMicroseconds(2);
  // Sets the trigPin on HIGH state for 10 micro seconds
  digitalWrite(trigPin, HIGH);
  delayMicroseconds(10);
  digitalWrite(trigPin, LOW);
  // Reads the echoPin, returns the sound wave travel time in microseconds
  duration = pulseIn(echoPin, HIGH);
  // Calculating the distance
  dist= duration*0.34/2;
    // Make calibration to sensor reading//

    if ((dist>100) && (dist <289))  {
      dist2=(365-dist);}
      else
    if ((dist>=289) && (dist <=360))  {
      dist2=(360-dist);}

  // Prints the distance on the Serial Monitor
  Serial.print("Distance: ");
  Serial.println(dist2);
  mb.Hreg(TEST_HREG, dist2); // :-)
  delay(20); // Should decrease loop delay for reliable network communication
}

emelianov, 2020-09-30:
I think you should examine distance management routine result. Havn't tested such kind of sensors with ESP8266 but can imagine that while executing
```c
digitalWrite(trigPin, LOW);
// Reads the echoPin, returns the sound wave travel time in microseconds
duration = pulseIn(echoPin, HIGH);
```
some kind of interrupt can take place that will lead to unpredictible `dist` value.
Also if `dist` is 360 TEST_HREG value will be `0`.

methaqali, 2020-09-30:
I check output of HC-SR04 on IDE serial monitor every thing is normal.

methaqali, 2020-09-30:
When check the output of HC-SR04 in IDE serial port every thing is ok.

emelianov, 2020-10-01:
Add callback to monitor how client is requesting data from th modbus server. Probably that will help to understand erratic data flow.
```c
uint16_t cb(TRegister* reg, uint16_t val) {
  Serial.printf("Actual value sent: %d\n", val);
  return val;
}

void setup() {
...
mb.onGetHreg(TEST_HREG, cb);
...
}
```

methaqali, 2020-10-03:
Now I am able to write register to PLC , how can I use above sketch to read register from PLC?

emelianov, 2020-10-04:
If you mean read from remote Modbus server by ESP device, easiest way is 
```c
uint16_t val = 0;
uint16_t tr = mb.readHreg(ip, REMOTE_HREG, &val);
while (mb.isTransaction(tr)) mb.task();
Serial.println(val);
```

closed 2020-10-14 by emelianov