emelianov/modbus-esp8266 #64: FEATURE REQUEST: Add a possibility of having a callback for complete write and read operations in slave mode.

gonzabrusco, 2020-09-27:
Hello. I see that your library provides callbacks on register basis (that is, the callback function is called for each address written or read) but there's no callback for the complete write or read operation.

For example, if my ESP32 is configured as slave and I have the holding registers 0,1,2,3 and the master wants to write all four registers on one operation, it can. But the callback function will be called four times (for address 0,1,2 and 3).

It there a possibility to have a single callback that is called before the read operation (but only once, no matter how many registers are being read) and after the write operation (again but only once). That is, if the master wants to read 0,1,2,3 in one single operation, the callback should be called once BEFORE sending the answer back (maybe the callback could inform the user the addresses that are being read). And if the master wants to write 0,1,2,3 in one single operation, the callback should be called once AFTER the operation so the software can take the neccesary actions  (maybe the callback could inform the user the addresses that were written).

Basically it would be more or less the same as you already have, but instead of being called lots of times, I want just one callback call informing the complete ranges of addresses. This would be more useful because sometimes addresses are related to one each other and it does not make sense to have a separate callback for each of them.

Thanks!
Gonzalo

emelianov, 2020-09-28:
Hello,
I'm not sure that it's good idea to duplicate onSet/onGet callbacks by adding another agregated callback that is called after registers have beed updated. But i think it's reasonable to add callback before actual read/write operation on registers to implement some way of access control. Callback function will be
```c
bool cb(Modbus::FunctionCode, TRegister reg, uint16_t regCount) {
// return true if operation is allowed and false otherwise.
}
```
That do you think?

gonzabrusco, 2020-09-28:
What I really need is the library to give me the oportunity of updating the holding registers before sending the answer back. But I would like that the library calls me only one time per operation, and not per register address. That way I can update the whole memory map before the read operations.
The same goes to writing operations but AFTER the writing operation is done. That way I can check if I need to perform some actions based on the registers written. But this call has to be ONE per writing operation, not several calls (one per address written).

The callback you are proposing is OK but only for reading operations. Maybe you could add a second callback. One that calls you back before ANY operation (specified by Modbus::FunctionCode, TRegister reg, uint16_t regCount) and one that calls you back after ANY operation (again specified by Modbus::FunctionCode, TRegister reg, uint16_t regCount and Execution-Result?) 

Thanks for the support you are giving to this library.!!

gonzabrusco, 2020-09-28:
Just to sum up the idea: The address based callbacks are a nice feature, but I think it would be useful to have something more general too. 
One callback before any operation (specifiying the type of operation, register address and  register count) and another callback after any successful operation (specifiying the type of operation, register address and register count)

emelianov, 2020-09-29:
Let me day or so to add the functionality to upcoming [4.0.0](https://github.com/emelianov/modbus-esp8266/tree/4.0.0).
Pre-operation callback (access control functionality) will be generally available.
Pos-operation callback will also be implemented but i prefere to not include it to public API (at least by now).

gonzabrusco, 2020-09-29:
Thank you! Pos-operation callback will be useful. 

For example, you have a device with some calibration parameters and the user has to change them all at the same time and not one by one. So the user has to perfom a multiple holding register write.  If the callback is called once per each register, it gets difficult (from the program point of view) to know when those paremeters are FULLY changed (and not just one random register). With a callback after the operation telling me the register address and the amount of registers changed, I can be completely sure that the user finished writing those registers (and wrote them all). In this particular situation, having just one callback is more usefull because a lot of times registers are correlated to one each other.

emelianov, 2020-10-01:
https://github.com/emelianov/modbus-esp8266/tree/4.0.0/examples/Callback

gonzabrusco, 2020-10-01:
Wow thanks Alexander! Is it ready for testing?

You are making a lot of changes in this new version, when do you think it will be ready for production?

gonzabrusco, 2020-10-01:
I'm testing the library now. Take a look at this compiler errors (not related to this):

    In file included from .pio\libdeps\esp32dev\modbus-esp8266\src/ModbusTCP.h:17:0, from src\main.cpp:15:
    .pio\libdeps\esp32dev\modbus-esp8266\src/ModbusTCPTemplate.h:32:2: warning: 'typedef' was ignored in this declaration
     };
      ^
    .pio\libdeps\esp32dev\modbus-esp8266\src/ModbusTCPTemplate.h:45:2: warning: 'typedef' was ignored in this declaration
      };
      ^
It seems you defined two typedefs but you miss to write the type name.

gonzabrusco, 2020-10-01:
I tried to test version 4.0.0 but I have a problem with modbusRTU. The pin direction argument is missing in the definition of the method. 

```
In file included from .pio\libdeps\esp32dev\modbus-esp8266\src/ModbusAPI.h:8:0,
                 from .pio\libdeps\esp32dev\modbus-esp8266\src/ModbusRTU.h:13,
                 from src\main.cpp:10:
.pio\libdeps\esp32dev\modbus-esp8266\src/Modbus.h:27:0: note: this is the location of the previous definition
 #define HREG(n) (TAddress){TAddress::HREG, n}
 ^
src\main.cpp: In function 'void setup()':
src\main.cpp:42:39: error: no matching function for call to 'ModbusRTU::begin(HardwareSerial*, int, bool)'
   mb.begin(&Serial2, MODBUS_DIR, false);
                                       ^
In file included from src\main.cpp:10:0:
.pio\libdeps\esp32dev\modbus-esp8266\src/ModbusRTU.h:47:8: note: candidate: bool ModbusRTUTemplate::begin(HardwareSerial*, int16_t)
   bool begin(HardwareSerial* port, int16_t txPin=-1);
        ^
.pio\libdeps\esp32dev\modbus-esp8266\src/ModbusRTU.h:47:8: note:   candidate expects 2 arguments, 3 provided
.pio\libdeps\esp32dev\modbus-esp8266\src/ModbusRTU.h:48:8: note: candidate: bool ModbusRTUTemplate::begin(Stream*)
   bool begin(Stream* port);
        ^
.pio\libdeps\esp32dev\modbus-esp8266\src/ModbusRTU.h:48:8: note:   candidate expects 1 argument, 3 provided
*** [.pio\build\esp32dev\src\main.cpp.o] Error 1
```
I really don't fully understand what you are doing, but now PlatformIO is taking ModbusRTUTemplate as the class of ModbusRTU and there the only methods for begin are:

`bool begin(HardwareSerial* port, int16_t txPin=-1);
		bool begin(Stream* port);`

While it should be 

`bool begin(HardwareSerial* port, int16_t txPin=-1, bool direct=true);`

emelianov, 2020-10-02:
> 'typedef' was ignored in this declaration

It's allowable c++ struct definition so i suggest just ignore warning.

> While it should be
> `bool begin(HardwareSerial* port, int16_t txPin=-1, bool direct=true);`

Ops! Just havn't incorporated changes from last master branch updates. Fixed.

> You are making a lot of changes in this new version, when do you think it will be ready for production?

Core of Modbus processing for ESP8266/ESP32 havn't changed too much. Most changes are related to create wrappers for easer library expansion and to eliminate code duplication.
So for ESP8266/ESP32 new build should be already stable. No strict timeframe defined for the production release.

gonzabrusco, 2020-10-03:
Thank you. I will test now and report back. In the meanwhile, the compiler is giving me some warnings because you are using void* in arithmetic operations. Shouldn't you cast the object before making an arithmetic operation ?

[image]

gonzabrusco, 2020-10-03:
Also, looking at the source: Shouldn't "_onRequestSuccess" return void? Why do you need to return "Modbus::ResultCode" ?

gonzabrusco, 2020-10-03:
I'm happy to tell you that the callbacks worked perfectly. I tested your example on my setup and it worked as expected. Thanks!

emelianov, 2020-10-04:
> Shouldn't you cast the object before making an arithmetic operation ?

That is C-style semantic. It looks ok to me as it's used to walk through buffer that can contain different datatypes. So direct address increment is performed on void pointer and data cast is taking place while data is processing:
```c
*((uint16_t*)output) = __bswap_16(*((uint16_t*)frame));
```

> Shouldn't "_onRequestSuccess" return void?

`_onRequestSuccess` is declared to be the same as _onRequest just to decrease variety API. Currently return value of _onRequestSuccess is not used.

> Thanks!

You are welcome!

|If the library is helpful for your projects you can support it by a glass of beer|[image]|
|---|---|

(It's not mandatory for sure)

gonzabrusco, 2020-10-04:
> That is C-style semantic. It looks ok to me as it's used to walk through buffer that can contain different datatypes. So direct address increment is performed on void pointer and data cast is taking [...]

Yes but you cannot perform arithmetic on a void pointer because pointer arithmetic is defined in terms of the size of the pointed-to object.

To omit the warning you should insted use: 

`output = (uint16_t*)output + 1;` or `output = (uint8_t*)output + 2;`

> > Shouldn't "_onRequestSuccess" return void?
> 
> `_onRequestSuccess` is declared to be the same as _onRequest just to decrease variety API. Currently return value of _onRequestSuccess is not used.

Understood.

> > Thanks!
> 
> You are welcome!
> 
> If the library is helpful for your projects you can support it by a glass of beer	[image]
> (It's not mandatory for sure)

I will later. You bet on it. 

> > 'typedef' was ignored in this declaration
> It's allowable c++ struct definition so i suggest just ignore warning.

About this, I know it's allowed but just doing this would make the warning disappear:

```
typedef struct TTransaction {
	uint16_t	transactionId;
	uint32_t	timestamp;
	cbTransaction cb = nullptr;
	uint8_t*	_frame = nullptr;
	void*		data = nullptr;
	TAddress	startreg;
	Modbus::ResultCode forcedEvent = Modbus::EX_SUCCESS;	// EX_SUCCESS means no forced event here. Forced EX_SUCCESS is not possible.
	bool operator ==(const TTransaction &obj) const {
		    return transactionId == obj.transactionId;
	}
} TTransaction;  <--------- Repeat struct name
```

And

```
typedef union MBAP_t {
		struct {
			uint16_t transactionId;
			uint16_t protocolId;
			uint16_t length;
			uint8_t	 unitId;
		};
		uint8_t  raw[7];
} MBAP_t ;   <--------- Repeat struct name
```

I think it's a better practice.

emelianov, 2020-10-05:
> Yes but you cannot perform arithmetic on a void pointer because pointer arithmetic is defined in terms of the size of the pointed-to object.
> 
> To omit the warning you should insted use:
> 
> `output = (uint16_t*)output + 1;` or `output = (uint8_t*)output + 2;`

If this piece of code should be refactored i prefere to make it clear (This will require to rewrite __bswap_16 but who cares ;) ):

```c
uint16_t* from = (uint16_t*)(frame + 2);
uint16_t* to = (uint16_t*)output;
while(field2--) {
   *(to++) = __bswap_16(*(from++));
```

> > > 'typedef' was ignored in this declaration
> > > It's allowable c++ struct definition so i suggest just ignore warning.
> 
> About this, I know it's allowed but just doing this would make the warning disappear:

It's reasonable. I'll review all structs defenitions.

gonzabrusco, 2020-10-05:
Thanks. I'll close this now!

gonzabrusco, 2020-10-25:
Hello Alexander. I was writing some code using your amazing library and I began to wonder... these preRequest and postRequest callbacks are not global like the other callbacks, right? They are built inside the class as protected members. Correct me if I'm wrong.

Also, when adding or modifying local registers, it would be nice to have a generic way of accessing them, without using an specific modbusIP or modbusRTU object. As the registers are global, it feels not right to access them using one specific object. But programmatically I don't know who to do this. Just an idea.

Thanks
Gonzalo

gonzabrusco, 2020-10-25:
Sorry to bother you again. I'm looking to the library in detail. When using global regs, you define this arrays of regs and callbacks. But what I'm seeing is that when you detroy the object, that memory does not get freed. You have a possible memory leak there. The problem gets more difficult to solve if you have two objects (modbusIP and modbusRTU) and you destroy only one of them, then the registers should remain in place. But if you destroy both of them, then de memory should be freed. I think this leads to the conclusion that maybe the library needs a bit of restructuring. 

`#ifdef MB_GLOBAL_REGS
 std::vector<TRegister> _regs;
 std::vector<TCallback> _callbacks;
#endif`

emelianov, 2020-10-26:
Exactly. preRequest and postRequest are defined for each instance.

As you write in anothe issue "control" Modbus instance is suitable solution for generalizing access to global registers.

closed 2020-10-26 by gonzabrusco

gonzabrusco, 2020-11-02:
Hello Alexander, Sorry to bother with this again, is there a chance you could backport this callbacks to current version? Thanks
Gonzalo

emelianov, 2020-11-03:
Hello Gonzalo,
I'm afraid no backport is planned as i'm going to witch master to 4.0 soon.