emelianov/modbus-esp8266 #65: control transmission/reception function on MAX485 chip

Bach-Danielsson, 2020-09-28:
Is ther any plans to implement the TX/RX function in this libirary as described bellow?

From the repo mentioned as base https://github.com/andresarmento/modbus-arduino, you can read this:

Create the mb instance (ModbusSerial) to be used.

mb.config (&Serial, 38400, SERIAL_8N1);
mb.setSlaveId (10);
Sets the serial port and the slave Id. Note that the serial port is passed as reference, which permits the use of other serial ports in other Arduino models. The bitrate and byte format (8N1) is being set. If you are using RS-485 the configuration of another pin to control transmission/reception is required. This is done as follows:

mb.config (& Serial, 38400, SERIAL_8N1, 2);
In this case, the pin 2 will be used to control TX/RX.

emelianov, 2020-09-28:
Hello,
That've already implemented:
```c
bool begin(SoftwareSerial* port, int16_t txPin=-1, bool direct=true); // For ESP8266 only
bool begin(HardwareSerial* port, int16_t txPin=-1, bool direct=true);
```
 Assing Serial port. txPin controls transmit enable for MAX-485. Pass direct=false if txPin uses inverse logic.
```

Bach-Danielsson, 2020-09-28:
Thank you for a promt answer.  
Im looking at you example , examples/RTU-slave/RTU-slave.ino.  
I´m using Visual code and platformIO as development envormental.  That means that i rename to .CPP (C++) i expect that this ha no influence for me.
I have problem to see witch GPIO on my ESP32 to control the MAX485 TX/RX function.

emelianov, 2020-09-28:
For ESP32 it should be something like
==
Serial2.begin(9600, SERIAL_8N1, RXD2, TXD2);
mb.begin(&Serial2, DE_RE);
mb.slave(SLAVE_ID);
==
DE_RE is direction (RX/TX) control pin
RXD2, TXD2 and DE_RE can be assigned to pins you wish.

Bach-Danielsson, 2020-09-29:
In your example RTU-Slave you define slave id to 1  #define SLAVE_ID 1, 
in your example RTU-Master, where do you define the slave ID to call ?  If i´m correct informed this two shall match ?

closed 2020-10-12 by emelianov