emelianov/modbus-esp8266 #66: MODBUS RTU 0xE4

TelleRen2k, 2020-09-28:
Hi,
I have a sketch (master) running accessing a slave (RTU) using your fine modbus library.

All readHreg and writeHreg work fine, but when I do a readIreg I always get the 0xE4 error back.

```
  if (!mb.slave()) { 
    mb.readIreg(SLAVE_ID, FanPowerRegister, &FanReadPower, 1, cbRead);
    delay(100);
    Serial.print(FanReadPower);
  }
```
Any idea why this could happen?

emelianov, 2020-09-28:
Hello,
Could you tell if both smaster and slave are driven by the library or just master?

TelleRen2k, 2020-09-28:
Hi, It's just the master. The slave is a high end motor

emelianov, 2020-09-28:
Have no ireas at this piont. let's try to debug. Please uncomment
```c
//#define MODBUSRTU_DEBUG
```
in `ModbusRTU.h` and collect Serial output processed during `readIreg()` call.

TelleRen2k, 2020-09-28:
I enabled debugging but generally speaking that did nothing much? 
When querying the slave (readIreg) I do not see any specific byte sequence (debugging info). Only once I saw the string "00 00 C8", most other reads did not produce any debugging info. Another time I saw "F7", but that was only 2 times in maybe 100 calls to readIreg.

In setup I query the motor for its ID (readHreg) and I get a nice debug string: "03 0C 33 4B 32 47 30 35 52 52 31 30 32 48 D8 DB"

But afterwards I do not see additional debug info - not even if I do a writeHreg.

emelianov, 2020-09-28:
Debug output should contain the only response data from slave.
Basing on fact that you are getting absolutely impossible for normal flow data bits i suspect that it some kind of hardware problems. I can imagine for example that on writing Ireg motor state is changed that lead to electrical noise that affecting data line.
So try to reduce communication speed if possible. Isolate power sources of ESP, motors, motor's modbus controller as much as possible. Play with connecting or disconnecting ground wire between ESP and motor.

TelleRen2k, 2020-09-30:
I have copied the MODBUS RTU code from my project and made a test sketch. Here everything works fine. 

The only thing happening in the main loop is this:

```

void loop() {
  delay(1100);
  readFanPower();
  delay(1100);
  mb.task();  
}

void readFanPower() {
  if (!mb.slave()) { 
    mb.readIreg(SLAVE_ID, FanPowerRegister, &FanReadPower, 1, NULL);
    delay(100);
    Serial.print("MODBUS Read Fan power: ");
    Serial.print(FanReadPower);
    Serial.println(" ");
  }
  mb.task();
}
```
`
Everything works fine. I just realised I could probably remove the call to mb.task() in readFanPower?

When the exact same function is run in my other sketch it fails with the 0xE4 error.

The code is as follows (to give you an indication of what happens in the main loop):

```
void pollFanPower(void) {
  if ((millis() - lastPollFanTime) > POLLINTERVAL_FAN) {
    readFanPower();
    lastPollFanTime  = millis();
  }
}

void loop() {
  checkPushButtons();  
  pollWiFiConnection();
  if (displayStatus == ON) {
    checkSwitchOffDisplay();
  }
    // Poll Fan Power Consumption
    if (ENABLE_FAN_COMMUNICATION == 1) {
      pollFanPower();
      mb.task();  
    }

  if(PushButtonLevel != 0) {   
    if (ENABLE_LOG_DATA_TO_MQTT_SERVER == 1) {
      checkMQTTConnection();
    }
    pollSensors();
  } 
}

```
But this is not working?

emelianov, 2020-09-30:
As the library executess calls async way (that is mb.readIreg(...) just sends read request to slave\server but responce processed by mb.task()) mb.task() should be called at leas once (multiple times is way better) per MODBUS*_TIMEOUT (that is one second by default). At least first fragment
```c
void loop() {
  delay(1100);
  readFanPower();
  delay(1100);
  mb.task();  
}
```
may lead to timeout error.
If you need to exec ModbusRTU request  sync way use
```c
mb.readIreg(...);
while (mb.slave()) mb.task();
```

TelleRen2k, 2020-09-30:
I use the "while (mb.slave()) mb.task();" method in setup() 

The code with the delay(1100); works fine. It's the other example code that does not work :-) 

There's nothing time consuming in the loop that fails (0xE4), but I just tried to make is synchronus (while (mb.slave()) mb.task();) and the result is still the same. All calls to readFanPower result in 0xE4

I find this very strange.

TelleRen2k, 2020-09-30:
OK, this is strange. I have removed everything from loop() in the sketch not working. Only thing in there now is:

```
      pollFanPower();
      mb.task();  
```
It's still producing 0xE4 errors.

TelleRen2k, 2020-09-30:
Allright, I think I found the problem - but not the solution :-)

I use a dust sensor and I use this library: https://github.com/luciansabo/GP2YDustSensor

It seems your code does not like this other library (or the other way around). I have no idea why, but some MODBUS RTU communication is not working after the code below is executed (the readIreg does not work, but it seems the writeHreg is working):

```
GP2YDustSensor dustSensor(GP2YDustSensorType::GP2Y1010AU0F, SHARP_LED_PIN, SHARP_VO_PIN, SAMPLES_TO_AVERAGE);
dustSensor.begin();

```

I assume if this was a hardware issue then all MODBUS RTU communication should be affected? But this is not the case as described above.

Do you know what can cause this?

emelianov, 2020-09-30:
Probably that's because of getDustDensity(uint16_t numSamples). What count of numSamples is user in uor code?
Each sample locks execution by ~10mS.

```c
uint16_t GP2YDustSensor::getDustDensity(uint16_t numSamples)
{
    uint32_t total = 0;
    uint16_t avgRaw;
  
    for (uint8_t i = 0; i < numSamples; i++) {
        total += this->readDustRawOnce();
        // Wait for remainder of the 10ms cycle = 10000 - 280 - 100 microseconds.
        delayMicroseconds(9620);
    }
...
```

TelleRen2k, 2020-09-30:
I call getDustDensity without any parameters

`value = dustSensor.getDustDensity();
`

And use: 

```
#define SAMPLES_TO_AVERAGE    20
GP2YDustSensor dustSensor(GP2YDustSensorType::GP2Y1010AU0F, SHARP_LED_PIN, SHARP_VO_PIN, SAMPLES_TO_AVERAGE);
```

emelianov, 2020-10-01:
Default .getDustDensity() samplles are 20. So it takes at ~200 ms. So i hope loop() have a chance to run multiple times per second. 
Try
1. Change Serial port used for Modbus. Probably even try to use SoftwareSerial
2. Test patch to the library
( replace ModbusRTU::rawSend() function in ModbusRTU.cpp with the following implementation)
```c
bool ModbusRTU::rawSend(uint8_t slaveId, uint8_t* frame, uint8_t len) {
    uint16_t newCrc = crc16(slaveId, frame, len);
    if (_txPin >= 0) {
        digitalWrite(_txPin, _direct?HIGH:LOW);
        delay(1);
    }
    #ifdef ESP32
    portENTER_CRITICAL(&mux);
    #endif
    _port->write(slaveId);  	//Send slaveId
    _port->write(frame, len); 	// Send PDU
    _port->write(newCrc >> 8);	//Send CRC
    _port->write(newCrc & 0xFF);//Send CRC
    _port->flush();
    if (_txPin >= 0)
        digitalWrite(_txPin, _direct?LOW:HIGH);
    #ifdef ESP32
    portEXIT_CRITICAL(&mux);
    #endif 
    return true;
}
```

TelleRen2k, 2020-10-02:
I got an error compiling your code.

```
error: '_direct' was not declared in this scope
         digitalWrite(_txPin, _direct?LOW:HIGH);
```
I changed the lines with: digitalWrite(_txPin, HIGH); You are probably working with a newer version than is available as an Arduino library.

Unfortunately it is still the same problem.

I'm not sure what you mean by: 
"Change Serial port used for Modbus. Probably even try to use SoftwareSerial"?

mb.master is using serial2 at the moment

Strangely writeHreg works most of the times, but readIreg never works

Thanks.

emelianov, 2020-10-02:
> I'm not sure what you mean by:
> "Change Serial port used for Modbus. Probably even try to use SoftwareSerial"?

Try to use Serial3. It make difference in some cases.

> Strangely writeHreg works most of the times, but readIreg never works

Probably it depends on amount of the data and interframe timings.

TelleRen2k, 2020-10-02:
I tried to change the code to:
```
  Serial3.begin(19200, SERIAL_8E1, RXD, TXD);
  mb.begin(&Serial3, RXTXEnablePin);
  mb.master();
```
But I get an error: "'Serial3' was not declared in this scope" 

I then added the following code:

```
#include <HardwareSerial.h>
HardwareSerial Serial3(2);
```

And changed my initialization to:

```
  Serial3.begin(19200, SERIAL_8E1, RXD, TXD);
  mb.begin(&Serial3, RXTXEnablePin);
  mb.master();
```
I'm not sure that is how to work with serial3, but the code compiles fine.

Unfortunately is is still the exact same issue.

Thank you for your support.

TelleRen2k, 2020-10-02:
I do not know the inner workings of your library - but I thought it was strange that the MUTEX did not solve the issue (assuming that would allow the modbus code to execute fully without being interrupted)?

I tried to lower the dust sensor "samples to average" from 20 to 10 but that has no effect.

emelianov, 2020-10-02:
MUTEX is about to protect data send stage.
How about to ivestigate hardware problems? Is that possible to replace RS-485 tranciever with another type one?

TelleRen2k, 2020-10-02:
I'm affraid it's not easy to change the RS485 interface.

Do you think it's a possibility that the RS485 board is faulty? I mean everything runs fine if the Dust Sensor code is disabled.

I tried to remove the sensors (they are attached using a cable to a the sensor PCB) but the problem is still the same.

emelianov, 2020-10-02:
They said that some RS-485 tranceivers with auto RX/TX switching can be affected by powerline noise. Also hardware problems are reported from time to time: [for example](https://github.com/emelianov/modbus-esp8266/issues/44#issuecomment-640441039)
What RS-485 data rate is used? 9600 or faster?

TelleRen2k, 2020-10-02:
Data rate is 19200 (default value recommended by motor manufacturer). I will try to lower it.

emelianov, 2020-10-02:
If it will not help and as i understand the problem persist even with sensors are not connected it possible to perform simulation in my lab if you share complete code (not for public; by e-mail for example).

TelleRen2k, 2020-10-02:
Do you work with hardware and PCB design too?

emelianov, 2020-10-02:
No unfortunatly not.
I can just simulate PCL and ESP data exchange for specific sketch and debug software problems if any.

TelleRen2k, 2020-10-02:
I have sent you an email :-)

emelianov, 2020-10-04:
I've run code you provided in my environment.
Note that after enabling `pollDustSensor();` `mb.task();` executed only once every 5 seconds:
```log
11:41:22.688 -> Current Fan Power: 100 
11:41:22.688 -> Read result: 0x00
11:41:27.817 -> Current Fan Power: 100 
11:41:27.817 -> Read result: 0x00
11:41:32.907 -> Current Fan Power: 100 
11:41:32.907 -> Read result: 0x00
```
I suspect that the library version you are using not the latest one -- since 3.0.2 the library should be tolerant to rare .task() execution.

TelleRen2k, 2020-10-04:
In the README.md file I see this:

## Last Changes
```diff
// 3.0.2
+ ModbusTCP Client: ESP32 fix unexpected transaction timeout 
// 3.0.1
+ ModbusRTU: ESP32 possible send\receive failure fix
+ ModbusRTU: Non-ESP devices support
+ Restriction to registers count removed
+ Added bridge example
// 3.0.0
...
```
I assume this means I use version 3.02.

I have no idea where to go from here.

emelianov, 2020-10-05:
Finally i found no fatal errors of Modbus communication in your code. Without dust sensor connected it's working ok (have no such sensor in my lab)

> The fun thing is if pollDustSensor is not called (but only initDustSensor() in setup()) then it's still the same problem.

Note that `initDustSensor()` actually just sets pin mode.

```c
void GP2YDustSensor::begin()
{
    pinMode(this->ledOutputPin, OUTPUT);
}
```
Basing on the rest of the GP2Y library it also turns sensor led on.

Also note that GP2Y datascheet points that power socket of the sensor should contain 220uF capacitor:
[image]
I think there is a reason. Power consumption is labaled as 20mA but probably there are peaks that should be compensated by the capacitor.
In other words i beleave that there is hardware problem preventing stable Modbus communication.

TelleRen2k, 2020-10-05:
The capacitor as well as the resistor are implemented in the circuit board. 

Your observation that the initDustSensor() does only set a pin is interesting. Especially because I see the same problem if the sensor PCB is not connected to the ESP32/MODBUS PCB.

TelleRen2k, 2020-10-10:
FYI I have ordered some new PCBs because it seems the one I have now has a problem on the tracing near some of the ESP32 pins. Hopefully it is a hardware problem after all.

closed 2020-10-12 by TelleRen2k