emelianov/modbus-esp8266 #68: Fatal error

workpage2, 2020-09-29:
I am trying to use your library for my esp32 code.
When trying to connect, I get this error:
    assertion "heap != NULL && "free() target pointer is outside heap areas"" failed: file "/home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/heap/heap_caps.c", line 267, function: heap_caps_free
    abort() was called at PC 0x401512af on core 1

    Backtrace: 0x4008c7bc:0x3ffb1d40 0x4008c9ed:0x3ffb1d60 0x401512af:0x3ffb1d80 0x40084b57:0x3ffb1db0 0x400850b9:0x3ffb1dd0 0x4000bec7:0x3ffb1df0 0x400fca43:0x3ffb1e10 0x400d9bee:0x3ffb1e50 0x400fefb9:0x3ffb1fb0 0x40088d1d:0x3ffb1fd0

    Rebooting...
However, sometimes a connection occurs. When there is no connection, the program works and there is no error.
I made labels in the program. The error occurs when the
 mb.task () ;.
My program is large and complex. Uses webserver, mqtt, websocket, HTTPClient.h
p.s. IP-server-HoldingReg works fine.

emelianov, 2020-09-29:
On that limited information can only suspect that problem is about multitasking.
The Mobbus library is not thread-safe so if you calling modbus methods from different threads without external semaphore locking it will crash sooner or later.
Otherwise addintional info is required to identify the issue.

workpage2, 2020-09-29:
Александр, вы оказывается мой земляк!
Вобщем проблема была как в том анекдоте про лом и ботинки.
Что бы отправить данные в websocket, нужно преобразовать переменную и string в charArray. Для этого используется промежуточная переменная char data1[]. Я указал слишком маленькую длину. После увеличения размера data1 библиотека modbus падать перестала!
Спасибо за Ваш труд. Отличная библиотека!

    data_temp="╚23"+ntp_adr;
    data_temp.toCharArray(data1,(data_temp.length()+1));
    webSocket.broadcastTXT(data1,data_temp.length());

emelianov, 2020-09-30:
Мир тесен, Владимир ;).
Отдаю долг OpenSource'у, так сказать. + Даже иногда донаты прилетают, но не окупают время, конечно.
Glad the problem have resolved.

workpage2, 2020-09-30:
Куда донатить? Дайте ссылку или карту.

emelianov, 2020-09-30:
|If the library is helpful for your projects you can support it by a glass of beer|[image]|
|---|---|

Уже вставил ссылку, но вспомнил, что внутри России PayPal теперь person-to-person не переводит.
Можно через систему быстрых платежей на номер +7-912-922-7-666 (Альфа)

closed 2020-09-30 by workpage2