emelianov/modbus-esp8266 #69: Using only ModbusRTU on PlatformIO

TSO-tech, 2020-09-29:
Hi,

I'm using the lib on VSC with PlatformIO to create a simple sensor with modBus RTU (salve).

on file platformio.ini I have "lib_deps = melianov/modbus-esp8266 @ ^3.0.3"

on main.cpp (renamed RTU-slave.ino) I only reference is to #include <ModbusRTU.h> but I always get compiler error on file ModBusIP_ESP8266.cpp.

To solve the compiler error I have to delete ModBusIP_ESP8266.h and ModBusIP_ESP8266.cpp.
There is any change I have to do?

thanks

gonzabrusco, 2020-09-29:
You could try changing the way PlatformIO scans for libraries. 

    lib_ldf_mode = deep

More info: https://docs.platformio.org/en/latest/librarymanager/ldf.html

TSO-tech, 2020-09-30:
Hi @gonzabrusco,

Thank you for your reply. I'm new to PlatformIO and my programing skills are limited (I didn't program anything for years) So I have some difficulty to understand the full impact of the options. 

The initial suggestion didn't work, but using the lib_ldf_mode = deep+ worked. I'm assuming the main difference is taking in consideration C/C++ Preprocessor conditional syntax.

closed 2020-09-30 by TSO-tech