emelianov/modbus-esp8266 #70: Fractional number.

workpage2, 2020-09-30:
How do I send a fractional number? 
Как отправить дробное число? Подскажите пожалуйста.

emelianov, 2020-09-30:
[Thanks for donate ;)]
```c
float value32 =  1.00;
mb.writeHreg(ip, REG_NR, (*uint16_t)&value32, 2); // write 32-bit variable to two sequental registers
```
Modbus оперирует только16bit регистрами. Т.е. любые другие типы данных нужно упаковывать в несколько последовательных регистров. Это нормальная общепринятая практика.
Возможно, перед отправкой придется поменять порядок слов: `(value32>>16) | (value32<<16)` в зависимости от того, что ожидает приемник.

workpage2, 2020-09-30:
Я хочу реализовать такую конструкция.
    mb.Ireg(REG,value32);
Как это сделать?

emelianov, 2020-10-01:
Так прямо не сделать: float 32 бита, регистр Modbus 16 bit. Как сделать, во многом зависит от того, что будет на второй стороне и что она сможет понять/принять
1. Универсальный вариант - задействовать два регистра (старшую часть храним в доном, младшую в другом и на приемнике собираем вместе)
2. Если значение небольшое можно использовать целое 16bit для хранения дробного с фиксированной точкой. Т.е 12.3 хранить как 123.

workpage2, 2020-10-01:
Сейчас всё работает по второму варианту. Но хотелось бы привести к общепринятому стандарту.
Приёмники могут быть самые разные. 
Как в один пакет затолкать два числа? Это возможно?

emelianov, 2020-10-01:
Стандартное (в промышленных контроллерах вполне применятся) Но стандарт де-факто; не описанный в спецификации Modbus. И от вендора бывает по разному задан порядок слов и байт.

closed 2020-10-12 by emelianov