emelianov/modbus-esp8266 #71: change port tcp

l337quez, 2020-10-05:
thanks for shared. i want change port tcp client from code in file .ino.   I dont like edit your library

emelianov, 2020-10-05:
Hello.
```c
bool connect(IPAddress ip, uint16_t port = MODBUSIP_PORT);
```
That is
```c
mb.connect(remoteIP, 1555); // Connect to port 1555 instead of default 502
```

l337quez, 2020-10-05:
this var remoteIP, is IP for server?  or ip asigned esp8266

this is a constant ?  bool connect(IPAddress ip, uint16_t port = MODBUSIP_PORT);

l337quez, 2020-10-05:
I need to connect an esp with a raspberry. The Raspberry is as served and works. The esp8266 is not working for me as a slave .. I have always seen that both the ip and the server port are written in the slave .. in your library the server's ip is never written

emelianov, 2020-10-06:
If RPi is server and ESP is client use [client example](https://github.com/emelianov/modbus-esp8266/blob/master/examples/IP-client-SimpleRead/IP-client-SimpleRead.ino) and specify port in mb.connect:
```c
mb.connect(remote, 1555); // 1555 is remote Modbus server port
```

if ESP is server ad RPi is client use [server example](https://github.com/emelianov/modbus-esp8266/blob/master/examples/IP-server-Led/IP-server-Led.ino) and specify port in mb.server():
```
mb.server(15555); // local ESP Modbus port to listen for clients
```

l337quez, 2020-10-06:
I had already tried the example and nothing happened .. Today I sat down to see in detail and it does not enter this conditional

if (mb.isConnected (remote))

then it seems that it is not connecting. I am using a python library on the Raspberry (server) Pymodbus and when a client connects a message appears. But when I load the code of your example it does not connect and there is no error.

Just edit this line from your example, what did you tell me

mb.connect (remote, 5020);

do you have any idea what's going on. Everything is in the same local network, the ip of the esp is correct, it is in the dhcp range, but there is no connection

l337quez, 2020-10-06:
I was able to connect now .. I forgot to change the remote ip to the server ip. thank you very much for responding

matlesc, 2020-10-13:
Hello,
I'm facing the same issue while attempting connecting ESP8266 as a client to RPI running a Modbus  (pyModbusTCP) server, port 502 by default. The server is OK while there are another python clients. The RPI is configured as an access point. ESP8266 is connecting  the network. IP correct. The Remote IP address is the RPI address. Try to run the example IP-client-SimpleRead.ino. loop on mb.connect()  without error. Have you got any ideas there is no connection?

emelianov, 2020-10-13:
Hello,
Try to use [IP-client-WriteMultiple.ino](https://github.com/emelianov/modbus-esp8266/blob/master/examples/IP-client-WriteMultiple/IP-client-WriteMultiple.ino) as a base. The sketch contains callback that reported error code of operation. So you can get more detail on issue cause.

matlesc, 2020-10-13:
Thank you for your help.  Same result with IP-client-WriteMultiple.ino.  The callback is not triggered.
The WiFi connection is OK  but mb.isConnected() always False since the sketch print "."  then "#" endlessly.
Same issue than l337quez but with the correct IP for the remote server.  Do you have any idea of what to do to characterize or  to solve

emelianov, 2020-10-13:
I'm afraid there are some connectivity problems. Point you attention to `mb.connect(...)` duration: in case of connectivity problems it will freeze for 5 seconds. Also check accessibility of ESP from RPi by ping.

matlesc, 2020-10-13:
ESP responds to the ping from the RPI. And yes, there is about 5 seconds between 2 prints  "." "#".
Maybe the connectivity issue came from the Access point configuration of the RPI. I'll search in that direction. Thank a lot for your help

matlesc, 2020-10-14:
I notice that the example is OK when uploaded to an ESP32 but fail to connect when uploaded to an ESP8266.
When the command   WiFi.mode(WIFI_STA);   is added in the setup(), then it's OK with ESP8266.
Therefore it's solved. Thx for your work and your help

emelianov, 2020-10-14:
You are welcome, @matlesc
Also thanks for donation ;).

closed 2020-10-14 by emelianov