emelianov/modbus-esp8266 #72: how send array tcp modbus

l337quez, 2020-10-07:
Hello Emilianov, could you help me. Thank you very much. I need to send this record to the server

uint16_t au16data [16] = {
  111, 222, 333, 444, 555, 6, 7, 8, 9, 10, 11, 12, 13, 14, 15, 16};

I am using this function

`uint16_t ModbusIP :: writeHreg (IPAddress ip, uint16_t offset, uint16_t * value, uint16_t numregs, cbTransaction cb, uint8_t unit)`

but I don't know what values ​​to put on it.

I know that
**IPAddress** is the ip of the server.

**offset** I think it's 16, it's the size of the array I think.

**value** I think is the data, that is au16data

**numregs** I don't know what it is

**cb** is the callback

**unit** I don't know what it is

gonzabrusco, 2020-10-07:
> **IPAddress** is the ip of the server.
> 
Correct
> **offset** I think it's 16, it's the size of the array I think.
> 
It's the address of the register where you are writing on the server. It could be any value, it depends on the modbus server you are writing to.
> **value** I think is the data, that is au16data
**Value** is the pointer to the data, in this case **au16data** (the name of the array works as a pointer).
> **numregs** I don't know what it is
The size of the array. In this case 16
> **cb** is the callback
You can pass a function that will be called when the action is finished. You can leave this blank.
> **unit** I don't know what it is
This is only used when you are talking to a modbus TCP->RTU bridge. You send the modbus IP packet to the bridge (by addressing its IP) and then with the UNIT you specify to which slave you want to send the modbus packet behind the bridge. It's a very specific situation that I don't think applies to you. You can leave this blank.

l337quez, 2020-10-11:
Hola @gonzabrusco @emelianov  no he podido enviar data. Estoy enviando data por red local, me sale este error, la variable unit que es para hacer un puente con rtu no la necesito y sino la pongo me da error, si pongo 0 o null tambien. Prometo hacer un tutorial de uso de esta libreria en mi blog. Agradezco su ayuda.

Hello I have not been able to send data. I am sending data over the local network, I get this error, the unit variable that is to make a bridge with rtu I don't need it and if I don't put it, it gives me an error, if I put 0 or null too. I promise to make a tutorial on using this library on my blog. I appreciate your help.

////////////////////////////////////////////////////////////// codigo

```

#ifdef ESP8266
 #include <ESP8266WiFi.h>
#else
 #include <WiFi.h>
#endif
#include <ModbusIP_ESP8266.h>

const int REG = 100;                    // Modbus Coils Offset
const int COUNT = 5;                    // Count of Coils
IPAddress remote(192, 168, 20, 102);    // Address of Modbus Slave device

ModbusIP mb;  // ModbusIP object

//////////////////////////////////////////////////////////////
uint16_t value[16] = {
  111, 222, 333, 444, 555, 6, 7, 8, 9, 10, 11, 12, 13 , 14, 15, 16 };
  
uint16_t offset[16] = {
  0, 0, 0, 0, 0, 0, 1, 1 };  // direccion del esclavo

uint16_t numregs[16] = {
  0, 1 };  // direccion del esclavo
  
uint16_t writeHreg(IPAddress remote, uint16_t offset, uint16_t* value, uint16_t numregs, cbTransaction cb, uint8_t unit); 
/////////////////////////////////////////////////////////////

void setup() {
  Serial.begin(115200);
 
  WiFi.begin();
  
  while (WiFi.status() != WL_CONNECTED) {
    delay(500);
    Serial.print(".");
  }
 
  Serial.println("");
  Serial.println("WiFi connected");  
  Serial.println("IP address: ");
  Serial.println(WiFi.localIP());

  mb.client();
}

bool cb(Modbus::ResultCode event, uint16_t transactionId, void* data) { // Modbus Transaction callback
  if (event != Modbus::EX_SUCCESS)                  // If transaction got an error
    Serial.printf("Modbus result: %02X\n", event);  // Display Modbus error code
  if (event == Modbus::EX_TIMEOUT) {    // If Transaction timeout took place
    mb.disconnect(remote);              // Close connection to slave and
    mb.dropTransactions();              // Cancel all waiting transactions
  }
  return true;
}

bool res[COUNT] = {false, true, false, true, true};

void loop() {
    if (!mb.isConnected(remote)) {   // Check if connection to Modbus Slave is established
        mb.connect(remote);           // Try to connect if no connection
        Serial.print(".");
    }
    if (!mb.writeCoil(remote, REG, res, COUNT, cb)) // Try to Write array of COUNT of Coils to Modbus Slave
        Serial.print("#");  
    mb.task(); // Modbus task
    delay(50); // Pushing interval

writeHreg( remote,  offset,  value,  numregs,  cb, NULL); 

    
}
```
/////////////////////////////////////////////////////////////

//////////////////////////   **error**
error1
In function 'void loop()':
modbus_tcp_IP-client-WriteMultiple:76:50: error: invalid conversion from 'uint16_t* {aka short unsigned int*}' to 'uint16_t {aka short unsigned int}' [-fpermissive]
 writeHreg( remote,  offset,  value,  numregs,  cb); 
                                                  ^
modbus_tcp_IP-client-WriteMultiple:31:10: error:   initializing argument 2 of 'uint16_t writeHreg(IPAddress, uint16_t, uint16_t*, uint16_t, cbTransaction)' [-fpermissive]
 uint16_t writeHreg(IPAddress ip, uint16_t offset, uint16_t* value, uint16_t numregs, cbTransaction cb); 
          ^
modbus_tcp_IP-client-WriteMultiple:76:50: error: invalid conversion from 'uint16_t* {aka short unsigned int*}' to 'uint16_t {aka short unsigned int}' [-fpermissive]
 writeHreg( remote,  offset,  value,  numregs,  cb); 
                                                  ^
modbus_tcp_IP-client-WriteMultiple:31:10: error:   initializing argument 4 of 'uint16_t writeHreg(IPAddress, uint16_t, uint16_t*, uint16_t, cbTransaction)' [-fpermissive]
 uint16_t writeHreg(IPAddress ip, uint16_t offset, uint16_t* value, uint16_t numregs, cbTransaction cb); 
          ^

invalid conversion from 'uint16_t* {aka short unsigned int*}' to 'uint16_t {aka short unsigned int}' [-fpermissive]

//////////////////////////   error

emelianov, 2020-10-12:
```c
writeHreg( remote, offset, value, numregs, cb);
```
- `remote` ip address of Modbus server (is correct in code above)
- `offset` Address of the first Modbus Hreg to write. Should be `uint16_t` value not an array.
- `value` values of registers to set. Should be array of at least `numregs` `uint16_t` elements.
- `numregs` number of sequental registers to be written strarting form `offset`. That is for `offset = 10` and `numregs = 3` Hregs 10, 11 and 12 will be written to Modbus server. Should be `uint16_t` not an array.
`cb` I suggest jus skip it:
```c
writeHreg( remote, offset, value, numregs);
```
`offset` and `numregs` are used incorrectly in code above. Unable to understand what you are trying to implement with these settings.

Also don't forget to change target server's IP

```c
IPAddress remote(192, 168, 20, 102);    // Address of Modbus Slave device
```

l337quez, 2020-10-13:
Hello, thanks for answering. I have solved the problems I had. One of them was that it put a different numregs value to the value array.

Example:
  uint16_t value [16] = {111,222,333,444,555,6,7,8,9,10,11,12,13,14,15,16}; // value to send

then numregs must be. numregs = 16.

now offset if it can be any number always and when it is between the range 0 and 254, which is the range that allows Modbus to work.

And numregs must be a maximum of 16, according to what I have read .. Well thanks for the help

closed 2020-10-16 by l337quez