emelianov/modbus-esp8266 #73: Is there a way of writting multiple values on local regs?

gonzabrusco, 2020-10-07:
Hello friend. I keep studying your library in depth and I can't find a way of writing multiple reg values to local values.

For example, if I want to write multiple values to remote Hreg I can use:

`uint16_t writeHreg(uint8_t slaveId, uint16_t offset, uint16_t* value, uint16_t numregs = 1, cbTransaction cb = nullptr);
`
But If I want to do the same with the local Hregs I can't.

`bool Hreg(uint16_t offset, uint16_t value);`

The same applies for the creation of the Hregs. The library lets you create multiple Hregs but all with the same value.

`bool addHreg(uint16_t offset, uint16_t value = 0, uint16_t numregs = 1);`

Maybe it would be useful to have a way of dealing with local registers by groups, like the library lets you do with remote ones.

For example:
`bool Hreg(uint16_t offset, uint16_t* value, uint16_t numregs);
bool addHreg(uint16_t offset, uint16_t* value, uint16_t numregs);

It makes life much easier if for example you are dealing with data bigger than uint16_t . For example, if you want to provide two registers for one uint32_t. You would use:

`
uint32_t data;
bool addHreg(100, (uint16_t*)&data, 2);`

The same goes if you want to update the registers:

`
data = 5453515;
bool Hreg(100, (uint16_t*)&data, 2);`

Do you think is a good idea? 

Thanks
Gonzalo

emelianov, 2020-10-08:
> ```
> bool Hreg(uint16_t offset, uint16_t* value, uint16_t numregs);
> bool addHreg(uint16_t offset, uint16_t* value, uint16_t numregs);
> ```
It's easy to implement but 
1. `addHreg()` assumed to be used only one per sketch. If it's really reasonable to extend the library API?
2. `Hreg()` to set multiple registers affected by major problem of error processing. I mean it's not clear what is right behaviuor if some of registers were not added. It's may be ok to just skip the values or spread values across existing registers or cancel complete operation.
So I suggest to leave operation to be implemented by sketch code. Just add `isDefinedHreg(uint16_t offset);` for more flexibility.

gonzabrusco, 2020-10-08:
> It's easy to implement but
>
> 1. `addHreg()` assumed to be used only one per sketch. If it's really reasonable to extend the library API?
> 2. `Hreg()` to set multiple registers affected by major [...]

1. You are right. Maybe it's not justified because it is used once. Although it would be nice to have it.
2. This is easy, the criteria should be the same as with writeHreg. It's the user responsibility to know the memory map. If the memory map has "gaps" then the user should use multiple calls of the function (once for each block of registers). That is the criteria with writeHreg correct?

emelianov, 2020-10-12:
```c
bool Hreg(uint16_t offset, uint16_t* value, uint16_t numregs = 1);
bool Coil(uint16_t offset, bool* value, uint16_t numregs = 1);
bool Ists(uint16_t offset, bool* value, uint16_t numregs = 1);
bool Ireg(uint16_t offset, uint16_t* value, uint16_t numregs = 1);
```
e7411b50f0a45d29720f496a2a8a556dab7438b1

gonzabrusco, 2020-10-13:
Thanks as always Alexander!

closed 2020-10-13 by gonzabrusco