emelianov/modbus-esp8266 #74: Each modbus TCP slave has an id or an address?

l337quez, 2020-10-16:
I need to differentiate each slave, this can be differentiated by means of the address. But I was thinking that the address is the offset. I'm not clear on that. In such case that offset is not the id or address of the slave, then what is offset

At the moment my code works, I am reading with another library, specifically with Python's Pymodbus. I need to know from which the client of the aunt you are receiving.

Currently on the server I am receiving data from 2 slaves, but I am studying how to make it read the data only from a certain address or id, I do not know what you call it from a slave and with a sampling time of 1 second then read the data from the other slave

Attached my code, excuse me for the comments of the code.

`//########################################################### Definimos Librerias
#ifdef ESP8266
 #include <ESP8266WiFi.h>
#else
 #include <WiFi.h>
#endif
#include <ModbusIP_ESP8266.h>
// usamos libreria modbus   https://github.com/emelianov/modbus-esp8266

//########################################################### Definimos variables
bool msg_connect= false;
IPAddress remote(192, 168, 11, 46); // Ip del servidor
const int LOOP_COUNT = 16;          // Contador de Bobinas 16bit
int16_t numregs=16;                 // es el tamaño del value 
const int REG = 528;                // Modbus Hreg Offset, 528 es maximo si se usa 32bit creo...
uint16_t offset=0;                  // direccion del esclavo Modbus permite 254
uint16_t value [16] = {4,222,333,444,555,6,7,8,9,10,11,12,13,10,10,10}; // valor a enviar 

//declaramos el metodo paara escribirde la clase mb
uint16_t writeHreg(IPAddress remote, uint16_t offset, uint16_t* value, uint16_t numregs);
uint16_t readHreg(IPAddress remote, uint16_t offset, uint16_t* value, uint16_t numregs );
uint16_t pullHreg(IPAddress remote, uint16_t from, uint16_t to, uint16_t numregs );
uint16_t pushHreg(IPAddress remote, uint16_t to, uint16_t from, uint16_t numregs);
//uint16_t writeHreg(IPAddress ip, uint16_t offset, uint16_t * value, uint16_t numregs, cbTransaction cb, uint8_t unit)

const int LED_COIL = 1;               // Modbus Coil Offset   ----> del ejemplo pull

// instanciamos la clase 
ModbusIP mb;  //ModbusIP object

void setup() {
  Serial.begin(115200);
  WiFi.begin("Hack_ronald","fmlaforero1234");
  
  while (WiFi.status() != WL_CONNECTED) {
    delay(800);
    Serial.print(".");
  }
 
  Serial.println("");
  Serial.println("WiFi connected");  
  Serial.println("IP address: ");
  Serial.println(WiFi.localIP());
  mb.client( );
}

uint16_t res = 0;
uint8_t show = LOOP_COUNT;

void loop() {
  
  if (mb.isConnected(remote)) {   // Check if connection to Modbus Slave is established
    msg_connect== true;
    mb.writeHreg( remote, offset, value, numregs ); // metodo para enviar data
    delay(3000);
    //Serial.println("Se ha conectado con el Servidor exitosamente");
    if (msg_connect== true){ // enviamos mensaje al servidor de conexion true, solo cuando se conecte
    mb.readHreg(remote, REG, &res);  // Initiate Read Coil from Modbus Slave
    }
  } else {
    msg_connect== false;
    mb.connect(remote, 5020);           // Try to connect if no connection
    }

   mb.readHreg(remote, offset, value, numregs );// metodo para leer data
   Serial.println(value[0]); // imprimimos el valor que esta llegando

  // mb.pullCoil(remote, LED_COIL, LED_COIL);  // Initiate Read Coil from Modbus Slave

// writeHreg(IPAddress remote, uint16_t offset, uint16_t* value, uint16_t numregs);
// readHreg(IPAddress remote, uint16_t offset, uint16_t* value, uint16_t numregs );
// pullHreg(IPAddress remote, uint16_t from, uint16_t to, uint16_t numregs );
// pushHreg(IPAddress remote, uint16_t to, uint16_t from, uint16_t numregs);

  
  mb.task();                      // Common local Modbus task  
  delay(100);                     // Pulling interval
  if (!show--) {                   // Display Slave register value one time per second (with default settings)
    Serial.println(res);
    show = LOOP_COUNT;
  }
}`

emelianov, 2020-10-17:
To make code simpler i suggest to wait response for each read/write operation:
```
mb.writeHreg( remote, offset, value, numregs ); // metodo para enviar data
// You may send next request without waiting response after [write] operation for ModbusTCP. 

uint16_t tid = mb.readHreg(remote, REG, &res);
while (mb.isTransaction(tid)) {
  mb.task();
  yield();
}
// At this point res is filled by response data (if it were no error)

tid = mb.readHreg(remote, offset, value, numregs );// metodo para leer data
while (mb.isTransaction(tid)) {
  mb.task();
  yield();
}
// Same as above. Operation is finished and value is filled by the response data

//mb.task(); // Common local Modbus task
// No need for mb.task() as all data processed above just after request is performed.
```

l337quez, 2020-10-20:
the code doesn't work for me, I have to get mb.task (); of the while. What could be happening. This is the code that works for me

```
//########################################################### Definimos Librerias
#ifdef ESP8266
 #include <ESP8266WiFi.h>
#else
 #include <WiFi.h>
#endif
#include <ModbusIP_ESP8266.h>
// usamos libreria modbus   https://github.com/emelianov/modbus-esp8266

//########################################################### Definimos variables
IPAddress remote(192, 168, 11, 46); // Ip del servidor
const int LOOP_COUNT = 16;          // Contador de Bobinas 16bit
int16_t numregs=16;                 // es el tamaño del value 
const int REG = 528;                // Modbus Hreg Offset, 528 es maximo si se usa 32bit creo...
uint16_t offset=0;                  // direccion del esclavo Modbus permite 254
uint16_t value [16] = {4,222,333,444,555,6,7,8,9,10,11,12,13,10,10,10}; // valor a enviar 
//uint8_t unit=99;
 
#define LED  2 //GPIO2
//int LED_BUILTIN = 2;  ESP32
uint16_t res = 0; //inicializar todo en cero

//declaramos el metodo paara escribirde la clase mb
//uint16_t writeHreg(IPAddress remote, uint16_t offset, uint16_t* value, uint16_t numregs, cbTransaction cb, uint8_t unit = MODBUSIP_UNIT);
uint16_t writeHreg(IPAddress remote, uint16_t offset, uint16_t* value, uint16_t numregs );
uint16_t readHreg(IPAddress remote, uint16_t offset, uint16_t* value, uint16_t numregs );
uint16_t pullHreg(IPAddress remote, uint16_t from, uint16_t to, uint16_t numregs );
uint16_t pushHreg(IPAddress remote, uint16_t to, uint16_t from, uint16_t numregs);
//uint16_t writeHreg(IPAddress ip, uint16_t offset, uint16_t * value, uint16_t numregs, cbTransaction cb, uint8_t unit)

// instanciamos la clase 
ModbusIP mb;  //ModbusIP object

void setup() {
  Serial.begin(115200);
  WiFi.begin("Hack_ronald","fmlaforero1234");
  //WiFi.begin("IM Ronald","fmlaforero1234");

    pinMode(LED, OUTPUT);
  
  while (WiFi.status() != WL_CONNECTED) {
    digitalWrite(LED,LOW);
    delay(100);     Serial.print(".");
    digitalWrite(LED,HIGH);
    delay(50);    Serial.print(".");
    digitalWrite(LED,LOW);
    delay(100);    Serial.print(".");
    digitalWrite(LED,HIGH);
    delay(50);    Serial.print(".");
    digitalWrite(LED,LOW);
    delay(100);    Serial.print(".");
    digitalWrite(LED,HIGH);
    Serial.print(".");
  }

  //prendemos el led del nodemcu, quiere decir que si hizo conexion
  digitalWrite(LED,LOW);
  Serial.println("");
  Serial.println("WiFi connected");  
  Serial.println("IP address: ");
  Serial.println(WiFi.localIP());
  mb.client( );
}

bool cb(Modbus::ResultCode event, uint16_t transactionId, void* data) { // Modbus Transaction callback
  if (event != Modbus::EX_SUCCESS)                  // If transaction got an error
    Serial.printf("Modbus result: %02X\n", event);  // Display Modbus error code
  if (event == Modbus::EX_TIMEOUT) {    // If Transaction timeout took place
    mb.disconnect(remote);              // Close connection to slave and
    mb.dropTransactions();              // Cancel all waiting transactions
  }
  return true;
}

void loop() {
  
  if (mb.isConnected(remote)) {   // Check if connection to Modbus Slave is established
    //mb.writeHreg( remote, offset, value, numregs, nullptr, 230); // metodo para enviar data
    Serial.println("conectado"); 
    mb.writeHreg( remote, offset, value, numregs);
    delay(3000); 
    }else {
    mb.connect(remote, 5020);           // Try to connect if no connection
    }

   
// writeHreg(IPAddress remote, uint16_t offset, uint16_t* value, uint16_t numregs);
// readHreg(IPAddress remote, uint16_t offset, uint16_t* value, uint16_t numregs );
// pullHreg(IPAddress remote, uint16_t from, uint16_t to, uint16_t numregs );
// pushHreg(IPAddress remote, uint16_t to, uint16_t from, uint16_t numregs);

    mb.readHreg(remote, offset, value, numregs );// metodo para leer data
   
  mb.task();      

}
```

l337quez, 2020-10-20:
Hi Emilianov, for my client I am using your library, but for the server I am using pymodbus. The previous question was to know how the id unit works.

in python I must read those unit id, but in your library I can't send a 0x00. It says that the default is 1. I don't understand.

 0x01: ModbusSlaveContext(...),

```
EXAMPLE PYTHON PYMODBUS LIBRARY
#---------------------------------------------------------------------------# 
# configure the service logging
#---------------------------------------------------------------------------# 
import logging
logging.basicConfig()
log = logging.getLogger()
log.setLevel(logging.DEBUG)

#---------------------------------------------------------------------------# 
# initialize your data store
#---------------------------------------------------------------------------# 
# The datastores only respond to the addresses that they are initialized to.
# Therefore, if you initialize a DataBlock to addresses from 0x00 to 0xFF, a
# request to 0x100 will respond with an invalid address exception. This is
# because many devices exhibit this kind of behavior (but not all)::
#
#     block = ModbusSequentialDataBlock(0x00, [0]*0xff)
#
# Continuing, you can choose to use a sequential or a sparse DataBlock in
# your data context.  The difference is that the sequential has no gaps in
# the data while the sparse can. Once again, there are devices that exhibit
# both forms of behavior::
#
#     block = ModbusSparseDataBlock({0x00: 0, 0x05: 1})
#     block = ModbusSequentialDataBlock(0x00, [0]*5)
#
# Alternately, you can use the factory methods to initialize the DataBlocks
# or simply do not pass them to have them initialized to 0x00 on the full
# address range::
#
#     store = ModbusSlaveContext(di = ModbusSequentialDataBlock.create())
#     store = ModbusSlaveContext()
#
# Finally, you are allowed to use the same DataBlock reference for every
# table or you you may use a seperate DataBlock for each table. This depends
# if you would like functions to be able to access and modify the same data
# or not::
#
#     block = ModbusSequentialDataBlock(0x00, [0]*0xff)
#     store = ModbusSlaveContext(di=block, co=block, hr=block, ir=block)
#
# The server then makes use of a server context that allows the server to
# respond with different slave contexts for different unit ids. By default
# it will return the same context for every unit id supplied (broadcast
# mode). However, this can be overloaded by setting the single flag to False
# and then supplying a dictionary of unit id to context mapping::
#
#     slaves  = {
#         0x01: ModbusSlaveContext(...),
#         0x02: ModbusSlaveContext(...),
#         0x03: ModbusSlaveContext(...),
#     }
#     context = ModbusServerContext(slaves=slaves, single=False)
#
# The slave context can also be initialized in zero_mode which means that a
# request to address(0-7) will map to the address (0-7). The default is
# False which is based on section 4.4 of the specification, so address(0-7)
# will map to (1-8)::
#
#     store = ModbusSlaveContext(..., zero_mode=True)
```

emelianov, 2020-10-22:
If you need to access spscific unit just set it when calling read/write:

```c
//...
uint8_t unit=1; // You can specify any value from 0 to 255 according to Python side
//...
mb.readHreg(remote, offset, value, numregs, nullptr, unit);
//...
```

```c
uint16_t readCoil(IPAddress ip, uint16_t offset, bool* value, uint16_t numregs = 1, cbTransaction cb = nullptr, uint8_t uint = MODBUSIP_UNIT);
uint16_t readCoil(const char* host, uint16_t offset, bool* value, uint16_t numregs = 1, cbTransaction cb = nullptr, uint8_t uint = MODBUSIP_UNIT);
uint16_t readCoil(String host, uint16_t offset, bool* value, uint16_t numregs = 1, cbTransaction cb = nullptr, uint8_t uint = MODBUSIP_UNIT);
uint16_t writeCoil(IPAddress ip, uint16_t offset, bool value, cbTransaction cb = nullptr, uint8_t uint = MODBUSIP_UNIT);
uint16_t writeCoil(IPAddress ip, uint16_t offset, bool* value, uint16_t numregs = 1, cbTransaction cb = nullptr, uint8_t uint = MODBUSIP_UNIT);
uint16_t readIsts(IPAddress ip, uint16_t offset, bool* value, uint16_t numregs = 1, cbTransaction cb = nullptr, uint8_t uint = MODBUSIP_UNIT);
uint16_t writeHreg(IPAddress ip, uint16_t offset, uint16_t value, cbTransaction cb = nullptr, uint8_t uint = MODBUSIP_UNIT);
uint16_t writeHreg(IPAddress ip, uint16_t offset, uint16_t* value, uint16_t numregs = 1, cbTransaction cb = nullptr, uint8_t uint = MODBUSIP_UNIT);
uint16_t readHreg(IPAddress ip, uint16_t offset, uint16_t* value, uint16_t numregs = 1, cbTransaction cb = nullptr, uint8_t uint = MODBUSIP_UNIT);
uint16_t readIreg(IPAddress ip, uint16_t offset, uint16_t* value, uint16_t numregs = 1, cbTransaction cb = nullptr, uint8_t uint = MODBUSIP_UNIT);
```

- `ip` IP Address of Modbus server to get registers from
- `host` Hostname fo Modbus server to get registers from
- `offset` Address of first Modbus register to read/write
- `numregs` Count of registers to read/write
- `cb` Transaction callback function. `NULL` if not used
- `unit` Modbus unit

closed 2020-10-29 by l337quez