emelianov/modbus-esp8266 #76: A way to access local registers without depending on the existence of a modbusIP or modbusRTU object?

gonzabrusco, 2020-10-25:
Hello. The title is self descriptive. I think it would be nice to have a way to access local registers without depending on the existence of a modbusIP or modbusRTU object.

Imagine you are developing a software that can enable or disable modbusTCP or modbusRTU on demand by the user, so the programmer does not know which object will exist. The modbusTCP and modbusRTU objects if enabled and then disabled, will get destroyed. So you need to access the registers to update them, but you don't know which object exist. I would be nice to have some sort of handler for accessing local registers (that are global) without needing to know beforehand which objects actually exists. I don't know if I'm being clear here. It is somewhat difficult to explain. I guess that to achieve this, it would take a big restructuring of the library. 

What I want to do is a library that has modbusTCP and modbusIP and it is used definable. And the user can change that on the fly. So I need to be able to stop them (freeing all the memory and resources). But at the same time, I want to be able to update the registers (indepedently of which modbus object exists or not). So here is the problem, If I stopped both servers (destroyed them), then I loose access to the registers. How can I keep the holding registers updated independently of which server is running?

gonzabrusco, 2020-10-25:
I think I figured it out... I could just create an object of the class "Modbus" just for having access to the registers without needing to create a ModbusIP or ModbusRTU. Right?

emelianov, 2020-10-26:
Hello Gonzalo,
That's right decision to just create "control" Modbus instance to access registers etc.
Also note that in 4.0.0 registers can be accessed using generalized notation
```c
bool addReg(TAddress address, uint16_t value = 0, uint16_t numregs = 1);
bool Reg(TAddress address, uint16_t value);
uint16_t Reg(TAddress address);
```
Example:
```c
mb.addReg(HREG(100)); // == mb.addHreg(100);
TAddress reg = HREG(100);
mb.Reg(reg, 200);
reg++; // reg now is HREG(101);
if (reg.isHreg()) // do something
```
It may be useful to build some flexible parameters-driven code.

emelianov, 2020-10-26:
Anyway it's good point for clean registers and callbacks on destroying last instance of Modbus. But I'm not sure that it's really critical thing.

gonzabrusco, 2020-10-26:
For me it is ok if registers do not get cleaned. I only needed to delete modbus objects if the user changes the settings (TCP port, serial speed or format, etc). But the registers in my program stay always the same. I think maybe it could be handy a method for clearing all the memory. In case anyone wants to delete every registers at once. **But I agree with you, don´t think is critical.**

closed 2020-11-02 by gonzabrusco