emelianov/modbus-esp8266 #78: write log when it detects a connected client

l337quez, 2020-10-29:
Hi Emiliano. I am making a server with ESP32 and I want that when a client connects to send this variable value [16] = {4,222,333,444,555,6,7,8,9,10,11,12,13,10,10,10};

From what I see in your examples the server expects something from the client. But I don't see that I have an example sending data. I want the server to stay open. At the moment the connection works, I am connecting with a raspberry pi, I know that the connection works, because it prints the client's ip by serial monitor. The problem I have is that when I send data from the client I get an error and when I put it to read it says none. yes well I'm not sending anything.

**What I really want is that when a client connects to send that array of value** 

my code
`//########################################################### Definimos Librerias
#ifdef ESP8266
 #include <ESP8266WiFi.h>
#else //ESP32
 #include <WiFi.h>
#endif
#include <ModbusIP_ESP8266.h>
// usamos libreria modbus   https://github.com/emelianov/modbus-esp8266
//########################################################### Definimos variables

// Modbus Registers Offsets
uint16_t value [16] = {4,222,333,444,555,6,7,8,9,10,11,12,13,10,10,10}; // valor a enviar 

const int LED_HREG = 16;
//ModbusIP object
ModbusIP mb;

// callback
bool cbConn(IPAddress ip) {
Serial.println(ip);
return true;
}

// Callback function to read corresponding DI
uint16_t cbRead(TRegister* reg, uint16_t val) {
  return val;
}

// Callback function to write-protect DI
uint16_t cbWrite(TRegister* reg, uint16_t val) {
  return val;
}

void setup() {
  Serial.begin(115200);
 
  WiFi.begin("Ronal","234234234234");
  while (WiFi.status() != WL_CONNECTED) {
    delay(500);
    Serial.print(".");
  }
 
  Serial.println("");
  Serial.println("WiFi connected");  
  Serial.println("IP address: ");
  Serial.println(WiFi.localIP());
  mb.onConnect(cbConn);   // conocer la ip del cliente conectado
// puerto de comunicacion 5020
  mb.server(5020);
  mb.addHreg(1, 2, 16);

//bool onSetHreg(uint16_t address, cbModbus cb = nullptr, uint16_t numregs = 1);
  mb.onSetHreg(LED_HREG, cbWrite);

} // Fin del setup
 
void loop() {

   //Call once inside loop() - all magic here
   mb.task();
   delay(10);

}`

l337quez, 2020-10-30:
hello @emelianov  

I already saw that which function that writes from the server, is this

uint16_t writeHreg (uint8_t slaveId, uint16_t offset, uint16_t value);

when it comes to the client, all clients know the ip of the server, but when it comes to the server not. Although there is a callback that if it returns, I can read it from a serial monitor, ok. I thought that ip was the slaveId. but it seems not. I have problems because apparently it is not sending the data from the server to the client.

//#################################################### Definimos Librerias
#ifdef ESP8266
 #include <ESP8266WiFi.h>
#else //ESP32
 #include <WiFi.h>
#endif
#include <ModbusIP_ESP8266.h>
// usamos libreria modbus   https://github.com/emelianov/modbus-esp8266
//############################################## Definimos variables

// Modbus Registers Offsets
uint16_t value [16] = {4,222,333,444,555,6,7,8,9,10,11,12,13,10,10,10}; // valor a enviar 
uint16_t offset=0;  
const int LED_HREG = 16;

// Declaramos el metodo writeHreg
//uint16_t writeHreg(uint8_t slaveId, uint16_t offset, uint16_t value, cbTransaction cb = nullptr);
uint16_t writeHreg(uint8_t slaveId, uint16_t offset, uint16_t value);

// Modbus Registers Offsets
const int TEST_HREG = 100;
IPAddress slaveId; // ip del cliente que es la Raspberry pi

//ModbusIP object
ModbusIP mb;

// callback
bool cbConn(IPAddress ip) {
Serial.println(ip);
slaveId= ip;
return true;
}

// Callback function to read corresponding DI
uint16_t cbRead(TRegister* reg, uint16_t val) {
  return val;
}

// Callback function to write-protect DI
uint16_t cbWrite(TRegister* reg, uint16_t val) {
  return val;
}

  void setup() {
  Serial.begin(115200);
 
  WiFi.begin("Hack_ronald","fmlaforero1234");
  
  while (WiFi.status() != WL_CONNECTED) {
    delay(500);
    Serial.print(".");
  }
 
  Serial.println("");
  Serial.println("WiFi connected");  
  Serial.println("IP address: ");
  Serial.println(WiFi.localIP());
  mb.onConnect(cbConn);   // conocer la ip del cliente conectado
// puerto de comunicacion 5020
  mb.server(5020);
  mb.addHreg(TEST_HREG, 0xABCD);

} // Fin del setup
 
void loop() {

   // al parecer conoce el slaveid que es la ip del servidor. esto porque es sincrono, pero en tal caso se pueden usar
   // los otros campo como unit y se pone el unit id que tenga el server pienso yo
   mb.writeHreg(slaveId, offset, value);

   //Call once inside loop() - all magic here
   mb.task();
   delay(10);

}

l337quez, 2020-10-30:
@emelianov  ok I understand. How can I do so that when a request arrives, the server sends the value array. What I have tried has not worked for me, thanks for answering

emelianov, 2020-11-01:
Probably, yo may send read request from client? I mean request multiple registers containing array values.
Something like
```c
vois setup();
//...
for (size_t i=0; i < 16; i++)
  mb.addHreg(i, value[i]); 
//...
}
```
In example above Hreg 0-15 will be filled with array values. Then client will be able to read all these 16 Hreg to get the array values.

l337quez, 2020-11-03:
many thanks. Thanks for responding quickly, I am grateful. I made a post on my blog about some things. The publication will feed it little by little.

https://ronaldl337.wordpress.com/2020/11/02/comunicacion-tcp-modbus-con-arduino/

closed 2020-11-03 by l337quez